adafruit/circuitpython #4884: List comprehensions and function call performance penalty is high

matemaciek, 2021-06-11:
**Description**

I was looking for ways to optimize some of my programs. It turns out that function calls and list comprehensions are pretty costly. Is there any syntax to do some inlining maybe? 

**Firmware**

```python
Adafruit CircuitPython 6.3.0 on 2021-06-01; Raspberry Pi Pico with rp2040
```

**Code/REPL**
Let's start with this, which runs at `41 Hz`:

```python
import time

def mul(x, p):
    return x*x*p

def vect(x):
    return [mul(x, p) for p in [2,3,5]]

checkpoint = int(time.monotonic())
cycles = 0
while True:
    v = 0
    for i in range(100):
        v += sum(vect(i))
    t = time.monotonic()
    if checkpoint < int(t):
        print(v)
        print(cycles, 'Hz')
        checkpoint = int(t)
        cycles = 0
    else:
        cycles = cycles + 1

```
Manually expanding list comprehension boosted performance to `102 Hz`:
```python
def vect(x):
    return [mul(x, 2), mul(x, 3), mul(x, 5)]
```
Inlining `mul` boosted to `154 Hz`:
```python
def vect(x):
    return [x*x*2, x*x*3, x*x*5]
```
And after getting rid of `mul` and `sum` I got `275 Hz`:
```python
import time

checkpoint = int(time.monotonic())
cycles = 0
while True:
    v = 0
    for i in range(100):
        v += i*i*2 + i*i*3 + i*i*5
    t = time.monotonic()
    if checkpoint < int(t):
        print(v)
        print(cycles, 'Hz')
        checkpoint = int(t)
        cycles = 0
    else:
        cycles = cycles + 1
```
 And now I have a dilemma, whether to write good code or repeated-but-fast code (I guess if I want both I should switch to C). Can you think of something obvious that I'm missing? Or maybe I can help to make such scenarios faster?

dhalbert, 2021-06-11:
When the code in the function is small, the overhead of the call becomes significant. Also you are avoiding storage allocation: you are no longer creating lists in your last example.

Try using `monotonic_ns()` instead of `monotonic()`, to avoid software floating point.

Python has never been particularly fast. If you try the same thing in CPython, you may get similar relative differences.

I'm going to remove the bug label because this is not a bug.

dhalbert, 2021-06-15:
Closing for now.

closed 2021-06-15 by dhalbert