adafruit/circuitpython #4886: COM port for nRF52840 Micro Dev Kit USB Dongle stopped working after 6.0.1

ThomasAtBBTF, 2021-06-13:
**Firmware**

All CircuitPython versions _after_ :

```python
Adafruit CircuitPython 6.0.1 on 2020-12-28; MakerDiary nRF52840 MDK USB Dongle with nRF52840
```
are **installing and running fine** on the nRF52840 Micro Dev Kit USB Dongle. 
This is observed by checking ```boot_out.txt``` in the root of the CIRCUITPY drive.

**However**

Starting with version ```6.1.0-beta 3``` up to version ```7.0.0-alpha.3``` the **comport for the REPL is not working** anymore.
Trying to connect with PUTTY the terminal-window will not open and after a long time out a error message window shows up.
Note that in device manager the comport shows up "normal" for all CircuitPython versions.
Other than that CircuitPython itself seems to work fine as code.py seems to gets loaded and boot_out is written.

**Additional Info**

I am not so sure about the whole configuration of my board as I am not sure about the UF2 Bootloader.
I created a UF2 file from ```mdk_nrf52840_dongle_bootloader-0.5.1_s140_6.1.1.hex``` and after copying this things started to work. e.g. the drive ```circuitpy``` showed up after booting the board.
But when "double tapping" reset a drive named ```MDK-DONGLE``` is showing up and ```INFO_UF2.TXT``` there reads:
```
UF2 Bootloader 0.2.13-42-g82464f9-dirty lib/nrfx (v2.0.0) lib/tinyusb (legacy-1500-g23df777b) s140 6.1.1
Model: nRF52840 MDK USB Dongle
Board-ID: nRF52840-Dongle-v1
Date: Feb  3 2020
```
Why is there no ```mdk_nrf52840_dongle_bootloader-0.5.1_s140_6.1.1.uf2``` available for download?
What is ```update-mdk_nrf52840_dongle_bootloader-0.5.1_nosd.uf2``` doing?
Does anyone else have later versions of CircuitPython running on the nRF52840 Micro Dev Kit USB Dongle?
What is the content of your ```INFO_UF2.TXT``` ?
Does anyone have the same problem on other nRF52840 boards? (I have a nRF ItsyBitsy on order)
Can this be configuration problem, which can be fixed with boot.py?

jerryneedell, 2021-06-13:
I just loaded 6.3.0
```
uto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.3.0 on 2021-06-01; MakerDiary nRF52840 MDK USB Dongle with nRF52840
>>> 
>>> 

```
7,0 alpha.3
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.3 on 2021-06-03; MakerDiary nRF52840 MDK USB Dongle with nRF52840
>>> 

```

and the current tip of main
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.3-45-g5f81f9ed1 on 2021-06-13; MakerDiary nRF52840 MDK USB Dongle with nRF52840
>>> 

```

All work as expected via "screen" from my Linux system.

My bootloader is:
```
UF2 Bootloader 0.3.2-184-g64ea299 lib/nrfx (v2.0.0) lib/tinyusb (0.6.0-272-g4e6aa0d8) lib/uf2 (heads/master)
Model: MDK nRF52840 USB Dongle
Board-ID: nRF52840-Dongle-v1
SoftDevice: S140 version 6.1.1
Date: Nov 25 2020
```

ThomasAtBBTF, 2021-06-13:
Thank you!
Great that this is not a common problem but most likely a configuration / code problem on my board.
Strange is only that the REPL-COM-PORT stopped working in CircuitPython after 6.0.1.

@jerryneedell How did you get the bootloader version 0.3.2 onto your board?
Did you update it yourself or was the board delivered to you like this?
Can you point me to a download link?

jerryneedell, 2021-06-13:
Since your SD is already s140-6.1.1 you may be able to just use the update .uf2  https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/download/0.5.1/update-mdk_nrf52840_dongle_bootloader-0.5.1_nosd.uf2

I built the bootloader locally  -- I cloned the repository https://github.com/adafruit/Adafruit_nRF52_Bootloader
I just updated it to the latest version 0.5.1-2
```
UF2 Bootloader 0.5.1-2-g881dac1 lib/nrfx (v2.0.0) lib/tinyusb (0.9.0-22-g7cdeed54) lib/uf2 (heads/master)
Model: MDK nRF52840 USB Dongle
Board-ID: nRF52840-Dongle-v1
SoftDevice: S140 version 6.1.1
Date: Jun 13 2021
```
using the dfu serial load command
`make BOARD=mdk_nrf52840_dongle SERIAL=/dev/ttyACM0 dfu-flash`
which loads the .zip file via the adafruit-nrfutil comand
```
adafruit-nrfutil --verbose dfu serial --package _build/build-mdk_nrf52840_dongle/mdk_nrf52840_dongle_bootloader-0.5.1-2-g881dac1_s140_6.1.1.zip -p /dev/ttyACM0 -b 115200 --singlebank --touch 1200
```
You should be able to use this to load the .zip file in the repository release

With the new bootloader -- the latest CP still works OK
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.3-45-g5f81f9ed1 on 2021-06-13; MakerDiary nRF52840 MDK USB Dongle with nRF52840
>>> 

```

jerryneedell, 2021-06-13:
FYI -- I tried just copying the update...nosd.uf2 file to my  board and it still works....

jerryneedell, 2021-06-13:
@ThomasAtBBTF  I'm curious if you made any progress on this and if I can close this issue?
I should have started by adding a warning that it is good to have a J-Link to rescue things when messing with the bootloader!
I hope you  got it working.

ThomasAtBBTF, 2021-06-13:
Just copying the different UF2 files didnot help and did not change anything.

Trying this:

``` 
C:\Users\frieh\AppData\Roaming\Python\Python37\Scripts>adafruit-nrfutil --verbose dfu serial --package mdk_nrf52840_dongle_bootloader-0.5.1_s140_6.1.1.zip -p COM26 -b 115200 --singlebank --touch 1200
Upgrading target on COM26 with DFU package C:\Users\frieh\AppData\Roaming\Python\Python37\Scripts\mdk_nrf52840_dongle_bootloader-0.5.1_s140_6.1.1.zip. Flow control is disabled, Single bank, Touch 1200
Touched serial port COM26
Opened serial port COM26
Starting DFU upgrade of type 3, SoftDevice size: 151016, bootloader size: 39000, application size: 0
Sending DFU start packet
Sending DFU init packet
Sending firmware file
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
############
Activating new firmware

DFU upgrade took 21.173063039779663s
Device programmed.
``` 

Initially looked good but after reboot I have a **bricked** board.

Only tomorrow in the office I can try to bring back the board back to life using a segger adapter.

jerryneedell, 2021-06-13:
Ugh! Sorry about that -- So now if you double Tap RESET, it does not mount the Bootloader drive?
I loaded it to mine that way several times this morning. I don't know what else to suggest.
Good luck!

tannewt, 2021-06-22:
Was this also fixed by #4900?

ThomasAtBBTF, 2021-06-22:
I am pretty sure that this is also fixed by #4900, but I can not confirm as my only nRF52840 Micro Dev Kit USB Dongle is bricked after messing around with the bootloader. 
I ordered more in China but it takes a couple of weeks for them to arrive.
We can close this for now and I will reopen in the unlikely case, that #4900 does not also fix this.

closed 2021-06-22 by ThomasAtBBTF