adafruit/circuitpython #4888: print throws error with "flush" argument

jerryneedell, 2021-06-14:
Just ran across this on  CLUE (nrf52840)
using the "flush" argument for print throws an error

```
Adafruit CircuitPython 7.0.0-alpha.2-527-g47947dc23 on 2021-05-13; Adafruit CLUE nRF52840 Express with nRF52840
>>> print("hi",flush=True)
hi
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] Invalid argument
>>> 
```

jerryneedell, 2021-06-14:
hmmm -- may just be on the nRF52840
```

Adafruit CircuitPython 7.0.0-alpha.3-45-g5f81f9ed1 on 2021-06-13; Adafruit Feather M0 Adalogger with samd21g18
>>> print("hi",flush=True)
hi
>>> 

```

jerryneedell, 2021-06-14:
Nope....
```

Adafruit CircuitPython 7.0.0-alpha.3-45-g5f81f9ed1 on 2021-06-13; Adafruit Feather M4 Express with samd51j19
>>> 
>>> print("hi",flush=True)
hi
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] Invalid argument
>>> 

```

dhalbert, 2021-06-14:
I believe this is due to:
https://github.com/adafruit/circuitpython/blob/84e1d77664bc7425673f33395a7dd0e53ab0f063/py/stream.c#L494-L496
but I don't know why there is no `ioctl` support for stdout on those ports. It may just be a typo somewhere.

dhalbert, 2021-06-15:
This works on Metro M0 and fails on Metro M4. I believe the difference is whether `MICROPY_PY_IO` is `0` (M0) or `1` (M4).

jerryneedell, 2021-06-16:
MICROPY_PY_IO gets enabled if JSON is enabled. 
https://github.com/adafruit/circuitpython/blob/main/py/circuitpy_mpconfig.h#L516
I think JSON is enable for all the "full" builds so that would be consistent with what we are seeing.

Is this likely an upstream issue?

jerryneedell, 2021-06-16:
It looks like MICROPY_PY_IO gets enabled if JSON is enabled.
https://github.com/adafruit/circuitpython/blob/main/py/circuitpy_mpconfig.h#L516
I think JSON is enable for all the "full" builds so that would be consistent with what we are seeing.

Is this likely an upstream issue?

dhalbert, 2021-06-16:
Yes, it is all a micropythonism. It provides the io module, whose code also seems to provide extra functionality for stdin/stdout (maybe...), and is also needed by `json`.

tannewt, 2021-06-22:
Why does PY_IO disable `flush`? CPython always has it: https://docs.python.org/3/library/functions.html#print

Flameeyes, 2021-08-01:
FWIW this is also affecting the FeatherS2 — and it's a regression from CircuitPython 6.2.

```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Feather M4 Express with samd51j19
>>> print("foo", flush=True)
foo
```

```
Adafruit CircuitPython 7.0.0-alpha.5 on 2021-07-21; FeatherS2 with ESP32S2
>>> print("foo", flush=True)
foo
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] Invalid argument
```

tannewt, 2021-08-04:
This is actually more correct. The change comes from [the 1.13 merge](https://github.com/adafruit/circuitpython/pull/4720/files?file-filters%5B%5D=.c#diff-d10683776c4bfa5c707b7af8a20a136c9d59131e9803ba1bc547020e43fe0704L93) where I removed the "pretend to flush" https://github.com/adafruit/circuitpython/pull/1555/files#diff-d10683776c4bfa5c707b7af8a20a136c9d59131e9803ba1bc547020e43fe0704R92.

Basically, the old version was lying to you that it actually flushed. So I propose we leave it as-is. We're on a major version boundary so breaking something "that used to work" is ok.

tannewt, 2021-08-11:
I don't think it's worth worrying about this. Very little code uses flush. If you have a reason you'd like to to work, then please file a new issue. If needed, we can make it work without IO instead of faking the flush.

closed 2021-08-11 by tannewt