adafruit/circuitpython #4892: ESP32-S2 webserver

reppad, 2021-06-16:
Hello makers,
Maybe I missed it but I couldn't find any way to use an ESP32-S2 board as a webserver with CircuiPython.
There is a library for the ESP32 throught SPI (adafruit_esp32spi_wsgiserver), I would very much like to have something comparable for the ESP32-S2, maybe in the wifi module of the ESP32-S2 port or via a dedicated library? 
Thanks for your great job on CircuitPython and microcontroller support in general!

Neradoc, 2021-06-16:
There's a port of wsgiserver for the S2, I think I should add it to the community bundle, but it's usable as is.
https://github.com/Neradoc/circuitpython-native-wsgiserver

reppad, 2021-06-16:
Thank you Neradoc, I'll look at it.
Indeed it would deserve to be in the bundle with an example which displays a web page !

tannewt, 2021-06-22:
Closing. Thanks @Neradoc.

closed 2021-06-22 by tannewt