adafruit/circuitpython #4893: 7.0.0-alpha.3 time.mktime() returns wrong seconds

anecdata, 2021-06-17:
```py
Adafruit CircuitPython 6.3.0 on 2021-06-01; Adafruit FunHouse with ESP32S2
>>> import time
>>> 
>>> s = time.mktime((2021, 06, 16, 20, 0, 0, -1, -1, -1))
>>> s
1623873600
>>> 
```
vs.
```py
Adafruit CircuitPython 7.0.0-alpha.3 on 2021-06-03; Adafruit FunHouse with ESP32S2
>>> import time
>>> 
>>> s = time.mktime((2021, 06, 16, 20, 0, 0, -1, -1, -1))
>>> s
677188800
>>> 
```

dhalbert, 2021-06-17:
This might be an epoch thing. Does that work out to seconds since 1/1/2000?

Neradoc, 2021-06-17:
The epoch has changed from 1970 to 2000 with the Micropython updates. Technically this is not a bug I guess, since the epoch in python is port dependent. It causes issues with existing libraries, like when they try to update the time from a unix epoch online.

It can be changed back in Circuitpython with `MICROPY_EPOCH_IS_1970`.
It can be tested in libraries by looking at `time.mktime((2000, 1, 1, 0, 0, 0, 0, 0, 0))`.
The time difference is `NTP_TO_UNIX_EPOCH = 2_208_988_800`.

anecdata, 2021-06-17:
So it's an intended breaking change, and we should tell users and library writers to add 30 years before doing more manipulations to avoid `OverflowError: timestamp out of range for platform time_t`?

Neradoc, 2021-06-17:
Unix timestamps based on 1/1/1970 need to have 30 years removed from them (the exact number of seconds above, taking leap years and such into accounts). For example this code from a clock demo converting the NTP 1900-based given time into the 1970 epoch representation before calling `localtime` has to be updated:
```py
def get_ntp_time(pool):
    packet = bytearray(48)
    packet[0] = 0b00100011

    for i in range(1, len(packet)):
        packet[i] = 0

    with pool.socket(pool.AF_INET, pool.SOCK_DGRAM) as sock:
        sock.settimeout(None)
        sock.sendto(packet, (NTP_SERVER, NTP_PORT))
        sock.recv_into(packet)
        destination = time.monotonic_ns()

    seconds = struct.unpack_from("!I", packet, offset=len(packet) - 8)[0]
    monotonic_start = seconds - 2_208_988_800 - (destination // 1_000_000_000)
    return time.localtime(time.monotonic_ns() // 1_000_000_000 + monotonic_start + TZ_OFFSET)
```
With something like this for CP7:
```py
    monotonic_start = monotonic_start - 946_684_800
```
Or like that for compatibility:
```py
    # time.mktime((2000, 1, 1, 0, 0, 0, 0, 0, 0)) is 946_684_800 in CP6 and zero in CP7
    monotonic_start = monotonic_start + time.mktime((2000, 1, 1, 0, 0, 0, 0, 0, 0)) - 946_684_800
```
(Which can be optimized by calculating the total delta with 1900 once in advance).

But for `time.localtime()` to work with dates before 2030, the Overflow Error test must be removed. Now I don't actually know the reason for that error, was the code actually unable to correctly compute times bellow 2000 ? In that case is removing the error as I proposed in that PR enough or does it need more scrutiny ?

Also I think the value of `TIMEUTILS_SECONDS_1970_TO_2000` could be exposed in Circuitpython in some way, so that people don't have to find it out and to avoid mistakes (like using `60*60*24*365*30` for 30 years, missing 7 leap days). Libraries like datetime might be the right place for that, and for abstracting away the CP6/CP7 differences (until CP6 is deprecated or something).

tannewt, 2021-06-22:
I think my preference would be to keep the old behavior by changing `MICROPY_EPOCH_IS_1970`. Any objections?

caternuson, 2021-06-30:
No objections from me. In general, preference is to do whatever cpython does. AFAIK, that's still 1970 based?

Neradoc, 2021-06-30:
I agree, since I don't know that switching to 2000 has any advantage (at least until 2038) I don't see why we should break compatibility with existing code and the Unix Epoch.

Note that technically the epoch is platform dependent in CPython <https://docs.python.org/3/library/time.html>
> The epoch is the point where the time starts, and is platform dependent. For Unix, the epoch is January 1, 1970, 00:00:00 (UTC). To find out what the epoch is on a given platform, look at time.gmtime(0).

In fact Circuitpython seems to break from CPython (at least the docs) by not having `time.gmtime()` (which would be identical to `localtime` I guess since there is no notion of timezone either).

anecdata, 2021-06-30:
Agreed. Unless there's a compelling reason, we should avoid breakage.

dhalbert, 2021-07-02:
So we are looking for a PR to restore `MICROPY_EPOCH_IS_1970 = 1`, right? I will make such a PR.

PR #4961 (2021-07-19): Restore default epoch to 1970
Fixes #4893. Rolls back changing the epoch to 2000, which happened due to #4744.
closed 2021-07-19 by jepler