adafruit/circuitpython #4894: COM port for ItsyBitsy NRF52840 Express stopped working after 6.0.1

ThomasAtBBTF, 2021-06-17:
After filing issue #4886 where I finally bricked my board. I got two ItsyBitsy NRF52840 Express boards from a vendor in Berlin / Germany.

And to my surprise I can report exactly the same findings as with the nRF52840 Micro Dev Kit USB Dongle !

My ```INFO_USB.TXT``` looks like the following:
```
UF2 Bootloader 0.2.13 lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-755-g55874813) s140 6.1.1
Model: Adafruit ItsyBitsy nRF52840 Express
Board-ID: nRF52840-ItsyBitsy-revA
Date: Sep 25 2019
```
It seems to me that these were not "brand new"  ItsyBitsy NRF52840 Express boards.

Running the example code from https://learn.adafruit.com/ble-hid-keyboard-buttons-with-circuitpython works fine on CP 6.0.1
(as it did on the Micro Dev Kit USB board)

I don't want to mess for the moment with the bootloader update as this did brick the previous board.

I am happy to accept / understand that the root cause of the problem may lie in the configuration of my Windows 10 system, but I am curious what can be wrong with my computer when all versions up to and including 6.0.1 are working fine and as soon as I install CP 6.1.0 or later the COM port is not working anymore.  After downgrading to 6.0.1 all become fine again. Without any change to the Windows Computer.

I would like to get step by step suggestions here how to proceed to update the ItsyBits in a meaningful way!

Any ideas?

ladyada, 2021-06-17:
you dont need to update the bootloader, but you should be using the latest releases. did you try 7.0.0 alpha?

dhalbert, 2021-06-17:
Do you know how to do a bisect? That would narrow down what the issue is, if it does not also work with 7.0.0-alpha.3.

dhalbert, 2021-06-17:
@ThomasAtBBTF Do you have a Raspberry Pi or some other non-Windows computer to try it on to try to further narrow the problem?

ThomasAtBBTF, 2021-06-17:
> Do you know how to do a bisect? That would narrow down what the issue is, if it does not also work with 7.0.0-alpha.3.

Yes, I do know bisect and the O-Notation effort of O(log N). But with low n-sample count log-n is nearly equivalent to O(n)

Anyways: I started with the current version 6.3 and when I realized that I got the same results as with the other boards.
Then I found that 5.x was working and then I tried 6.0.1 which worked for me before.

ThomasAtBBTF, 2021-06-17:
> @ThomasAtBBTF Do you have a Raspberry Pi or some other non-Windows computer to try it on to try to further narrow the problem?

Sure I have a Raspberry PI but it is running OctoPrint for a 3D-Printer.

But I have s stable of FriendlyArm NanoPi Neo Air's running different Ubuntu versions. 
Also I can revive a old MacBook to check if Versions of after CP 6.0,1 are working with for example MiniTerm....

But this does not answer the question what is different between version CP 6.0.1 and later versions!

ThomasAtBBTF, 2021-06-17:
> you dont need to update the bootloader, but you should be using the latest releases. did you try 7.0.0 alpha?

I will try 7.0.0 alpha next. But as I wrote in issue #4886 this version exhibited the same behaviour on the MicroDev board using version 7.0.0 alpha.

I don't want to put unnecessary work on you.

But I think you should investigate what changed after version 6.0.1 for the REPL-Com port.

dhalbert, 2021-06-17:
> But this does not answer the question what is different between version CP 6.0.1 and later versions!

Right, I just want to know if it's specific to your Windows box or not.

ThomasAtBBTF, 2021-06-17:
Sure, I am with you on thatl
I will report back tomorrow morning my time here in Germany.

ThomasAtBBTF, 2021-06-17:
> you dont need to update the bootloader, but you should be using the latest releases. did you try 7.0.0 alpha?

[image]

I copied ````adafruit-circuitpython-itsybitsy_nrf52840_express-en_US-7.0.0-alpha.3.uf2 ````
to my ItsyBitsy NRF52840 Express and got the same result as with any other version above CP 6.0.1

Sorry for my German "device manager output" but I think you can see what I get.

ThomasAtBBTF, 2021-06-17:
And here the content of the boot_out.txt after the update:
```Adafruit CircuitPython 7.0.0-alpha.3 on 2021-06-03; Adafruit ItsyBitsy nRF52840 Express with nRF52840```

I also tried CP 6.0.1 and CP 7.0.0-alpha.3 on the computer of my wife with a fresh installed PUTTY.
The results were the same as on my "old" Windows machine.

What I noticed between CP 6.0.1 and  CP 7.0.0-alpha.3 that the COM port changed on the computer # of my wife.
So there must be a difference after CP 6.0.1 on the implementation of the USB-CDC implementation.
Or?

And from a personal or "birds-eye" perspective: I am always amazed how slightly confrontative and/or untrustful threads can become.
I can only say: I want to make things better and less problematic for everyone.

dhalbert, 2021-06-17:
If you felt we were being annoyed I assure you that was not the case at all. Apologies if you felt that way. We are just trying to divide and conquer the problem.

ThomasAtBBTF, 2021-06-18:
**Good news under Linux !**

The ItsyBitsy NRF52840 Express shows up as ```/dev/ttyACM0```
[image]

And works with CP 6.0.1:
[image]

And also with CP 7.0.0-alpha.3
[image]

This was, kind of, to be expected as **jerryneedell** confirmed that also for #4886

ThomasAtBBTF, 2021-06-18:
Additional info:

This is how the ItsyBitsy NRF52840 Express shows up running CP 6.0.1:
[image]

And this is running CP 7.0.0-alpha.3:
[image]

And this in UF2 BOOTLOADER mode:
[image]

hathach, 2021-06-18:
hmm, I think it is related to #4440, it probably got too long and not responding. I am trying to troubleshoot it again to see if  I could figure out something. Please be patient

ThomasAtBBTF, 2021-06-18:
> hmm, I think it is related to # 4440, it probably got too long and not responding. I am trying to troubleshoot it again to see if I could figure out something. Please be patient

No problem, I will be patient.

hathach, 2021-06-18:
> > hmm, I think it is related to # 4440, it probably got too long and not responding. I am trying to troubleshoot it again to see if I could figure out something. Please be patient
> 
> No problem, I will be patient.

Good news, after tons of trying, I seems to found a fix,it is indeed race condition within nRF5x USB DMA. The issue depends on the timing of usb command sequence from host, and it is coincidence that it only happens with windows driver (not Linux). I will need to do a proper testing with PR, but if you have time, please give this uf2 a try.

[itsybitsy-cpy.zip](https://github.com/adafruit/circuitpython/files/6677165/itsybitsy-cpy.zip)

ThomasAtBBTF, 2021-06-18:
Wow, this was fast and required very little patience on my side!
The firmware.uf2 in the above zip-file **works for fine for me** on my windows machine.

Thank you very much for sharing your test-build.

hathach, 2021-06-18:
> Wow, this was fast and required very little patience on my side!
> The firmware.uf2 in the above zip-file **works for fine for me** on my windows machine.
> 
> Thank you very much for sharing your test-build.

Great, thanks for testing, please continue to use that for awhile to see if there is any `side-effect`, I will wrap up the fix with PR. It could take a day or so. It would be great if you could give more feedback after a few days of usage :)

ThomasAtBBTF, 2021-06-18:
Will do!

hathach, 2021-06-18:
@ThomasAtBBTF this is the final uf2 that generated by PR #4900 Please try it out to see if that "still" solve problems.
[itsy-pr4900.zip](https://github.com/adafruit/circuitpython/files/6678993/itsy-pr4900.zip)

ThomasAtBBTF, 2021-06-19:
Happy to confirm that **it works fine** for me,
@hathach: Thank You!

ThomasAtBBTF, 2021-06-19:
Also tested the usb_cdc.data object which also works fine.

hathach, 2021-06-19:
thank for confirmation, glad that works out for you.

PR #4900 (2021-06-21): update tinyusb to fix nrf control race condition
fix race condition with control since TASKS_EP0RCVOUT also require EsyDMA. Should fix #4894 and fix #4440 .
closed 2021-06-21 by dhalbert