adafruit/circuitpython #4895: USB MIDI sending extra 1-byte messages in CP 7.x

todbot, 2021-06-17:
**Firmware**

```
- Adafruit CircuitPython 7.0.0-alpha.3 on 2021-06-03; Adafruit QT Py M0 Haxpress with samd21e18
- Adafruit CircuitPython 7.0.0-alpha.3 on 2021-06-03; Raspberry Pi Pico with rp2040
```

**Code/REPL**

```py
import time, usb_midi, adafruit_midi
from adafruit_midi.note_on import NoteOn
from adafruit_midi.note_off import NoteOff
midi = adafruit_midi.MIDI( midi_in=usb_midi.ports[0], midi_out=usb_midi.ports[1] )
while True:
    midi.send(NoteOn("A4",10))
    time.sleep(1)
    midi.send(NoteOff("A4"))
    time.sleep(1)
```

**Behavior**

Extra "invalid" truncated MIDI message sent for every `midi.send()`.  
CircuitPython 6.3.0 does not emit this extra message.

**Description**

Originally reported in [#help-with-circuitpython discord](https://discord.com/channels/327254708534116352/537365702651150357/855175670241034270) and @dhalbert asked me to submit issue. 

In [MIDI Monitor](http://www.snoize.com/MIDIMonitor/) on MacOS (11.4 Big Sur), the error looks like this:
[image]

This does not appear to affect most MIDI-consuming programs, and some other MIDI listeners like [Pocket MIDI](https://www.morson.jp/pocketmidi-webpage/) do not display the extra truncated messages.

**Additional Info**

CircuitPython 6.3.0 does not emit this extra message.

dhalbert, 2021-06-17:
@hathach is there anything in recent TinyUSB changes that might have caused this? I will also look at the USB descriptors, which may have changed inadvertently when I did the dynamic USB descriptors in 7.0.0. Thanks.

todbot, 2021-06-18:
To help honing in on the bug, it appears in both 7.x alphas:
- `Adafruit CircuitPython 7.0.0-alpha.2 on 2021-05-11;...`
- `Adafruit CircuitPython 7.0.0-alpha.1 on 2021-04-06;...`  (requires 6.x style mpys)

todbot, 2021-06-18:
If you're on Windows, [MIDI-OX](http://www.midiox.com/) is a good MIDI Monitor. It also will show the issue, in potentially more clarity than MIDI Monitor. It prints out:
```
90 45 0A
90 0A 00
80 45 00
80 00 00
```
as seen in the screenshot below.
[image]

hathach, 2021-06-18:
> @hathach is there anything in recent TinyUSB changes that might have caused this? I will also look at the USB descriptors, which may have changed inadvertently when I did the dynamic USB descriptors in 7.0.0. Thanks.

There has been too many changes to tinyusb, I did fix an MIDI buffer overflow recently, maybe there is other buffer issue with MIDI as well. I will test this out

dhalbert, 2021-06-24:
This problem is still present in the latest `main` as of this writing, so the updating of TinyUSB did not fix this. There were significant changes to MIDI storage management when dynamic USB descriptors were added, so that is a place to look as well.

hathach, 2021-06-24:
I will check this out once I got time, probably next week

jedgarpark, 2021-07-08:
I'm seeing this extra byte on Macropad RP2040 as well.

hathach, 2021-07-09:
I tried with tip of main on metro m0 with above code and got memory error as below. Just switch to test with M4 for now.

```C
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 1, in <module>
  File "/lib/adafruit_midi/__init__.py", line 29, in <module>
MemoryError: 
```

hathach, 2021-07-12:
I run MIDI-OX on VM and see the same output as reported with M4, but I don't know MIDI enough to know where is the issue. Could anyone help to point it out ?

[image]

todbot, 2021-07-12:
@hathach, the "90 45 0A" and "80 45 00" is the valid data being sent.  All other data in that screenshot are the errors. (e.g. "90 0A 00" and "80 00 00")

hathach, 2021-07-12:
> @hathach, the "90 45 0A" and "80 45 00" is the valid data being sent.
> All other data in that screenshot are the errors. (e.g. "90 0A 00" and "80 00 00")

AH ok, got it, sorry, I am not familiar with MIDI and wasn't too sure.

todbot, 2021-07-12:
This almost looks like an off-by-one error on a send buffer. Most MIDI USB packets are 4-bytes (cable number, midi status, data1, data2) but it's like it's got an extra byte in there that's being sent out with pre-existing cable number and status byte

hathach, 2021-07-12:
thanks for your extra information, I will take a closer look at this tomorrow and try to fix it asap.

hathach, 2021-07-13:
This should be fixed by #4995 , @todbot please give it a try when you have time :)

closed 2021-07-13 by tannewt

todbot, 2021-07-13:
I verify this issue that was present in 7.0.0-alpha.4 and is now fixed in the CI builds.
Tested with RP2040 and SAMD21 builds.
Thanks @hathach!

PR #4995 (2021-07-13): update tinyusb to fix midi write off by one
should fix #4895 . 

Note as side effect, tinyusb update also include RP2040 double buffer https://github.com/hathach/tinyusb/pull/891 which could increase throughput for rp2040.