adafruit/circuitpython #4899: Save space by removing slots from property proxies when possible

dhalbert, 2021-06-18:
A typical read-only property object in `shared-bindings` looks like this:
```c
const mp_obj_property_t keypad_keys_num_keys_obj = {
    .base.type = &mp_type_property,
    .proxy = {(mp_obj_t)&keypad_keys_get_num_keys_obj,    // getter
              MP_ROM_NONE,                                // setter
              MP_ROM_NONE},                               // deleter
};
```
There are a lot of these kinds of properties in the native modules, several hundred right now, so about 1200 bytes altogether for all modules. We could create a new `mp_obj_getter_only_property_t` that has a proxy that only contains a getter, and corresponding `mp_type_getter_only_property`. That would save hundreds of bytes, though of course smaller builds have fewer properties.

Deleters are also very rare even in native modules, so we could add another type that is only getter/setter. I did not count up the savings for those yet.

Checking for `mp_type_property` is done only in a few places, so those places would need extra code, but not a lot.

d-c-d, 2021-06-18:
Are these tables hand written?

If machine generated - many tweeks could be measured

dhalbert, 2021-06-18:
There is a table for each property in a native module. They are hand-written. I eyeballed the savings just with some grepping and wc.

dhalbert, 2021-06-19:
@jepler tried this in #4901, without a net gain on small builds. My original suggestion above was to create new property types corresponding to the contents of the `.proxy` array. Each type takes about 60 bytes, I think, and then the code has to check for these.

The point of the multiple types is to encode what slots are in the proxy array. I thought of some other ways to do this that might take less code space. These ways do not require creating new property types.Unfortunately there are no extra fields or bytes in the various structs where we can encode this information.  Instead the existence of proxy slots is encoded by the proxy values themselves.

1. For properties with a getter but no setter or deleter, we could use a different value than `MP_ROM_NONE` to encode the missing setter. We could use `MP_ROM_FALSE` or NULL or similar. If the setter is this special value, then it is equivalent to `MP_ROM_NONE`, _and_ it means there is no deleter slot. This only saves one slot instead of two.

2. Another alternative is trickier: The only things that can be in proxy slots are NONE or object pointers. So an int value would be invalid. In obj repr C, which we use, the low bit is 1 if the value is a 31-bit immediate int.  So we could use this bit to indicate that there are no following slots. That is, if the getter looks like an int, it is in fact not an int, but an object pointer, _and_ the bit means that there is no setter or deleter. If it looks like an int, this fact is noted, and then the bit is cleared when it needs to be considered to be an object pointer. This is maybe too tricky, and I'm not sure it would work for all obj reprs, so there would need to be conditional compilation. This can only work for the property tables in ROM, not the ones in the heap, since it would cause havoc with gc.

Maybe there are other ways to do this kind of like 2. I could not find spare bits anywhere else. I though perhaps it could also be indicated by something in the getter object itself, but I don't see a place to do that.

jepler, 2022-05-03:
I've tried again in #6337 and did get some savings. The approach it takes is to organize "get-only" properties into one region of flash and "get-set" properties into another; in each flash region, they're assumed to have just 1 or 2 proxy slots rather than the traditional 3. It avoids adding additional Python types or performing weird "low bit tricks", and saves 8 bytes for get-only and 4 bytes for get-set (each). However, ~100 bytes of code end up added. There is still a net savings on Trinket M0 (96 bytes); a bigger port like PyGamer saves 940 bytes.

PR #6337 (2022-05-06): Optimize flash storage of properties
In-flash properties with only a getter or only get+set are organized into two specific memory regions, so that it's possible to tell by their address whether they have 1, 2, or 3 proxy slots available. When accessing a property slot, code is added to check if the desired slot is available -- this affects setter & deleter paths, adding some additional conditional branching, so it probably has a minor negative performance effect as well (which I didn't attempt to measure)

A property with only a getter shrinks from 16 byte (type pointer + 3 proxy slots) to 8  [... cut]
closed 2022-05-06 by dhalbert