adafruit/circuitpython #4905: BLE HID -> NotImplementedError: Only one report id per Application collection supported

ThomasAtBBTF, 2021-06-21:
**Firmware**

```python
Adafruit CircuitPython 7.0.0-alpha.3-57-g683304fd3-dirty on 2021-06-19; Adafruit ItsyBitsy nRF52840 Express with nRF52840
```

**Code/REPL**

From the ble_keyboard example, I copied this code to my tester and created BRAILLE_HID_DESCRIPTOR based on code that works for OSX over USB.

```python

hid = HIDService(hid_descriptor=BRAILLE_HID_DESCRIPTOR)

```

**Behavior**

```python
Traceback (most recent call last):
  File "code.py", line 46, in <module>
  File "/lib/adafruit_ble/services/standard/hid.py", line 328, in __init__
  File "/lib/adafruit_ble/services/standard/hid.py", line 419, in _init_devices
NotImplementedError: Only one report id per Application collection supported

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

```

If I replace in the code line the descriptor to 

``` hid_descriptor=DEFAULT_HID_DESCRIPTOR ```

the example works as normal.

**Description**

I am not sure if I should call it a bug or feature request.
The descriptor which is working over USB is not accepted here.

**Additional Info**

Maybe I can split the descriptor into multiple smaller descriptors and create multiple devices, but I am really not sure how to and if VoiceOver IOS will then accept the braille display over BLE

Here the file with the descriptor:
[BRAILLE_HID.zip](https://github.com/adafruit/circuitpython/files/6688163/BRAILLE_HID.zip)

Also for reference: 
[hut1_22.pdf](https://github.com/adafruit/circuitpython/files/6688234/hut1_22.pdf)
where on page 281 the "Braille Display Page (0x41)" is documented.

dhalbert, 2021-06-21:
There is already an issue for this: #4868. Would it make sense to move your requirements to there?

ThomasAtBBTF, 2021-06-21:
Yes!

closed 2021-06-21 by ThomasAtBBTF