adafruit/circuitpython #4908: PulseOut.send sometimes leaving the line high afterwards

dmcomm, 2021-06-23:
**Firmware**

```python
Adafruit CircuitPython 7.0.0-alpha.3-97-g47a6b13d6 on 2021-06-22; Raspberry Pi Pico with rp2040
```

**Code/REPL**

```python
import array
import board
import time
import pulseio
import pwmio

pwm = pwmio.PWMOut(board.GP0, frequency=38000, duty_cycle=2**15)
pulseOut = pulseio.PulseOut(pwm)

pulses = array.array('H', [1000,1000,1000,1000,1000])

while(True):
    print(".")
    pulseOut.send(pulses)
    time.sleep(5)
```

**Behavior**

Occasionally it leaves the line high afterwards. The docs say "This method waits until the whole array of pulses has been sent and ensures the signal is off afterwards" which suggests it should be off whether I sent an odd or even length.

[image]

dmcomm, 2021-06-25:
Seems to be fixed for my example, thanks!

However, I noticed another case, after first creating the objects. I'm thinking it's the same issue, as creating the PWMOut object starts the PWM, then creating the PulseOut object stops it.

```python
Adafruit CircuitPython 7.0.0-alpha.3-418-g769805c9b on 2021-06-25; Raspberry Pi Pico with rp2040
```

```python

import board
import pulseio
import pwmio

pwm = pwmio.PWMOut(board.GP0, frequency=38000, duty_cycle=2**15)
pulseOut = pulseio.PulseOut(pwm)

while(True):
    pass
```

dhalbert, 2021-07-15:
Fixed by #4911.

closed 2021-07-15 by dhalbert