adafruit/circuitpython #4916: pulseio pulsein freezing

rsolow, 2021-06-24:
**Firmware**

Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Metro M4 Express with samd51j19

**Code/REPL**

import board
import pwmio
import pulseio
import time
import countio

def send_signal(freq,dc):
    pwm = pwmio.PWMOut(board.D5, frequency=freq)
    pwm.duty_cycle = int(65535*dc)

    print('Sending Signal')

def capture_cycles(p, num_cycles, timeout):
    p.clear()
    p.resume()
    start = time.time()
    while len(p) < num_cycles:
        if time.time() - start < timeout:
            pass
        else:
            print('TIME OUT')
            break

    p.pause()
    pulses = [p[i] for i in range(len(p))]
    return pulses

def measure_cycles(p, num_cycles, timeout):
    pulses = capture_cycles(p, num_cycles, timeout)
    total_time = sum(pulses)
    total_low = sum(pulses[::2])
    duty_cycle = total_low/total_time*100
    return duty_cycle, total_time, len(pulses)

def recieve_signal(num,timeout):
    p = pulseio.PulseIn(board.TX, maxlen=num)

    print('\nMeasuring PWM input (%s pulses):'%num)
    duty_cycle, total_time, total_pulses = measure_cycles(p, num, timeout)
    print('Duty cycle (percentage): %.2f' % duty_cycle)
    print('Total Run Time (s): %.2f' % (total_time/1000000))
    print('Total Pulses Captured: ', total_pulses)

#metro m4
f = 300
send_signal(f,.7)
recieve_signal(5000,20)

**Behavior**

Output data from pulsein test, shows that regardless of maxlen pulsein stops appending after 1s of data collected

**Description**

I send a pwmout signal from one pin connected to another pin that is reading pulsein. I can change the frequency of the pwmout signal and the number of elements pulsein will freeze at changes, but consistently freezes after one seconds worth of total data is collected by pulsein.

**Additional Info**

I'm using pulseio pulsein to read pulses coming off a tachometer on a motor to measure rpm. I read the pulses fine for one second of data collection, but after one second pulsein stops appending new pulses. I'll set maxlen to some large number, but after one seconds worth of data is collected no new pulses are captured and I never reach maxlen. Pulsein is not freezing at a consistent length, it's not that pulsein reaches x number of elements and then stops appending, but after one second of total data is collected. I've recreated the issue on an Metro M4 and that example code is what is included in this issue report.

DavePutz, 2021-06-24:
I believe the issue is because pulsein has no pre-defined protocol, so an arbitrary limit of 1 second was chosen to detect a runaway peripheral. If you chop up your capture cycles length into smaller chunks so that the 1 second limit isn't reached (a value of 500 worked for me) then you should be OK. I know this complicates your code, but there has to be some limit where pulsein can decide that "too much" data has come in.

rsolow, 2021-06-25:
To work around this 1s limit now I've been doing as you suggested where I capture a smaller set of pulses then append that to a separate list, clear pulsein, resume pulsein, and continue that process until I capture all the data for the time period I want. Would it be possible to remove the 1s limit and just use the maxlen as the limiter to prevent peripheral runaway in future updates of pulseio? Would it also be possible to update the current documentation of pulseio to include the caveat that there is an arbitrary 1s limit on pulsein that will take precedent over the maxlen limit?

DavePutz, 2021-06-25:
The problem with only using maxlen is if a peripheral just holds the line
high (i.e. a very long single pulse). In that
case the maxlen would never be hit, so we need some sort of timeout. The
current pulsein API doesn't include a
timeout value; although that might be a good idea to look at in the future.
I agree that a mention of the limit in the
documentation would be a good idea.

dmcomm, 2021-06-27:
Does that mean that if the total of the durations stored exceeds 1 second, it goes into the paused state, and then you have to remove some entries and call resume()?

rsolow, 2021-06-28:
Yes. After 1s pulsein automatically goes in to the paused state. I call clear() and then resume() to restart capturing pulses.