adafruit/circuitpython #4923: Allow sending less than the whole array in PulseOut.send

dmcomm, 2021-06-25:
Requiring the array.array to be exactly the right length means that if the messages have different numbers of pulses, there's going to be a lot of memory reallocation going on, potentially on quite large objects. An optional count argument would allow reuse of the same memory allocation for each message.

deshipu, 2021-06-27:
I wonder, can you use `memoryview` and slices to work around this?

dmcomm, 2021-06-28:
> I wonder, can you use `memoryview` and slices to work around this?

Can you create two Array with the same backing buffer? Though if you can, that would risk making really hard-to-trace bugs if you accidentally trigger a resizing operation.

Passing something that's not an Array, or is an Array with the wrong typecode, throws `TypeError: Array must contain halfwords (type 'H')`. I could investigate what it's checking exactly and whether there's a reasonable way to fool it.