adafruit/circuitpython #4926: Rename _pixelbuf and adafruit_pypixelbuf to match

kattni, 2021-06-25:
We included `colorwheel()` in both `_pixelbuf` and `adafruit_pypixelbuf`, but to use it across all CircuitPython-compatible, it requires a `try`/`except` import structure. This was something we did not consider from the beginning.

We should rename `_pypixelbuf` and `adafruit_pypixelbuf` to match so Pypixelbuf is a drop-in replacement for `_pixelbuf`. 

@dhalbert has suggested another option involving making the latter an alias for the former, so a single import works for both, but we're not entirely sure how priority of native modules over libraries works, so are therefore unsure how aliasing would work.

My suggestion is to rename `adafruit_pypixelbuf` to `adafruit_pixelbuf` and rename `_pixelbuf` to match. This is to maintain the fact that the native module is not Python, and therefore should not be named `py` anything, but I'm willing to concede to it not really mattering in the end, and that it may be easier to only rename the native module.

There are very few Learn guides that reference `adafruit_pypixelbuf`, and those that do, do so only in the required library lists, so there is very little that would need to be updated to match. The only direct references to either one in code appear to be when I have written code to call `color_wheel()`, which I don't believe has happened that often yet, so again, very little to be updated.

As we are breaking other things as well, I'd like to see this happen in 7.0.0.

kattni, 2021-06-28:
We discussed this in the CircuitPython Weekly and decided that rather than rename anything, we can make a `colorwheel()` native CircuitPython module with a matching Python library for instances where the module is not enabled for a particular board (in the event that this is an issue).

We also need to ensure that the native module and the Python version work the same (there was concern about what is returned, a tuple or an integer - it appears the current Python version is different from the current C version).

@dhalbert wanted to discuss naming the module something other than `colorwheel()`.

dhalbert, 2021-06-28:
i"m thinking:

Fork or otherwise copy the repo for `adafruit_pypixelbuf`, rename it to `adafruit_pixelbuf`, and remove `colorwheel()`. Add the new `adafruit_pixelbuf` to the bundle. Change guide code and neopixel/dostar libraries to use `adafruit_pixelbuf`.
Create a new `adafruit_colorwheel` Python library or whatever we want to call it. Implement `adafruit_colorwheel` natively in 7.0.0 as well. Also in 7.0.0, rename `_pypixelbuf` to `adafruit_pixelbuf` .

@jepler notes we can change `adafruit_pypixelbuf.py` to read `from adafruit_pixelbuf import *`.
MIght be slightly more complicated if wheel is out of `adafruit_pixelbuf`.

dhalbert, 2021-07-09:
@kattni Now that #4976 is merged, do you want to narrow the scope of this?

kattni, 2021-07-12:
@dhalbert I created a new issue (#4992) since this issue became specific to the native `rainbowio` module.

closed 2021-07-12 by tannewt