adafruit/circuitpython #4927: missing blank line in adafruit_bus_device docs

2bndy5, 2021-06-26:
https://github.com/adafruit/circuitpython/blob/5a4a80336fb21823c995fc8430baae69bbae7fe4/shared-bindings/adafruit_bus_device/SPIDevice.c#L47-L48

https://github.com/adafruit/circuitpython/blob/5a4a80336fb21823c995fc8430baae69bbae7fe4/shared-bindings/adafruit_bus_device/I2CDevice.c#L48-L49

The first `:param:` directive needs to be preceded by a blank line. Currently (see above snippets), the parameters are all clumped into 1 paragraph.

I also noticed [the `busio` docs still link to the pure python package's docs](https://github.com/adafruit/circuitpython/blob/5a4a80336fb21823c995fc8430baae69bbae7fe4/shared-bindings/busio/SPI.c#L68). To link to circuitpython's docs, (**I think**) it would be as easy as removing [a intersphinx mapping from the conf.py file](https://github.com/adafruit/circuitpython/blob/5a4a80336fb21823c995fc8430baae69bbae7fe4/conf.py#L417).

ps - If not all boards have `adafruit_bus_device`, it would be nice to see which boards (in the [module support matrix](https://circuitpython.readthedocs.io/en/6.3.x/shared-bindings/support_matrix.html#)) have `adafruit_bus_device` builtin. However, that might be a more complicated solution given that the matrix seems to be [generated with a py script](https://github.com/adafruit/circuitpython/blob/main/docs/shared_bindings_matrix.py) (**I think**). So, that should probably be a separate issue.

dhalbert, 2021-06-26:
> ps - If not all boards have `adafruit_bus_device`, it would be nice to see which boards (in the [module support [...]

The presence or absence of `adafruit_bus_device` is included in the `latest` (7.0.0) docs. There is also a very nice dynamic search capability: https://circuitpython.readthedocs.io/en/latest/shared-bindings/support_matrix.html.

If you would like to submit a simple PR for the doc errors, that would be great. If not, we can do ti.

2bndy5, 2021-06-26:
> The presence or absence of adafruit_bus_device is included in the latest (7.0.0) docs.

Awesome. Thanks.

> If you would like to submit a simple PR for the doc errors, that would be great.

I'd be glad to submit a PR for this.

PR #4928 (2021-06-28): update busio & adafruit_bus_device docs
resolves #4927 
I also found/fixed some typos while making sure that sphinx's interlinking from `busio` docs works properly.
closed 2021-06-28 by tannewt