adafruit/circuitpython #4929: ESP32S2 build does no complete

jerryneedell, 2021-06-26:
I just clones a fresh copy from the repository  after the update of the esp-idf.
I did the usual 
```
./install.sh
. ./esport.sh
```
then tried to build the merto_esp32s2 and the build stopped without completing. [ It stopped with ````
658/658] Generating ld/esp32s2.project.ld
```
Has something changed in the process or did I miss a step.

Here is the log
 ```
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/esp32s2$ make BOARD=adafruit_metro_esp32s2 clean
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
rm -rf build-adafruit_metro_esp32s2 
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/esp32s2$ make BOARD=adafruit_metro_esp32s2 -j8
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
IDF_PATH=/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf cmake -S . -B build-adafruit_metro_esp32s2/esp-idf -DSDKCONFIG=build-adafruit_metro_esp32s2/esp-idf/sdkconfig -DSDKCONFIG_DEFAULTS="esp-idf-config/sdkconfig.defaults;esp-idf-config/sdkconfig-opt.defaults;esp-idf-config/sdkconfig-4MB.defaults;boards/adafruit_metro_esp32s2/sdkconfig" -DCMAKE_TOOLCHAIN_FILE=/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/tools/cmake/toolchain-esp32s2.cmake -DIDF_TARGET=esp32s2 -GNinja
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/jerryneedell/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /home/jerryneedell/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /home/jerryneedell/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/jerryneedell/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++
-- Check for working CXX compiler: /home/jerryneedell/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s2
-- Checking Python dependencies...
Python requirements from /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/requirements.txt are satisfied.
-- Project sdkconfig file /home/jerryneedell/projects/circuitpython/ports/esp32s2/build-adafruit_metro_esp32s2/esp-idf/sdkconfig
Loading defaults file /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig.defaults...
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig.defaults:258 CONFIG_ESP32S2_ALLOW_RTC_FAST_MEM_AS_HEAP was replaced with CONFIG_ESP_SYSTEM_ALLOW_RTC_FAST_MEM_AS_HEAP
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig.defaults:375 CONFIG_ESP32_ENABLE_COREDUMP_TO_NONE was replaced with CONFIG_ESP_COREDUMP_ENABLE_TO_NONE
Loading defaults file /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig-opt.defaults...
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig-opt.defaults:5 CONFIG_ESP_CONSOLE_UART_NONE was replaced with CONFIG_ESP_CONSOLE_NONE
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig-opt.defaults:8 CONFIG_CONSOLE_UART_NUM was replaced with CONFIG_ESP_CONSOLE_UART_NUM
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig-opt.defaults:12 CONFIG_ESP32S2_PANIC_SILENT_REBOOT was replaced with CONFIG_ESP_SYSTEM_PANIC_SILENT_REBOOT
Loading defaults file /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig-4MB.defaults...
Loading defaults file /home/jerryneedell/projects/circuitpython/ports/esp32s2/boards/adafruit_metro_esp32s2/sdkconfig...
-- Found PythonInterp: /home/jerryneedell/.espressif/python_env/idf4.3_py3.8_env/bin/python (found version "2.7.18") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- App "circuitpython" version: 7.0.0-alpha.3-433-g5a4a80336
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/build-adafruit_metro_esp32s2/esp-idf/esp-idf/esp32s2/esp32s2_out.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp32s2/ld/esp32s2.project.ld.in
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp32s2/ld/esp32s2.peripherals.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.libgcc.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-data.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.spiflash.ld
-- Components: app_trace app_update bootloader bootloader_support cxx driver efuse esp-tls esp32s2 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hw_support esp_ipc esp_netif esp_pm esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py freertos hal heap log lwip mbedtls newlib nghttp nvs_flash partition_table pthread soc spi_flash tcpip_adapter vfs wpa_supplicant xtensa
-- Component paths: /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/app_trace /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/app_update /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/bootloader /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/bootloader_support /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/cxx /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/driver /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/efuse /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp-tls /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp32s2 /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_adc_cal /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_common /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_eth /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_event /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_gdbstub /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_hw_support /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_ipc /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_netif /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_pm /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_ringbuf /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_system /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_timer /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_wifi /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/espcoredump /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esptool_py /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/freertos /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf [... cut ...]
Generated /home/jerryneedell/projects/circuitpython/ports/esp32s2/build-adafruit_metro_esp32s2/esp-idf/bootloader/bootloader.bin
[647/658] Generating x509_crt_bundle
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py:36: CryptographyDeprecationWarning: Python 2 is no longer supported by the Python core team. Support for it is now deprecated in cryptography, and will be removed in a future release.
  from cryptography import x509

jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/esp32s2$ 

```

jerryneedell, 2021-06-26:
Updated to add:
I repeated this on my Mac and the build completed normally
the failure was on Ubuntu Linux (20.04)

Additional update: It also works normally on my Raspberry Pi4...
Can someone else try this on an Ubuntu Linux system and see if this is reproducible?

microDev1, 2021-06-26:
@jerryneedell It builds fine for me on `ubuntu-20.04` and the `CI` also uses that build environment. 
Perhaps there is something wrong with the state of the repo.

dhalbert, 2021-06-26:
I tried this as well. I first did a `git pull` on my clone of `adafruit/circuitpython`. I got some errors at the end of that:
```
Fetching submodule ports/esp32s2/esp-idf/components/mqtt/esp-mqtt
From https://github.com/espressif/esp-mqtt
   2b20b51..f10321a  master     -> origin/master
Fetching submodule ports/esp32s2/esp-idf/components/tinyusb/tinyusb
From https://github.com/espressif/tinyusb
 * [new branch]        esp-based_on_334e95f   -> origin/esp-based_on_334e95f
 * [new branch]        esp-based_on_43cb5fab  -> origin/esp-based_on_43cb5fab
 * [new branch]        esp-based_on_c5f440a6a -> origin/esp-based_on_c5f440a6a
 + edd91d28...075334af master                 -> origin/master  (forced update)
fatal: remote error: upload-pack: not our ref ca8f5c602ec2948a31146b5f5e429dea369060a4
Errors during submodule fetch:
	components/bt/controller/lib
Errors during submodule fetch:
	ports/esp32s2/esp-idf
```
This was easy to miss: I had to scroll back to see it.

After this, the commit of `esp-idf` was wrong, and did not match upstream, probably because the `git pull` did not really complete.
I then did `rm -rf esp-idf` and repeated the `git pull`, which completed successfully. Then I did my usual:
`git submodule sync --quiet --recursive && git submodule update --init`
Then
```
esp-idf/install.sh    # This installed a bunch of stuff
esp-idf/export.sh
```
and then did a build, which worked.

jerryneedell, 2021-06-26:
Hmm -- I did a complete reinstall and am getting the same failure.
If I run it without the -j8 I get an error message
```
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/esp32s2$ make BOARD=adafruit_metro_esp32s2 clean
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
rm -rf build-adafruit_metro_esp32s2 
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/esp32s2$ make BOARD=adafruit_metro_esp32s2 
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
IDF_PATH=/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf cmake -S . -B build-adafruit_metro_esp32s2/esp-idf -DSDKCONFIG=build-adafruit_metro_esp32s2/esp-idf/sdkconfig -DSDKCONFIG_DEFAULTS="esp-idf-config/sdkconfig.defaults;esp-idf-config/sdkconfig-opt.defaults;esp-idf-config/sdkconfig-4MB.defaults;boards/adafruit_metro_esp32s2/sdkconfig" -DCMAKE_TOOLCHAIN_FILE=/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/tools/cmake/toolchain-esp32s2.cmake -DIDF_TARGET=esp32s2 -GNinja
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/jerryneedell/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /home/jerryneedell/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /home/jerryneedell/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/jerryneedell/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++
-- Check for working CXX compiler: /home/jerryneedell/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s2
-- Checking Python dependencies...
Python requirements from /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/requirements.txt are satisfied.
-- Project sdkconfig file /home/jerryneedell/projects/circuitpython/ports/esp32s2/build-adafruit_metro_esp32s2/esp-idf/sdkconfig
Loading defaults file /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig.defaults...
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig.defaults:258 CONFIG_ESP32S2_ALLOW_RTC_FAST_MEM_AS_HEAP was replaced with CONFIG_ESP_SYSTEM_ALLOW_RTC_FAST_MEM_AS_HEAP
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig.defaults:375 CONFIG_ESP32_ENABLE_COREDUMP_TO_NONE was replaced with CONFIG_ESP_COREDUMP_ENABLE_TO_NONE
Loading defaults file /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig-opt.defaults...
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig-opt.defaults:5 CONFIG_ESP_CONSOLE_UART_NONE was replaced with CONFIG_ESP_CONSOLE_NONE
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig-opt.defaults:8 CONFIG_CONSOLE_UART_NUM was replaced with CONFIG_ESP_CONSOLE_UART_NUM
/home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig-opt.defaults:12 CONFIG_ESP32S2_PANIC_SILENT_REBOOT was replaced with CONFIG_ESP_SYSTEM_PANIC_SILENT_REBOOT
Loading defaults file /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf-config/sdkconfig-4MB.defaults...
Loading defaults file /home/jerryneedell/projects/circuitpython/ports/esp32s2/boards/adafruit_metro_esp32s2/sdkconfig...
-- Found PythonInterp: /home/jerryneedell/.espressif/python_env/idf4.3_py3.8_env/bin/python (found version "2.7.18") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- App "circuitpython" version: 7.0.0-alpha.3-433-g5a4a80336
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/build-adafruit_metro_esp32s2/esp-idf/esp-idf/esp32s2/esp32s2_out.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp32s2/ld/esp32s2.project.ld.in
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp32s2/ld/esp32s2.peripherals.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.libgcc.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld
-- Adding linker script /home/jerryneedell/projects/circuitpython/ports/esp32s2/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-data.ld
[... cut ...] s2/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jerryneedell/projects/circuitpython/ports/esp32s2/build-adafruit_metro_esp32s2/esp-idf
  File "../../py/genlast.py", line 50
    print(e, file=sys.stderr)
                 ^
SyntaxError: invalid syntax
make: *** [../../py/mkrules.mk:100: build-adafruit_metro_esp32s2/genhdr/qstr.splitjerrjerryneedell@jjerrynjerryjerrjjjerryneejerjerrynejerryjejjerryneedell@jerryneedell-ubuntu-macmini
```

jerryneedell, 2021-06-26:
Ah -- the error is there with -j8 -- just scrolls by too fast.

dhalbert, 2021-06-26:
```
-- Found PythonInterp: /home/jerryneedell/.espressif/python_env/idf4.3_py3.8_env/bin/python (found version "2.7.18") 
```
That explains the Python error. If you do this after the `source export.sh`, what do you see?
```
halbert@tuna:~/repos/adafruit/circuitpython/ports/esp32s2$ which python
/home/halbert/.espressif/python_env/idf4.3_py3.8_env/bin/python
halbert@tuna:~/repos/adafruit/circuitpython/ports/esp32s2$ python --version
Python 3.8.5
```

jerryneedell, 2021-06-26:
ahh 
```
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/esp32s2$ which python
/home/jerryneedell/.espressif/python_env/idf4.3_py3.8_env/bin/python
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/esp32s2$ python --version
Python 2.7.18
``` something is using python -- not python3

jerryneedell, 2021-06-26:
My Mac and Rasberry Pi only have Python 3 installed

dhalbert, 2021-06-26:
On my system:
```
halbert@tuna:~/.espressif/python_env/idf4.3_py3.8_env/bin$ ls -l python
lrwxrwxrwx 1 halbert halbert 15 Jun 26 10:02 python -> /usr/bin/python
```
So `/usr/bin/python` on your system is pointing to python2. I would suggest doing `sudo apt install python-is-python3`.

Normally there is no `python` on ubuntu, to prevent this kind of version problem. Unfortunately the esp-idf expects there to be a plain `python`.

Somehow yours got pointed to python 2.7

jerryneedell, 2021-06-26:
That fixed it -- something changed since it worked yesterday....  oh well, closing

closed 2021-06-26 by jerryneedell