adafruit/circuitpython #4944: Pin Stays On After Changing Code

LedbetterAlex, 2021-06-29:
**Firmware**

```python
Adafruit CircuitPython 6.3.0 on 2021-06-01; Teensy 4.0 with IMXRT1062DVJ6A
```

**Code/REPL**

```python
import board, digitalio
led = digitalio.DigitalInOut(board.D13)
led.direction = digitalio.Direction.OUTPUT
led.value = True
```

**Behavior**

```python
N/A
```

**Description**

Obviously the code runs fine, and turns on the on-board LED. However, I also have an LED wired up to pin D11, and when I switch D13 out for D11 in the code and hit save, the on-board LED stays on (i.e. the pin voltage stays high when it should reset to zero after the new code no longer uses the pin). If I then unplug and plug in my board again, the on-board LED is now off like it should be. The same happens in reverse.

Seems to me like the pin voltages should reset to 0 V after soft reset (reload).

A member of the Adafruit forums suggests it may be specific to iMX RT.

**Additional Info**

Just thanks!!! Really cool software.

-Alex

KurtE, 2022-04-13:
@dhalbert and all:

I don't think this is at all specific to the Teensy mimxrt... 

I put several debug outputs into the port code and there does not appear to be anywhere that the pin is logically released.

I then tried same stuff on RP2040:

```
Adafruit CircuitPython 7.3.0-beta.0-2-g87e59a444 on 2022-04-04; Adafruit Feather RP2040 with rp2040
>>> import board, busio
>>> from digitalio import DigitalInOut, Direction
>>> led = DigitalInOut(board.LED)
>>> led.direction = Direction.OUTPUT
>>> led.value = True
>>> led.value = False
>>> led.value = True
>>> led = DigitalInOut(board.D5)

```

And the LED is still on...

ladyada, 2022-04-13:
correct, if you want to reset the pin, you would use led.deinit() just like if you open a file on an SD drive, it must be 'close()'d manually - reassigning the file variable does not automatically close the file.

ladyada, 2022-04-13:
ill leave open as i know @tannewt was working on tweaking deinited/default pin settings for power management

KurtE, 2022-04-13:
> ill leave open as i know @tannewt was working on tweaking deinited/default pin settings for power management

Sounds good, you might either want to remove the mimxrt... label and/or add others as it looks like not specific to T4.x...

dhalbert, 2022-04-13:
Nothing will happen to the original `DigitalInOut(board.LED)` object until garbage collection, and even then, it may not reset the pin, if there is not deleter cleanup function on pins. And one cannot depend on deleter for cleanup anyway. Basically the object is "lost" but not forgotten, so the pin is not going to get changed. This is more about Python semantics than pin handling.

tannewt, 2022-05-09:
The pin should be reset between VM runs by this code: https://github.com/adafruit/circuitpython/blob/main/ports/mimxrt10xx/common-hal/microcontroller/Pin.c#L38=

It looks like I changed that code in #6125 and hopefully fixed @LedbetterAlex's original issue.

dhalbert, 2022-10-23:
Closing because this is probably fixed.

closed 2022-10-23 by dhalbert