adafruit/circuitpython #4945: PulseIn crashes if there are two of them

dmcomm, 2021-06-29:
**Firmware**

```python
Adafruit CircuitPython 7.0.0-alpha.3-452-g8a5a7457a on 2021-06-29; Raspberry Pi Pico with rp2040
```

**Code/REPL**

```python
import board
import digitalio
import pulseio
import time

#GP0 to GP1; GP2 grounded

print("start")
digitalOut = digitalio.DigitalInOut(board.GP0)
digitalOut.direction = digitalio.Direction.OUTPUT
time.sleep(5)

print("creating PulseIns")
pulsesIn1 = pulseio.PulseIn(board.GP1, maxlen=20)
pulsesIn2 = pulseio.PulseIn(board.GP2, maxlen=20)
time.sleep(3)

print("high")
digitalOut.value = True

while True:
    print(".", end="")
    time.sleep(1)
```

**Behavior**

After "high", the USB connection is dropped. If I comment out the `pulsesIn2` line, it prints the dots as expected.

tannewt, 2021-06-29:
@dmcomm Would you like to help fixes these pulseio issues you are finding? We'd be happy to help get you modifying and debugging circuitpython internals. I'd suggest joining the discord and saying hi in #circuitpython-dev to get started. You can join via https://adafru.it/discord

dmcomm, 2021-06-29:
Sorry, that's more than I want to take on at the moment. If I'd known how this was going to go, I probably wouldn't have started, but I hope it was useful ;)

DavePutz, 2021-07-04:
It looks like there is some conflict between the digitalio on GP0 and
the pulseins on GP1 and GP2. If the pulsein pins are changed to GP5
and GP6 then things run as expected. I will continue to investigate.

dmcomm, 2021-07-12:
Trying to break it now, 0c6dc6f crashes at 5. I guess it should throw an exception at the actual limit.

```python
import board
import digitalio
import pulseio
import time

#GP0..N chained together
numPulseIns = 5

print("start")
digitalOut = digitalio.DigitalInOut(board.GP0)
digitalOut.direction = digitalio.Direction.OUTPUT
time.sleep(7)

print("creating PulseIns")
pulseIns = []

for i in range(numPulseIns):
	pin = getattr(board, "GP%d" % (i+1))
	print(pin)
	pulseIn = pulseio.PulseIn(pin, maxlen=20)
	pulseIns.append(pulseIn)
time.sleep(3)

print("high")
digitalOut.value = True

while True:
	print(".", end="")
	time.sleep(1)
```

DavePutz, 2021-07-13:
The limit should be 8 (2 PIOs with 4 state machines each) assuming no other state machines are in use. . I'll look at why it hangs on 5, and make sure an error is raised when the limit is exceeded.

dmcomm, 2021-07-19:
Fixed!

closed 2021-07-19 by dmcomm