adafruit/circuitpython #4946: PulseIn drops very short pulses

dmcomm, 2021-06-29:
**Firmware**

```python
Adafruit CircuitPython 7.0.0-alpha.3-452-g8a5a7457a on 2021-06-29; Raspberry Pi Pico with rp2040
```

**Code/REPL**

```python
import board
import pwmio
import pulseio
import time

pwmOut = pwmio.PWMOut(board.GP0, frequency=10000, duty_cycle=6554)
pulsesIn = pulseio.PulseIn(board.GP1, maxlen=100)
#GP0 and GP1 are connected by a resistor

while True:
    pulsesIn.clear()
    while len(pulsesIn) < 20:
        pass
    for i in range(20):
        print(pulsesIn.popleft(), end=",")
    print()
    time.sleep(5)
```

**Behavior**

```python
89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,
89,89,88,88,88,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,
89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,89,
```

The biggest problem here is that it's broken the contract that we can tell whether a pulse duration is high or low by position in the list. If a pulse is really too short to measure, I would expect it to insert a 0 to fulfill that. It clearly knows that there was a pulse at those positions.

Noting that it seems to be measuring the gap, not pulse+gap.

Might be related to #4932.

GP1 scope trace:

[image]

tannewt, 2021-06-29:
I think we're sampling at 1mhz to get the 1us resolution. Any pulse shorter than that would break the contract for what we can measure and all bets are off.

dmcomm, 2021-06-29:
These are 10us pulses in the example, which should be within the range of what it can measure.

Supposing there was a <1us pulse, it would be fair to ignore it completely, but I'd say switching the polarity of the output is still a bug. Either we detected it or we didn't.

If the pulses and gaps are both too short, then indeed all bets are off.

DavePutz, 2021-07-02:
I put a somewhat arbitrary 10us minimum pulse check into the RP2040 pulsein
code.
This was necessary because while testing the irremote device (which uses
pulsein) I
was consistently seeing stray pulses from IR being picked up from the
surrounding environment.
10 us seemed to give me the most consistent readings for the irremote
device. If you
really need to make a  10us measurement I could drop that down to 7 or 8
us, which would have a
marginal impact on the infrared readings.

dmcomm, 2021-07-02:
@DavePutz thanks for your reply. Looks like the real signal which raised this issue is even a bit less than 10us. I understand if that's asking too much of such a system which needs to be optimized for more common use cases.

However, there's still a bug. Suppose the real incoming signal is 100,95,5,100,100. If you want to suppress the "5" as noise, the correct output is 100,200,100. The bugged output is 100,95,100,100 which shortens the total time and flips the polarity.

Picture of real use which is asking too much:

[image]

dmcomm, 2021-07-06:
Accepting that the intention is to ignore stray pulses <10us, here is an illustration of the bug:

[image]

(For my own problem, the protocol above has pulses of ~8us, gaps of >90us, and I only care about the time between the start of one pulse and the next. So I solved that by changing my circuit to stretch the pulses out a bit.)

dmcomm, 2021-07-10:
@DavePutz those "stray pulses from IR" could be #4983 where I'm getting 4-7us lows while the pin is definitely high. (I just checked that PulseIn uses the restart function.) If that's the case, you might be able to remove the MIN_PULSE filter.

DaftVagabond, 2021-11-03:
I've ben having a time with a similar issues. mikeysklar over on the Adafruit forums suggested I post this suggestion here.
(Here's the thread:  https://forums.adafruit.com/viewtopic.php?f=62&t=184158&start=15)

Would it be possible to add a parameter that can control the minimum lengths or even turn them off? This would allow people who aren't doing IR-related projects to get more functionality out of the PulseIn feature.

tannewt, 2021-11-03:
Could you explain further how you are using PulseIn? I'm curious what you are using it for.

DaftVagabond, 2021-11-03:
I have a small board that's designed for props, it controls lighting effects, sounds, leds, stuff like that. It has a feature where it can output a PWM base on the intensity of the sound it's currently producing, the idea being that you can make lights or other devices react to the sound.

I've been toying around with reading the pulses on Arduino using PulseIn, I even made a few small test projects. When I went to port the code to Circuitpython, I found that it's PulseIn returned much deferent values, and they were much less consistent. I thought it'd be simple enough to just adjust the values in my code to accommodate the new readings, but everything was behaving so unpredictably.

So my specific circumstance is that I'd like to be able to read PWM signals coming from other devices more reliably at shorter lengths. That's why I suggested a parameter that allowed us to adjust the 10us minimum pulse check that DavePutz mentioned. Again, I feel that would open up a range of functionality. Anyone using it for IR-related projects would just need to be mindful if the implications of adjusting the parameter.

tannewt, 2021-11-04:
@DaftVagabond Thanks for the info. What is the base frequency of the PWM? Do you need to measure individual pulses or could you use a different technique to measure the average? You could filter the signal and read it as an analog value. If the pulse frequency changes you could use countio to measure how many pulses happened.

DaftVagabond, 2021-11-09:
I don't feel like this will be the most insightful answer to your question. I wrote my original code in the Arduino IDE using PusleIn, so I used the results that this fed out. So, however Arduino's PulseIn measures things, that's what I'd prefer.

I think the PWM is around 30Hz.

tannewt, 2021-11-10:
That is helpful. It looks like it does measure a single pulse. https://www.arduino.cc/reference/en/language/functions/advanced-io/pulsein/

DaftVagabond, 2021-11-15:
Does that mean anything for the issue I'm experiencing?

DaftVagabond, 2021-12-06:
> @DaftVagabond Thanks for the info. What is the base frequency of the PWM? Do you need to measure individual pulses or could you use a different technique to measure the average? You could filter the [...]

I think filtering and reading as an analog value would be a good option. The list never gets any longer than index zero with the default code for reading pulses.

`while True:

    while len(pulse) == 0:
        pass

    pulse.pause()
    print(pulse[0])
    pulse.clear()
    pulse.resume(80)`

So I'm not entirely sure how to do that. Also, I do know that with that code if I were to read the pulses, they'd be between 31 and 1, but if I remove the pause() and resume() it reads the values as a 16-bit number, in the thousands.

I also tried to use a slice function but it told me that slices weren't supported.

dmcomm, 2022-07-11:
As of the excellent pull request above, RP2040 PulseIn has perfect 1us accuracy. Can my team safely rely on short pulses being detected in the future? Hoping it's not a matter of "the short pulse filter is coming back (non-optionally) when we figure out how to do it correctly". Noting that such a filter doesn't need to be in the PulseIn core and could be done in Python.

closed 2022-07-11 by tannewt