adafruit/circuitpython #4951: Kaluga not connecting via USB (since esp-idf update?)

jepler, 2021-06-30:
I am using the uf2 bootloader (0.4.0) installed with Adafruit ESPTool.

Firmware downloaded from s3 and installed via uf2/KALUGA1BOOT:
adafruit-circuitpython-espressif_kaluga_1.3-en_US-20210625-5a4a803.uf2

**Behavior**

With the above named firmware version, the CIRCUITPY drive does not appear.  The LEDs do blink, so CircuitPython appears to start.

Linux `dmesg` shows errors like `device descriptor read/64, error -110`

Reverting to adafruit-circuitpython-espressif_kaluga_1.3-en_US-20210625-d67fb85.uf2 from s3 works

I think this indicates the problem was introduced by #4195

I did not test any other boards or installing via .bin.

@askpatrickw you may be interested in taking a look.

askpatrickw, 2021-06-30:
@jepler i'm away from my desk for a few days. But I do have a Kaluga and can try to repo next week.

dhalbert, 2021-07-01:
Does the .bin work?

jerryneedell, 2021-07-01:
Using the .bin , mine boots but goes in to safe mode
```
uto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.3-474-g37c52110c on 2021-07-01; Kaluga 1 with ESP32S2
>>> 
```

jerryneedell, 2021-07-01:
ah -- but it is OK if I remove my old code.py (using the OV7670 demo code from a few weeks ago
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.3-474-g37c52110c on 2021-07-01; Kaluga 1 with ESP32S2
>>> 
>>> 

```

jepler, 2021-07-01:
My device (probably) had an OV____ camera demo on CIRCUITPY too, fwiw.

rsbohn, 2021-07-02:
When I copy the .UF2 `adafruit-circuitpython-adafruit_funhouse-en_GB-20210701-37c5211.uf2` to my funhouse the dotstars flash purple. Doesn't happen with the 7.0.0-alpha.3.

rsbohn, 2021-07-02:
I can then go into boot mode and install 7.0.0-alpha.3 (tries to run code.py, fails because I haven't updated the libraries yet).

dhalbert, 2021-07-02:
I tried the latest builds on a Metro ESP32 and a Feather S2. Without a code.py, they come up and I can use the REPL.

I had a random test program on the Feather S2, and found that this simple program causes a crash:
```py
import busio
import board
import displayio
import adafruit_il0373

displayio.release_displays()

spi = busio.SPI(board.SCK, board.MOSI)
epd_cs = board.D9
epd_dc = board.D10

display_bus = displayio.FourWire(spi, command=epd_dc, chip_select=epd_cs, baudrate=1000000)
```
The last line causes a crash. If I remove the `displayio.FourWire()`, there is no crash.

I didn't actually have a display attached, but it doesn't matter: the display is output-only.

jfurcean, 2021-07-02:
I was able to duplicate the crash on the Feather S2 with the latest from s3.

```py
import busio
import board
import displayio
import adafruit_displayio_ssd1306

displayio.release_displays()

i2c = busio.I2C(board.SCL, board.SDA)

display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)
```
The last line is what causes the crash as well.

I confirmed that it doesn't crass with `adafruit-circuitpython-unexpectedmaker_feathers2-en_US-20210625-d67fb85.uf2` from s3.

jerryneedell, 2021-07-02:
See comments on PR #4960 -- it fixes this issue on the Kaluga and on the Funhouse

dhalbert, 2021-07-02:
Confirming that #4960 fixes my crash as well. Will merge!

PR #4960 (2021-07-02): ESP32S2: Fix spi_never_reset()
closes #4951.
closed 2021-07-02 by dhalbert