adafruit/circuitpython #4954: DigitalIO OUTPUT to dinit() to AnalogIO Does Not Work

LedbetterAlex, 2021-07-01:
### CircuitPython version

```python
Adafruit CircuitPython 6.3.0 on 2021-06-01; Teensy 4.0 with IMXRT1062DVJ6A
```

### Code/REPL

```python
import digitalio, analogio

pot = digitalio.DigitalInOut(board.D23)
pot.direction = digitalio.Direction.OUTPUT
pot.deinit()
pot = analogio.AnalogIn(board.A9)

print((pot.value,))
```

### Behavior

If I use a digital / analog combo pin first as a digital output pin, then later use deinit() to release the pin, and then finally try to use the same pin as analog pin, the values read into the pin are all out of whack, technically speaking. 

I don't know exactly what is going on, but it APPEARS like the pin is staying digital when it reads in values later as an analog io pin, i.e. it can only really read high or low values, with small variances at the high and low states, e.g. 0-500 the straight to 63000-64000.

[image]

### Additional information

Now why exactly would someone want to use a pin once for digital, and again for analog? See my other bug report, "Pin Stays On After Changing Code #4944". I have the below work-around code with sets all pins to digital outputs and then sets them false before deinitializing them, and this bug makes that not work for analog pins :(

```python
import board
import digitalio

def boardReleaseAll():

    boardMembers = [a for a in dir(board) if a.startswith('D')]

    for boardPin in boardMembers:
        mPin = digitalio.DigitalInOut(eval("board."+boardPin))
        mPin.direction = digitalio.Direction.OUTPUT
        mPin.value = False
        mPin.deinit()
        print("Released: "+boardPin)
```

Okay, good luck!!

tannewt, 2021-07-01:
I suspect this is the same issue as #4944.

KurtE, 2022-04-13:
Sorry, I thought I would take a look at a couple of the Imxrt marked issues. 

Saw this one.  Took a little time to how to print debug stuff (printf) and then also setup that debug output went to my Teensy Serial4 (LPUART3)...

Test sketch
```
import digitalio, analogio, board

pot = analogio.AnalogIn(board.A9)
for i in range(10):
    print(pot.value)
pot.deinit()

print("try the digital")
pot = digitalio.DigitalInOut(board.D23)
pot.direction = digitalio.Direction.OUTPUT
pot.deinit()
pot = analogio.AnalogIn(board.A9)
for i in range(10):
    print(pot.value)
```

Note I added some prints int the init and deinit functions to see if they were called 

The first run showed differences like mentioned:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
592
704
624
688
640
656
560
544
656
576
try the digital
>>> common_hal_digitalio_digitalinout_construct called: 25
>>> common_hal_digitalio_digitalinout_deinited called: 25
>>> common_hal_digitalio_digitalinout_deinited called: 25
0
0
0
0
0
0
0
0
0
0
```
But I had nothing hooked up to the IO pin... So floating garbage... So I hooked up simple pot with 3.3/GND and up to the IO pin and reran:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
400
272
256
240
272
256
272
272
256
272
try the digital
>>> common_hal_digitalio_digitalinout_construct called: 25
>>> common_hal_digitalio_digitalinout_deinited called: 25
>>> common_hal_digitalio_digitalinout_deinited called: 25
400
272
256
272
272
256
272
272
256
272
```
So looks a lot more consistent.  

Note: I know the deinit is not really doing anything.    And I know from Teensy Forum on T4.x that you get better Analog
results if you do pinMode(23, INPUT_DISABLE)

So question is should that be the state that the pin should be set to when deinit is called?

As for the other issue.  Not sure if same issue or if it is an issue? 
That is if I in REPL say led.value=True

I would hope the LED would stay on after I typed that in... But that is again different issue.

Thoughts?

dhalbert, 2022-04-13:
We often use `mp_printf(&mp_plat_print, "something: %d\n", something);`. That goes to the REPL connection, so you don't have to wire up the serial debug.

As you surmise, it would be good to see if `DigitalInOut.deinit()` does a reset on the pin or not. Check some of the other ports and see if i.MX is doing something similar. That is how I would start on this.

PR #8928 (2024-02-16): Numerous iMX fixes and SDK update
* Fixes #4954. Leaving GPIO in output mode messed with ADC readings.
* Fixes #8018 and fixes #6537. UART rx was more complicated than it needed to be.
* Fixes #5581. uart reset is handled by finalizer. Added i2c_reset() call and reservation check.
closed 2024-02-16 by jepler