adafruit/circuitpython #4955: audiomixer module for ESP32S2

rsbohn, 2021-07-01:
I can play WaveFiles over I2SOut but I want to control the volume and interleave samples. The `audiomixer` module would allow both.

tannewt, 2021-07-01:
I believe this was disabled for ESP32-S2 because the implementation used ARM assembly. We now have a C implementation as well. I'd suggest trying to enable it yourself in a build. It may just work.

rsbohn, 2021-07-03:
I enabled the audiomixer module for ESP32-S2, built the .uf2 file, installed it on my feathers2, and was able to mix WaveFiles.

Here's the diff:

    diff --git a/ports/esp32s2/mpconfigport.mk b/ports/esp32s2/mpconfigport.mk
    index 1f5021bc1..479725211 100644
    --- a/ports/esp32s2/mpconfigport.mk
    +++ b/ports/esp32s2/mpconfigport.mk
    @@ -18,6 +18,7 @@ CIRCUITPY_AUDIOBUSIO = 1
     CIRCUITPY_AUDIOBUSIO_PDMIN = 0
     CIRCUITPY_AUDIOBUSIO_I2SOUT = 1
     CIRCUITPY_AUDIOIO = 0
    +CIRCUITPY_AUDIOMIXER = 1
     CIRCUITPY_CANIO = 1
     CIRCUITPY_COUNTIO = 1
     CIRCUITPY_DUALBANK = 1

closed 2021-07-03 by microDev1