adafruit/circuitpython #4956: OLED 128x64 FeatherWing stopped working

jepler, 2021-07-01:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.2-734-g6ee9acc90-dirty on 2021-07-01; Adafruit Feather RP2040 with rp2040
```

### Code/REPL

```python
import time
import board
import displayio
import adafruit_displayio_sh1107

displayio.release_displays()
# oled_reset = board.D9

# Use for I2C
i2c = board.I2C()
display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)

# SH1107 is vertically oriented 64x128
WIDTH = 128
HEIGHT = 64
BORDER = 2

display = adafruit_displayio_sh1107.SH1107(display_bus, width=WIDTH, height=HEIGHT)

while True:
    print(time.time())
    time.sleep(1)
```

### Behavior

The display is garbled or does not update

### Description

I manually determined that this appears to be introduced by c1c101ca2dcdfc81b008b933bf50d4053f4624dd (#4864), which changed addressing to support the sh1106g on the MacroPad.

### Additional information

Note that when switching from a working revision to a non-working one, the display may seem to retain its prior content and trick you into thinking all is well.

jepler, 2021-07-08:
@ladyada has indicated that this may be better solved by a change in the adafruit_displayio_sh1107 library.

ladyada, 2021-07-09:
fix submitted, plz try it :)

jepler, 2021-07-09:
Fixed in the library.

closed 2021-07-09 by jepler