adafruit/circuitpython #4959: Remove display.Group max_size parameter from libraries

lesamouraipourpre, 2021-07-01:
Now that `max_size` has been disabled in `shared-bindings/displayio/Group.c` ready for removal in CircuitPython 7, it needs to be removed from all the libraries that make use of it before this can happen.
https://github.com/adafruit/circuitpython/blob/37c52110c5fdcc4b47567e85cba628470f658f84/shared-bindings/displayio/Group.c#L45
I propose to work through the libraries and make the necessary PRs. This Issue is intended as a central tracking point which will eventually allow the removal of the parameter from `shared-bindings`, hopefully before the release-candidate of CP7.

- [ ] Adafruit_CircuitPython_Display_Text - PR https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/pull/153
- [ ] Adafruit_CircuitPython_FRAM - Not Group
- [ ] to be added

Below is the output of `grep -r max_size *` against `adafruit-circuitpython-bundle-py-20210629.zip`. Some of these do not appear to be related to Group but will be checked anyway.
```
examples/st7735r_simpletest.py:splash = displayio.Group(max_size=10)
examples/displayio_ssd1306_featherwing.py:splash = displayio.Group(max_size=10)
examples/display_shapes_circle_animation.py:main_group = displayio.Group(max_size=2)
examples/st7789_tft_gizmo_simpletest.py:splash = displayio.Group(max_size=10)
examples/st7789_tft_gizmo_simpletest.py:text_group = displayio.Group(max_size=10, scale=2, x=50, y=120)
examples/dash_display_advancedtest.py:rgb_group = displayio.Group(max_size=9)
examples/displayio_layout_gridlayout_simpletest.py:main_group = displayio.Group(max_size=10)
examples/displayio_layout_gridlayout_simpletest.py:    max_size=10,
examples/st7735r_128x160_simpletest.py:splash = displayio.Group(max_size=10)
examples/st7735r_128x160_simpletest.py:text_group = displayio.Group(max_size=10, scale=2, x=11, y=64)
examples/displayio_layout_cartesian_simpletest.py:my_group = displayio.Group(max_size=3)
examples/hx8357_simpletest.py:splash = displayio.Group(max_size=10)
examples/hx8357_simpletest.py:text_group = displayio.Group(max_size=10, scale=3, x=137, y=160)
examples/ili9341_shield_simpletest.py:splash = displayio.Group(max_size=10)
examples/ili9341_shield_simpletest.py:text_group = displayio.Group(max_size=10, scale=3, x=57, y=120)
examples/progressbar_magtag_simpletest.py:splash = displayio.Group(max_size=10)
examples/clue_ble_color_patchwork.py:color_select_group = displayio.Group(max_size=10)
examples/clue_ble_color_patchwork.py:color_select_text_group = displayio.Group(max_size=4, scale=3)
examples/ssd1331_simpletest.py:splash = displayio.Group(max_size=10)
examples/displayio_sh1106_simpletest.py:splash = displayio.Group(max_size=10)
examples/displayio_layout_icon_animated_simpletest.py:main_group = displayio.Group(max_size=2)
examples/progressbar_vertical_simpletest.py:splash = displayio.Group(max_size=10)
examples/progressbar_accelerometer.py:main_group = displayio.Group(max_size=10)
examples/st7735r_minitft_featherwing_simpletest.py:splash = displayio.Group(max_size=10)
examples/st7735r_minitft_featherwing_simpletest.py:text_group = displayio.Group(max_size=10, scale=2, x=11, y=40)
examples/display_button_customfont.py:splash = displayio.Group(max_size=20)
examples/display_shapes_sparkline_triple.py:my_group = displayio.Group(max_size=20)
examples/display_shapes_simpletest.py:splash = displayio.Group(max_size=20)
examples/cursorcontrol_custom_cursor.py:splash = displayio.Group(max_size=5)
examples/st7789_320x240_simpletest.py:splash = displayio.Group(max_size=10)
examples/st7789_320x240_simpletest.py:text_group = displayio.Group(max_size=10, scale=3, x=57, y=120)
examples/monsterm4sk_pumpkin_shifting_eyes.py:left_group = displayio.Group(max_size=4)
examples/monsterm4sk_pumpkin_shifting_eyes.py:right_group = displayio.Group(max_size=4)
examples/gizmo_tft_thermometer.py:    text_group = displayio.Group(max_size=1, scale=TEXT_SCALE, x=0, y=0)
examples/display_text_pyportal.py:splash = displayio.Group(max_size=len(fonts) * len(demos))
examples/gizmo_tft_demo.py:splash = displayio.Group(max_size=10)
examples/gizmo_tft_demo.py:text_group = displayio.Group(max_size=10, scale=2, x=50, y=120)
examples/displayio_sh1107_simpletest.py:splash = displayio.Group(max_size=10)
examples/cursorcontrol_buttons_text.py:splash = displayio.Group(max_size=22)
examples/ssd1325_simpletest.py:splash = displayio.Group(max_size=10)
examples/ssd1325_simpletest.py:    max_size=10,
examples/monsterm4sk_rainbow_stars.py:left_group = displayio.Group(max_size=4, scale=3)
examples/monsterm4sk_rainbow_stars.py:right_group = displayio.Group(max_size=4, scale=3)
examples/displayio_layout_simpletest.py:main_group = displayio.Group(max_size=10)
examples/displayio_layout_simpletest.py:    max_size=10,
examples/hx8357_pitft_simpletest.py:splash = displayio.Group(max_size=10)
examples/hx8357_pitft_simpletest.py:text_group = displayio.Group(max_size=10, scale=3, x=137, y=160)
examples/st7789_240x240_simpletest_Pimoroni_Pico_Explorer.py:splash = displayio.Group(max_size=10)
examples/st7789_240x240_simpletest_Pimoroni_Pico_Explorer.py:text_group = displayio.Group(max_size=10, scale=2, x=50, y=120)
examples/cursorcontrol_simpletest.py:splash = displayio.Group(max_size=5)
examples/advanced_examples/clue_ams_remote_advanced.py:group = displayio.Group(max_size=25)
examples/st7789_240x240_bonnet_simpletest.py:splash = displayio.Group(max_size=10)
examples/st7789_240x240_bonnet_simpletest.py:text_group = displayio.Group(max_size=10, scale=2, x=50, y=120)
examples/displayio_layout_switch_multiple.py:my_group = displayio.Group(max_size=8)
examples/ili9341_simpletest.py:splash = displayio.Group(max_size=10)
examples/ili9341_simpletest.py:text_group = displayio.Group(max_size=10, scale=3, x=57, y=120)
examples/ssd1327_simpletest.py:splash = displayio.Group(max_size=10)
examples/ssd1327_simpletest.py:    max_size=10,
examples/client_examples/battery_daughter.py:splash = displayio.Group(max_size=10)
examples/displayio_layout_dial_simpletest.py:my_group = displayio.Group(max_size=1)
[... 91 lines cut ...]
lib/adafruit_pybadger/pybadger_base.py:            self._background_group = displayio.Group(max_size=30)
lib/adafruit_pybadger/pybadger_base.py:        background_group = displayio.Group(max_size=30)
lib/adafruit_pybadger/pybadger_base.py:        business_card_splash = displayio.Group(max_size=4)
lib/adafruit_displayio_layout/layouts/grid_layout.py:    :param int max_size: (Optional) this will get passed through to the
lib/adafruit_displayio_layout/layouts/grid_layout.py:    def __init__(self, x, y, width, height, grid_size, cell_padding, max_size=None):
lib/adafruit_displayio_layout/layouts/grid_layout.py:        if not max_size:
lib/adafruit_displayio_layout/layouts/grid_layout.py:            max_size = grid_size[0] * grid_size[1]
lib/adafruit_displayio_layout/layouts/grid_layout.py:        super().__init__(x=x, y=y, max_size=max_size)
lib/adafruit_displayio_layout/widgets/switch_round.py:            my_group = displayio.Group(max_size=10) # make a group that can hold 10 items
lib/adafruit_displayio_layout/widgets/switch_round.py:        super().__init__(x=x, y=y, height=height, width=width, **kwargs, max_size=4)
lib/adafruit_displayio_layout/widgets/switch_round.py:        # using "max_size=XX"
lib/adafruit_displayio_layout/widgets/flip_input.py:        super().__init__(**kwargs, max_size=4)
lib/adafruit_displayio_layout/widgets/flip_input.py:            max_size=1,
lib/adafruit_displayio_layout/widgets/cartesian.py:        my_group = displayio.Group(max_size=10) # make a group that can hold 10 items
lib/adafruit_displayio_layout/widgets/cartesian.py:        super().__init__(**kwargs, max_size=3)
lib/adafruit_displayio_layout/widgets/icon_widget.py:    :param int max_size: (Optional) this will get passed through to the
lib/adafruit_displayio_layout/widgets/icon_widget.py:     displayio.Group constructor. ``max_size`` should be set to the maximum number of
lib/adafruit_displayio_layout/widgets/annotation.py:        super().__init__(max_size=3)
lib/adafruit_displayio_layout/widgets/widget.py:        # **kwargs should include `max_size`from the subclass implementation
lib/adafruit_displayio_layout/widgets/icon_animated.py:    :param int max_size: (Optional) this will get passed through to the
lib/adafruit_displayio_layout/widgets/icon_animated.py:     displayio.Group constructor. ``max_size`` should be set to the maximum number of
lib/adafruit_displayio_layout/widgets/icon_animated.py:                "`IconAnimated.init_class(display, max_scale, max_size, max_color_depth)`\n"
lib/adafruit_displayio_layout/widgets/dial.py:        super().__init__(**kwargs, max_size=3)
lib/adafruit_displayio_layout/widgets/dial.py:        # using "max_size=XX"
```

FoamyGuy, 2021-07-01:
Thanks for putting this together. We'll need to run a similar pass through the learn guide repo to change instances of this as well.

jposada202020, 2021-07-02:
@lesamouraipourpre I give all the thanks here for all these PRs. This is hard work!!!

jposada202020, 2021-07-05:
Arrowline Done --> https://github.com/jposada202020/CircuitPython_ArrowLine/releases/tag/0.8.1
Candlesticks Done --> https://github.com/jposada202020/CircuitPython_Candlesticks/releases/tag/0.6.1
Styles Done --> https://github.com/jposada202020/CircuitPython_styles/releases/tag/0.5.1

PR #5040 (2021-07-22): Remove the max_size parameter from displayio.Group
Remove `max_size` parameter from `displayio.Group`
Closes #4959

**This is potentially a breaking change as the constructor signature is changing.**

Status of dependents:

*Frozen modules*
```bash
> grep -r max_size frozen | wc
      0       0       0
```
All frozen modules that use `max_size` have been updated

*Adafruit_CircuitPython_Bundle libraries* - after the next release
```bash
>> Adafruit_CircuitPython_Bundle  main
> .update-submodules.sh
> grep -r max_size * | grep -v adafruit_fram
libraries/drivers/displayio_sh1106/examples/displa [... cut]
closed 2021-07-22 by tannewt