adafruit/circuitpython #4962: ESP32 hard crash on i2c display

jfurcean, 2021-07-02:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.3-493-gc37f354d2 on 2021-07-02; FeatherS2 with ESP32S2
```

### Code/REPL

```python
import busio
import board
import displayio
import adafruit_displayio_ssd1306

displayio.release_displays()

i2c = busio.I2C(board.SCL, board.SDA)

WIDTH = 128
HEIGHT = 32  # Change to 64 if needed
BORDER = 5
display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=WIDTH, height=HEIGHT)
```

### Behavior

This code crashes on the final line using a Feather S2 and the STEMMA QT 0.91" OLED Display 128x32.

### Description

- might be related to #4951
- might have started with the #4195
- other i2c devices worked like the i2c Stemma QT Rotary Encoder

meltdown03, 2021-07-03:
I get also a hard crash when using the the SPI display on the Kaluga V1.2 . It worked a month ago. Happens at the same line of code (except SPI bus)

meltdown03, 2021-07-03:
SPI seems to work now, as of build today (c37f354d2d9faca7bb827c775335421ffa98bf33). I see some SPI fixes were commited

jerryneedell, 2021-07-03:
I2C works OK on an ESP32s2 SAOLA-WROVER with the current tip of main.
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.3-493-gc37f354d2 on 2021-07-03; Saola 1 w/Wrover with ESP32S2
>>> import busio
>>> import board
>>> import displayio
>>> import adafruit_displayio_ssd1306
>>> 
>>> displayio.release_displays()
>>> i2c = busio.I2C(board.IO2, board.IO1)
>>> display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
>>> display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)
>>> 
>>> 
>>> 
```

jerryneedell, 2021-07-03:
And it also works now on a FeatherS2 
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.3-493-gc37f354d2 on 2021-07-03; FeatherS2 with ESP32S2
>>> import busio
>>> import board
>>> import displayio
>>> import adafruit_displayio_ssd1306
>>> 
>>> displayio.release_displays()
>>> 
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> 
>>> display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
>>> display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)
>>> 
>>> 
```

jerryneedell, 2021-07-03:
Trying to reproduce the failing test setup - I have a STEMMA OLED SSD1306 132x64.
It also works for me on the FeatherS2
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.3-493-gc37f354d2 on 2021-07-03; FeatherS2 with ESP32S2
>>> import busio
>>> import board
>>> import displayio
>>> import adafruit_displayio_ssd1306
>>> 
>>> displayio.release_displays()
>>> 
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> display_bus = displayio.I2CDisplay(i2c, device_address=0x3D)
>>> display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=64)
>>> 
>>> 
>>> 
```

jfurcean, 2021-07-03:
It works fine when running in REPL the first time, but crashes on `import adafruit_displayio_ssd1306` after a soft reboot.

```
Adafruit CircuitPython 7.0.0-alpha.3-493-gc37f354d2 on 2021-07-02; FeatherS2 with ESP32S2
>>> import busio
>>> import board
>>> import displayio
>>> import adafruit_displayio_ssd1306
>>> displayio.release_displays()
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
>>> display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.3-493-gc37f354d2 on 2021-07-02; FeatherS2 with ESP32S2
>>> import busio
>>> import board
>>> import displayio
>>> import adafruit_displayio_ssd1306

```

jerryneedell, 2021-07-03:
Ah -- that reminds me of a very old issue I ran across on the feathers2 -- I had that problem if I used the .uf2 bootloader, but not if I flash with the .bin file ....
https://github.com/adafruit/tinyuf2/issues/74

I have not been using the UF2 bootloader on the featherS2 or the Saola

jerryneedell, 2021-07-03:
Ah -- never mind -- I just crashed mine trying to run the code after a soft reboot....

jerryneedell, 2021-07-03:
but it crashes before the last line -- mine crashes on the import adafruit_displayio_ssd1306 line

jerryneedell, 2021-07-03:
@jfurcean If you'd taht it works if I do a 
```
import displayio
displayio.release_displays()
```
after the soft reboot

then I can run the code repeatedly after soft reboot

jerryneedell, 2021-07-03:
I wonder if this is a displayio issue -- or even an expected behavior if the display is not released.

jerryneedell, 2021-07-03:
Well -- that does not make senses sice the code does the release, but for some reason, it has to be done before importing the ssd1306 library...

```
Adafruit CircuitPython 7.0.0-alpha.3-493-gc37f354d2 on 2021-07-03; FeatherS2 with ESP32S2
>>> import displayio
>>> displayio.release_displays()
>>> import busio
>>> ipormt board
>>> import displayio
>>> import adafruit_displayio_ssd1306
>>> 
>>> displayio.release_displays()
>>> 
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> 
>>> display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
>>> display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)
>>> 
```

jfurcean, 2021-07-03:
It also works fine with QT PY Haxpress and the QT PY RP2040 without changing the order.

```
Adafruit CircuitPython 7.0.0-alpha.3-493-gc37f354d2 on 2021-07-02; Adafruit QT Py RP2040 with rp2040
>>> import busio
>>> import board
>>> import displayio
>>> import adafruit_displayio_ssd1306
>>> displayio.release_displays()
>>> i2c = busio.I2C(board.SCL1, board.SDA1)
>>> display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
>>> display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.3-493-gc37f354d2 on 2021-07-02; Adafruit QT Py RP2040 with rp2040
>>> import busio
>>> import board
>>> import displayio
>>> import adafruit_displayio_ssd1306
>>> displayio.release_displays()
>>> i2c = busio.I2C(board.SCL1, board.SDA1)
>>> display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
>>> display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)
>>> 
>>> 
>>> 
```

jerryneedell, 2021-07-03:
it does seem like an issue with the esp32s2 ...
FWIW -- this works repeatedly with soft reboot
```
import busio
import board
import displayio
displayio.release_displays()
import adafruit_displayio_ssd1306

i2c = busio.I2C(board.SCL, board.SDA)

display_bus = displayio.I2CDisplay(i2c, device_address=0x3D)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=64)
```

jerryneedell, 2021-07-03:
oops -- accidental close

deshipu, 2021-07-03:
> Well -- that does not make senses since the code does the release, but for some reason, it has to be done before importing the ssd1306 library...

The ``adafruit_displayio_ssd1306`` library doesn't do anything on import, other than setting a few constants: https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_SSD1306/blob/main/adafruit_displayio_ssd1306.py

This sounds like some kind of a race condition — maybe there is a delay required between releasing the displays and creating a new displayio display. Maybe because it's still updating one frame in the background?

dhalbert, 2021-07-03:
A garbage collection is done after each import, so I wonder if it's a storage management problem.

PR #5220 (2021-08-25): Allocate I2C mutex with IDF
This keeps the mutex info in the same spot in memory. "Statically
allocating it" with CircuitPython meant that the buffer moved when
the I2C object is moved to keep it alive for a display.

Fixes #4962
commit 0552ce2d (2021-08-25): Allocate I2C mutex with IDF

This keeps the mutex info in the same spot in memory. "Statically
allocating it" with CircuitPython meant that the buffer moved when
the I2C object is moved to keep it alive for a display.

Fixes #4962
closed 2021-08-25 by microDev1