adafruit/circuitpython #4971: nrf52 BLE examples ?

mtk11, 2021-07-07:
There is a folder[ circuitpython/ports/nrf/examples/](https://github.com/adafruit/circuitpython/tree/main/ports/nrf/examples), but the BLE examples are not working, no ubluepy module. Any BLE examples that actually work ?

microDev1, 2021-07-07:
The `ubluepy` module is no longer in use and is replaced by `_bleio`. You can find more info and examples at:
- https://github.com/adafruit/Adafruit_CircuitPython_BLE
- https://learn.adafruit.com/circuitpython-nrf52840/bluefruit-le-connect-basics

mtk11, 2021-07-07:
@microDev1 Thanks for the links. I expected the high-level BLE library to be included in the nrf port. Also isn’t too slow providing such functionality with python code ?

closed 2021-07-07 by dhalbert

microDev1, 2021-07-08:
@mtk11 Hope the following helps. :)

> I expected the high-level BLE library to be included in the nRF port.

- Native modules provide only basic connection between the python layer and hardware. 
- Libraries are built on top of native modules and act as a wrapper for them. These make the interaction with the native modules easier and also provide functions that aren't related to hardware. 

Flash space constraints is the primary reason for not including libraries with the port.

> Also isn’t too slow providing such functionality with python code?

Yes, executing code in c is faster and there are libraries implemented as modules but this happens only when gain in performance outweighs the flash space impact and also when code needs to be run in background (i.e. outside vm).

PR #4972 (2021-07-07): Delete ports/nrf/examples directory
closes #4971.