adafruit/circuitpython #4979: pewpew10: glitchy display, random bright lines

deshipu, 2021-07-09:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.3-501-g53c01176e on 2021-07-09; PewPew 10.2 with samd21e18
```

### Code/REPL

```python
.
```

### Behavior

With CircuitPython 5.3.1 — the last version of CircuitPython that ran on the pewpew10 without issues — the display is very stable: https://youtu.be/_gLVneUf4yQ — the disappearing bricks are due to interplay between my camera's refresh rate and the PWM of the LED.

With CircuitPython 6.0.0 and later, there were bad issues with flickering due to interrupts being disabled when time registers were read. I eventually worked around this by switching to keeping time by counting frames of the animation of the display itself: https://www.youtube.com/watch?v=bLqYwwU1rS0

In CircuitPython 7 there is a new problem: the display again flickers, but in addition a single random line of the display becomes super-bright for a moment. This suggests that there is again something disabling the interrupts, this time less randomly and less often, but for longer: https://youtu.be/rFE8Epz-syk

### Description

I do realize that Adafruit does not sell the PewPew 10 and doesn't have to support it. I also realize that I am reporting this problem late in the development cycle — I simply had no time to look into it earlier. Considering this, and the fact that I don't even know where to start looking for a fix myself, I wonder if it would make sense to remove PewPew 10, possbily together with uGame 10 (which has become so slow it is unusable) from further versions of CircuitPython, and only offer versions of firmware up to 5.3.1 for download on circuitpython.org for those boards.

### Additional information

.

ladyada, 2021-07-09:
is this a raw multiplexed display? are you doing the display muxing in c?

deshipu, 2021-07-09:
@ladyada it's an 8x8 LED matrix driven from a timer interrupt. The code for the interrupt is here: https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/common-hal/_pew/__init__.c#L36-L89

This looks like something is disabling the timer interrupt periodically (or otherwise preventing it from firing on time), so that it is "late" and some of the lines get much more time than others, and so become brighter.

deshipu, 2021-07-09:
I noticed that those lines appear much less randomly and for longer time when USB cable is connected. I'm going to try bisect tomorrow to see if I can find the commit responsible.

deshipu, 2021-07-09:
7.0.0-alpha.2 works fine, 7.0.0-alpha.3 is flickering, so it's something recent.

deshipu, 2021-07-09:
I found the reason. Somehow the PR https://github.com/adafruit/circuitpython/pull/4200 got reverted, reverting the workaround I made for problems with ``time.monotonic()`` in 6.0.

I found that my repository wasn't tagged properly for some reason.

I'm repeating it in https://github.com/adafruit/circuitpython/pull/4980 — with a proper tag, this should fix it.

closed 2021-07-12 by deshipu