adafruit/circuitpython #4982: Update libraries for the changes to OnDiskBitmap

lesamouraipourpre, 2021-07-10:
Following the changes to OnDiskBitmap merged in https://github.com/adafruit/circuitpython/pull/4823 there are a number of libraries that need to be updated and tracked here:

`Task list coming soon...`

`grep -r "pixel_shader=.*ColorConverter()" * | sort` at the root of `Adafruit_CiruitPython_Bundle` gives
```
libraries/drivers/il0373/examples/il0373_1.54_color.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il0373/examples/il0373_2.13_color.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il0373/examples/il0373_2.9_color.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il0373/examples/il0373_2.9_grayscale.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il0373/examples/il0373_flexible_2.13_monochrome.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il0373/examples/il0373_flexible_2.9_monochrome.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il0373/examples/il0373_simpletest.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il0373/README.rst:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il0398/examples/il0398_4.2_color.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il0398/examples/il0398_simpletest.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il0398/README.rst:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il91874/examples/il91874_simpletest.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/il91874/README.rst:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/ssd1608/examples/ssd1608_simpletest.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/ssd1608/README.rst:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/ssd1675/examples/ssd1675_2.13_monochrome.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/ssd1675/examples/ssd1675_simpletest.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/ssd1675/README.rst:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/ssd1680/examples/ssd1680_simpletest.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/ssd1680/README.rst:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/ssd1681/examples/ssd1681_simpletest.py:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/drivers/ssd1681/README.rst:    t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
libraries/helpers/displayio_layout/adafruit_displayio_layout/widgets/icon_widget.py:            tile_grid = TileGrid(image, pixel_shader=ColorConverter())
libraries/helpers/gizmo/examples/gizmo_eink_simpletest.py:    sprite = displayio.TileGrid(picture, pixel_shader=displayio.ColorConverter())
libraries/helpers/gizmo/examples/gizmo_tft_thermometer.py:    tile_grid = displayio.TileGrid(bitmap, pixel_shader=displayio.ColorConverter())
libraries/helpers/portalbase/adafruit_portalbase/graphics.py:                pixel_shader=displayio.ColorConverter(),
libraries/helpers/pybadger/adafruit_pybadger/pybadger_base.py:                    on_disk_bitmap, pixel_shader=displayio.ColorConverter()
libraries/helpers/pybadger/adafruit_pybadger/pybadger_base.py:                on_disk_bitmap, pixel_shader=displayio.ColorConverter()
libraries/helpers/pyoa/adafruit_pyoa.py:                background, pixel_shader=displayio.ColorConverter(), x=0, y=0
libraries/helpers/slideshow/adafruit_slideshow.py:                odb, pixel_shader=displayio.ColorConverter()
libraries/helpers/turtle/adafruit_turtle.py:                odb, pixel_shader=displayio.ColorConverter()
libraries/helpers/turtle/adafruit_turtle.py:                pixel_shader=displayio.ColorConverter(),
libraries/helpers/turtle/adafruit_turtle.py:                self._turtle_odb, pixel_shader=displayio.ColorConverter()
```

tannewt, 2021-07-12:
@jepler Mind tracking and reviewing these PRs?

FoamyGuy, 2021-07-22:
https://github.com/adafruit/Adafruit_Learning_System_Guides/pull/1604 in the learn guide repo as been merged, with that and the items listed above I believe the last outstanding change for this effort is the turtle library. I'm planning to take a closer look into that one this weekend.

dhalbert, 2022-05-14:
All done!

closed 2022-05-14 by dhalbert