adafruit/circuitpython #4984: ESP32-S2 DEBUG=1 builds failing

anecdata, 2021-07-11:
### CircuitPython version

```python
current main
```

### Code/REPL

```python
make (no code)
```

### Behavior

```
[658/658] Generating ld/esp32s2.project.ld
/Users/a/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: build-espressif_saola_1_wrover/firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/Users/a/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: region `dram0_0_seg' overflowed by 2816 bytes
collect2: error: ld returned 1 exit status
make: *** [build-espressif_saola_1_wrover/firmware.elf] Error 1
```

### Additional information

also observed by @microDev1 <https://discord.com/channels/327254708534116352/327298996332658690/863818850800500776>

tannewt, 2021-07-12:
Are there bugs we want to fix for 7.0.0 that need this to debug? Otherwise, I'd prefer this being a long term issue. The 7.0.0 milestone list is very long already.

anecdata, 2021-07-12:
I just wanted to look at the promiscuous wifi issue.

jepler, 2021-07-15:
```
section `.dram0.bss' will not fit in region `dram0_0_seg'
region `dram0_0_seg' overflowed by 2816 bytes
```

This is happening because debug mode changes optimization flags which in turn makes the firmware _much_ bigger.

You can work around this by disabling some modules you aren't interested in debugging (ulab is probably pretty big) when using DEBUG=1.

dhalbert, 2021-08-02:
> This is happening because debug mode changes optimization flags which in turn makes the firmware _much_ bigger.

I changed the optimization flag for DEBUG to be `-Os` and still had the same problem, so I wonder what `DEBUG` is adding to the build besides code size.

microDev1, 2021-08-02:
I had to turn off almost all of the `common-hal` modules to get a `DEBUG=1` build.
Also, @tannewt on discord suggested another way to get a debug build is by reducing tinyusb buffer size.

From [this](https://esp32.com/viewtopic.php?t=6699) esp32 forum post:
```
Because of some fixed RAM addresses used by the ESP32 ROM, there is a limit on the amount which can be statically allocated at compile time. 
Much more RAM is available at runtime via the heap (it is, as WiFive says, non-contiguous.)

There are some plans to work around this and allow more static allocation, but no ETA yet.

The best fix for the issue is to find a large statically allocated buffer in your program and change it to allocate dynamically (ie via malloc()).
```

microDev1, 2021-08-02:
`git bisect` poinst to `idf-v4.3` update commit https://github.com/adafruit/circuitpython/pull/4195/commits/e0eecd2f76b5e77271ee2dd3bb8b14d5bb3efad2.

anecdata, 2021-09-08:
Reducing wifi buffers was another suggestion to workaround this, but I couldn't find where / how to do that, so turned off modules so that it built. Just as a stake in the sand, I was able to build `DEBUG=1` on esp32-s2 `7.0.0-rc.1` with the following changes:
```
CIRCUITPY_AUDIOCORE ?= 0
CIRCUITPY_AUDIOBUSIO ?= 0
CIRCUITPY_AUDIOBUSIO_I2SOUT ?= 0
CIRCUITPY_AUDIOMIXER ?= 0
CIRCUITPY_CANIO ?= 0
CIRCUITPY_DUALBANK ?= 0
CIRCUITPY_IMAGECAPTURE ?= 0
CIRCUITPY_ROTARYIO ?= 0
CIRCUITPY_PS2IO ?= 0
```

microDev1, 2021-12-02:
Just tested `debug` build on `idf@v4.4`,  it seems to work now on boards with more than `4MB` flash.
The initial `dram0_0_seg` overflow issue seems to have been fixed in the newer idf.

jepler, 2021-12-05:
Thanks @microDev1 !

closed 2021-12-05 by jepler