adafruit/circuitpython #4985: NotImplementedError: opcode

lesamouraipourpre, 2021-07-11:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.4 on 2021-07-08; Adafruit PyPortal with samd51j20
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

"""Simpletest example using Adafruit PyPortal. Uses the touchscreen to advance between examples."""
import board
import adafruit_touchscreen
from adafruit_pybadger import pybadger
import time

# pylint: disable=invalid-name

# These pins are used as both analog and digital! XL, XR and YU must be analog
# and digital capable. YD just need to be digital
ts = adafruit_touchscreen.Touchscreen(
    board.TOUCH_XL,
    board.TOUCH_XR,
    board.TOUCH_YD,
    board.TOUCH_YU,
    calibration=((5200, 59000), (5800, 57000)),
    size=(320, 240),
)

pybadger.show_badge(
    name_string="Blinka", hello_scale=2, my_name_is_scale=2, name_scale=3
)
time.sleep(2)

cur_example = 0
prev_touch = None
while True:
    p = ts.touch_point
    if p and not prev_touch:
        cur_example += 1
        if cur_example >= 3:
            cur_example = 0
        print(cur_example)
    prev_touch = p

    if cur_example == 0:
        print("Current example", cur_example)
        pybadger.show_business_card(
            image_name="Blinka_PyPortal.bmp",
            name_string="Blinka",
            name_scale=2,
            email_string_one="blinka@",
            email_string_two="adafruit.com",
        )
    elif cur_example == 1:
        print("Current example", cur_example)
        pybadger.show_qr_code(data="https://circuitpython.org")
    elif cur_example == 2:
        print("Current example", cur_example)
        pybadger.show_badge(
            name_string="Blinka", hello_scale=2, my_name_is_scale=2, name_scale=3
        )
```

### Behavior

Run from Thonny with a PyPortal.
The test code is the `pybadger_pyportal_touchscreen.py` example from PyBadger, modified slightly with a `sleep()` and some extra `print`s
```python
%Run -c $EDITOR_CONTENT
Current example 0
Current example 0
Current example 0
Current example 0
Current example 0
Current example 0
Current example 0
Current example 0
Current example 0
Current example 0
Current example 0
Current example 0
Current example 0
Traceback (most recent call last):
  File "<stdin>", line 46, in <module>
NotImplementedError: opcode
```
The display locksup and triggers a slow device restart. It also locks up Thonny while this is happening.

### Description

The example code above works perfectly on `6.3.0` but not on `7.0.0-alpha.4` (I've not had a chance to try the previous alphas yet.)

According to Thonny, line 46 is
`            email_string_two="adafruit.com",` from
```
    if cur_example == 0:
        print("Current example", cur_example)
        pybadger.show_business_card(
            image_name="Blinka_PyPortal.bmp",
            name_string="Blinka",
            name_scale=2,
            email_string_one="blinka@",
            email_string_two="adafruit.com",
        )
```

The only place I can find which could throw that error is in CircuitPython core in `py/vm.c`
https://github.com/adafruit/circuitpython/blob/22e8a50585ad548b83dfb76b3cfa880bdf7479c7/py/vm.c#L1369

### Additional information

Also, (this might be a separate issue but I'll include it here for now,) if I touch the screen before the crash and trigger the second example to be shown (the QR code) I get the following error:
```
Current example 0
1
Current example 1
Traceback (most recent call last):
  File "<stdin>", line 50, in <module>
  File "/lib/adafruit_pybadger/pybadger_base.py", line 649, in show_qr_code
TypeError: function takes 3 positional arguments but 1 were given
```
But line 649 is a call to a 1-parameter staticmethod defined earlier in the file.
https://github.com/adafruit/Adafruit_CircuitPython_PyBadger/blob/addfdad90fbe201616e72ad632e2c3bf965d46bf/adafruit_pybadger/pybadger_base.py#L649
https://github.com/adafruit/Adafruit_CircuitPython_PyBadger/blob/addfdad90fbe201616e72ad632e2c3bf965d46bf/adafruit_pybadger/pybadger_base.py#L618-L619

lesamouraipourpre, 2021-07-11:
`7.0.0-alpha.1` - works correctly
`7.0.0-alpha.2` - works correctly
`7.0.0-alpha.3` - crashes
`7.0.0-alpha.4` - crashes

lesamouraipourpre, 2021-07-11:
In relation to the second (positional arguments) error, there was something on @FoamyGuy 's stream yesterday while working on library code that was generating a positional argument error that didn't appear to relate to the code referenced
https://youtu.be/0AwRphuQPys?t=5123

tannewt, 2021-07-12:
This sounds like a memory write issue that is stomping on byte code the VM is trying to execute.

lesamouraipourpre, 2021-07-12:
The example code above is not particularly efficient and is creating new displayio objects with every iteration and churning memory.
Changing the code so that it only updates the screen when required prevents the crashes in the short term.

lesamouraipourpre, 2021-07-22:
Re-tested with `7.0.0-alpha.5` and it longer crashes.

closed 2021-07-22 by lesamouraipourpre