adafruit/circuitpython #4988: PIO read/write slices not working

dmcomm, 2021-07-12:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.4-18-g0c6dc6f0a on 2021-07-12; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import array
import board
import time
import rp2pio
import adafruit_pioasm

asm = """
	pull
	mov isr osr
	push
"""
program = adafruit_pioasm.assemble(asm)

sm = rp2pio.StateMachine(
	program,
	frequency=1_000_000,
)

def readAnswers():
	answer = array.array("L", [0])
	time.sleep(0.001)
	while sm.in_waiting > 0:
		sm.readinto(answer)
		print("0x%08X" % answer[0], end=",")
	print()

for i in range(10, 0, -1):
	print(i, end=",")
	time.sleep(1)

print("\nwrite L 1,2,3", end="  ")
sm.write(array.array("L", [1,2,3]))
readAnswers()

print("write H 1,2,3", end="  ")
sm.write(array.array("H", [1,2,3]))
readAnswers()

print("write B 1,2,3", end="  ")
sm.write(array.array("B", [1,2,3]))
readAnswers()

print("write bytes 1,2,3", end="  ")
sm.write(bytes([1,2,3]))
readAnswers()

print("write bytes 1,2,3 [1:2]", end="  ")
sm.write(bytes([1,2,3]), start=1, end=2)
readAnswers()

print("write L 1,2,3 [1:2]", end="  ")
sm.write(array.array("L", [1,2,3]), start=1, end=2)
readAnswers()

print("write L 1,2,3 [4:8]", end="  ")
sm.write(array.array("L", [1,2,3]), start=4, end=8)
readAnswers()

"""
print("write L 1,2,3 [3:7]", end="  ")
sm.write(array.array("L", [1,2,3]), start=3, end=7)
readAnswers()
#hangs
"""

arrL4 = array.array("L", [1,2,3,4])

print("read 4 L", end="  ")
answers = array.array("L", [0] * 4)
sm.clear_rxfifo()
sm.write(arrL4)
sm.readinto(answers)
print(answers)

print ("read L [1:3]", end="  ")
answers = array.array("L", [0] * 4)
sm.clear_rxfifo()
sm.write(arrL4)
sm.readinto(answers, start=1, end=3)
print(answers)

print ("read L [4:8]", end="  ")
answers = array.array("L", [0] * 4)
sm.clear_rxfifo()
sm.write(arrL4)
sm.readinto(answers, start=4, end=8)
print(answers)

print ("read 4 B", end="  ")
answers = array.array("B", [0] * 4)
sm.clear_rxfifo()
sm.write(arrL4)
sm.readinto(answers)
print(answers)

arrL4high = array.array("L", [0x01000000,0x02000000,0x03000000,0x04000000])

print ("read 4 B high", end="  ")
answers = array.array("B", [0] * 4)
sm.clear_rxfifo()
sm.write(arrL4high)
sm.readinto(answers)
print(answers)

print ("read B high [1:3]", end="  ")
answers = array.array("B", [0] * 4)
sm.clear_rxfifo()
sm.write(arrL4high)
sm.readinto(answers, start=1, end=3)
print(answers)
```

### Behavior

```
write L 1,2,3  0x00000001,0x00000002,0x00000003,
write H 1,2,3  0x00010001,0x00020002,0x00030003,
write B 1,2,3  0x01010101,0x02020202,0x03030303,
write bytes 1,2,3  0x01010101,0x02020202,0x03030303,
write bytes 1,2,3 [1:2]  0x02020202,
write L 1,2,3 [1:2]  
write L 1,2,3 [4:8]  0x00000002,
read 4 L  array('L', [1, 2, 3, 4])
read L [1:3]  array('L', [0, 0, 0, 0])
read L [4:8]  array('L', [0, 1, 0, 0])
read 4 B  array('B', [0, 0, 0, 0])
read 4 B high  array('B', [1, 2, 3, 4])
read B high [1:3]  array('B', [0, 1, 2, 0])
```

### Description

I didn't see anything in the API docs about what array type is wanted. Logically it would be 32 bits, and that behaves as expected when no slicing is done. I've seen tutorial examples using 8 or 16 bits. Writing bytes seems to fill the word with duplicates, and reading bytes seems to take the high bits of each word. However, the slicing seems to work only for bytes.

tannewt, 2021-07-12:
What is the behavior you expect? Its not clear to me what you think the issue is.

I *think* this is all correct. We use the array typecode to do an 8, 16 or 32 bit transfer. The memory structure of the RP2040 will then do 32 bit writes to the peripheral and mirror the 8 or 16 bit value to the other bits.

dmcomm, 2021-07-12:
Based on what you just said, I expect `start` and `end` to refer to the indexes of the input array for 16 and 32 bits, as well as for 8. (The mirroring was just a little surprising, and had me confused about what the intended behaviour is.)

jepler, 2021-12-22:
I think it would be good to consider that this current behavior (though surprising!) is probably shared with e.g., i2c readinto.

tannewt, 2021-12-23:
Does i2c readinto support array input? I suppose it does but doesn't do the offset by element either. Maybe we should consider changing it as well for CP 8. I think going by element size makes more sense.

tannewt, 2022-02-11:
FYI I've added more info about element size here: https://github.com/adafruit/circuitpython/pull/6011

fivesixzero, 2022-04-22:
Ran into this issue while working on a PIO driver for the HX711 load cell amplifier.

It sends out 24 bits at a time and Python's built-in types don't offer anything to map to that properly. Based on a quick read of the docs I figured I could just throw a 3-element `bytearray()` into `readinto()` and it'd retrieve RX FIFO data in 24-bit chunks, 8 bits at a time. That resulted in some really weird output that didn't match what was on the wire and didn't seem to have a discernable pattern.

After a more thorough doc re-read, I caught the info @tannewt dropped in the above-mentioned PR. Gave me the idea to try padding the data going into the RX FIFO to make sure that it aligns with Python's 32-bit signed int type. That got things working as expected after tweaking the `StateMachine` config for a 32-bit threshold and auto-push.

Still not a perfect "binary read-in", which is to be expected given Python's limited bit-level built-ins. But hopefully if someone else runs into this in the future the idea of padding things out to fill an appopriate Python type might help.

dmopalmer, 2023-01-11:
This, I would say, is a bug and is not consistent with the documentation.  Until it is fixed, there should be a note to that effect in the documentation with a work-around, but then that will cause the user's code to fail when the bug is fixed.

If there are, e.g. 3 entries in the FIFO, and they are to be read into an array of uint32, starting at element 1, 
```
dest = array.array('L', [99]*10)
wordsize=4
sm.readinto(dest, start=1*wordsize, end=(1+sm.in_waiting)*wordsize)
```
is the way to successfully read them in, even though the docs imply that you should not multiply by wordsize.  I have had no luck at all using slices of dest.

I am not set up to compile circuitpython (is there a docker with the required tools etc?) or I would take a go at it.
`ports/raspberrypi/bindings/rp2pio/StateMachine.c:613`
```
    bool ok = common_hal_rp2pio_statemachine_readinto(self, ((uint8_t *)bufinfo.buf) + start, length, stride_in_bytes, args[ARG_swap].u_bool);
```
should probably be something like
```
    bool ok = common_hal_rp2pio_statemachine_readinto(self, ((uint8_t *)bufinfo.buf) + start*stride_in_bytes, length*stride_in_bytes, stride_in_bytes, args[ARG_swap].u_bool);
```
but handling the striding is tricky and I don't know the library routines.

Attached is a test program.

[piotest.py.gz](https://github.com/adafruit/circuitpython/files/10389223/piotest.py.gz)

PR #7444 (2023-01-14): Have start and end kwargs respect element size
The comment says it is `buffer[start:end]` but it assumed elements were a single byte long. Now it correctly does multibyte elements from array.array.

Fixes #4988

I only compiled it. I'm hoping @dmopalmer and @dmcomm will give it a try.
commit aa2ba4d8 (2023-01-14): Have start and end kwargs respect element size

The comment says it is `buffer[start:end]` but it assumed elements
were a single byte long. Now it correctly does multibyte elements
from array.array.

Fixes #4988
closed 2023-01-14 by dhalbert