adafruit/circuitpython #4989: PIO read/write interruption?

dmcomm, 2021-07-12:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.4-18-g0c6dc6f0a on 2021-07-12; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import array
import time
import rp2pio
import adafruit_pioasm

asm = """
	pull
	mov isr osr
	push
"""
program = adafruit_pioasm.assemble(asm)

sm = rp2pio.StateMachine(
	program,
	frequency=1_000_000,
)

item = array.array("L", [0])

for i in range(10):
	time.sleep(2)
	print(i)
	sm.write(item)
```

### Behavior

StateMachine.write and .readinto became non-interruptible in a recent change, presumably ab52a92 . The problem is, they can easily block forever due to a mistake in the user program, so this is effectively a crash. There must have been a good reason for doing that, but I thought I should mention it in case this result was overlooked.

tannewt, 2021-07-12:
@dhalbert How would you like to handle PIO mistakes?

dhalbert, 2021-07-15:
Leave as interruptible for user code, but internal uses of PIO should be able to be non-interruptible.