adafruit/circuitpython #4998: Add support for the STM32L4R5

bsatrom, 2021-07-13:
We are working on a new Feather board based on the STM32L4R5 and I would love to have CircuitPython support for the board. I know CPy already supports the F4, but I would love any guidance you all could provide on how to get started with a new port, where to look, etc.

Thanks!

tannewt, 2021-07-13:
Hi @bsatrom, that chip looks neat! Looks like TinyUSB already has support for it: https://github.com/hathach/tinyusb/tree/master/hw/bsp/stm32l4r5nucleo

Our stm port (in ports/stm) relies on the STM HAL so it shouldn't be too bad to add support within that port. The README includes build instructions: https://github.com/adafruit/circuitpython/tree/main/ports/stm It doesn't look like there is any L4R5 support yet in CP so you'll likely want to mimic an existing board and mirror the chip family specific files with ports/stm.

I recommend joining our Discord chat as well to ask any CP related questions you have: https://adafru.it/discord

hierophect, 2021-07-13:
I would encourage you to check out #3069 and #3192, which is where we discussed adding the F1 series - I laid out the steps required to add a new STM32 family in some detail there, and you can see the commits involved in adding the new family. While we ultimately closed it (the selection dev boards for the high end F1s wasn't great) there are parts of that implementation I'd encourage you to try and re-use, such as the page-based flash implementation.

I've made a number of python utilities in the `ports/stm/tools` directory for parsing the tables of pins and peripherals in STM32 datasheets, if you can use a program like [Tabula](https://tabula.technology/) to turn them into a .csv file. They might need some tweaking to fit the L4 setup but save a LOT of time if you can get them working.

hathach, 2021-07-14:
There is issue running tinyusb on crystaless L4, I haven't got time to look at this issue since https://github.com/hathach/tinyusb/issues/126

bsatrom, 2021-07-14:
Awesome, this is plenty to get me started, thanks all!

bsatrom, 2021-07-28:
@hathach the board we're building has a 32KHz crystal on it (https://www.digikey.com/en/products/detail/abracon-llc/ABS06-107-32-768KHZ-T/4291565) so I would presume we would be ok, but please let me know if that's not the case.

hathach, 2021-07-29:
I dont know for sure if it is due to crystalless or not. But I haven't got tinyusb running on L4 Discovery board, and didn't have time to look further at it.You could give it a try.

bsatrom, 2021-08-03:
Hey @hierophect thanks for the punch-list of items to port at the link above. I've pulled those into my port and have started working through them. One question about this item...

> Create a board family and SoC folder in stm/peripherals/, with clock setup code brought in from CubeMX, and pin/peripheral/ definitions for the SoC. Pin and peripheral files can be created with the python scripts in tools/.

Specifically on getting clock code from CubeMX, can you elaborate on where I can find this? I've created a new project for the STM32L4R5ZIY6 and am poking through the code, but nothing is jumping out. Do i need to pull this from the Clock configuration screen? 

[image]

tannewt, 2021-08-04:
My guess is that you'd move the auto-generated code into something like: https://github.com/adafruit/circuitpython/blob/main/ports/stm/peripherals/stm32f7/clocks.c

bsatrom, 2021-08-05:
Thanks @tannewt I created a clone of the peripherals for the f4r5 but was struggling to connect the dots based on the code generated by CubeMX. I'll take another look though, I'm sure I'm close to finding the right spot.

tannewt, 2021-08-05:
Feel free to join our Discord. We are usually in #circuitpython-dev during the US workday. https://adafru.it/discord

m-mcgowan, 2021-09-15:
Thanks for the help so far - our port is coming along nicely, USB and digitalio are working! On our feather board (as with others), it makes sense to initialize certain pins as output, e.g. the LED pin, and in our case ENABLE_3V3.   In the CP examples, the script sets the direction of the pin before using it. Is there some way we can have this setup beforehand and also have the corresponding digitalInOut object created so we can avoid the boilerplate?  
 
 ```
led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT
 ```
 
 and with I2C, we see code like
 ```
import board
import busio as io
i2c = io.I2C(board.SCL, board.SDA)
 ```
Is there a way we can add these attributes to `board` so that the script doesn't have to define them and the peripherals are available for immediate use?
 
 Something like
 ```
board.led = digitalio.DigitalInOut(board.LED)
board.led.direction = digitalio.Direction.OUTPUT
 board.i2c = io.I2C(board.SCL, board.SDA)
 board.i2c2  = ...
 # similarly for I2C1, I2C2 etc..
 board.Serial1 = busio.UART(board.TX, board.RX)
 # similarly for UARTS1..6
 board.spi = ...

 ```
 
 
I have used the STM32 HAL in board_reset() to initialize the pins as output which gives them the correct state, but they are overwritten when the pins are reset.  Marking the pin as "never reset" in board_reset() claims the pin and prevents creating any wrapper for it (e.g. DigitalInOut)
 
My motivation here is that I feel having the peripherals available on the board makes developing even easier, but feel free to tell me why this might not be a good idea, e.g. memory consumption and contention for pins perhaps?

Thanks for any guidance here!

tannewt, 2021-09-16:
Nice work! Sounds like you are ready to make a PR.

Generally we don't pre-initialize any individual pins. The one exception I could see is the power enable pin that you may want to turn on in board_init and off in board_deinit. However, this will prevent code.py control of it which some people usually want.

For I2C, SPI and UART we usually have default busses that are available as board.I2C etc. This requires entries in the board table and definition of the `DEFAULT_*` macros like here: https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/boards/feather_m4_express/mpconfigboard.h#L13-L21

We don't usually have more than one of each though. Remember that CircuitPython is more flexible than Arduino by not requiring you to fix functionality at compile time.

bsatrom, 2021-09-22:
Hey @tannewt! Thanks again for all of your help here! We're pretty close to ready for a PR and should have it up in the next few days. I did have a related question for you: Once we get that up and merged, what's the process for getting the port listed on CircuitPython.org? I can start running down any content you all would need for that in the meantime.

FoamyGuy, 2021-09-22:
@bsatrom You make a PR to circuitpython-org repo, check out this guide for details: https://learn.adafruit.com/how-to-add-a-new-board-to-the-circuitpython-org-website

It may be helpful to look at the changes from a previous PR as well: https://github.com/adafruit/circuitpython-org/pull/768

bsatrom, 2021-09-24:
Thanks @FoamyGuy I have a PR up against our fork of the `circuitpython-org` repo, and once we make an official PR here, I'll circle back and make an official PR against your main repo.

bsatrom, 2021-09-28:
PR is up [here](https://github.com/adafruit/circuitpython/pull/5403)!

closed 2022-03-24 by bsatrom