adafruit/circuitpython #5000: 远程更新代码code.py

ZHNscl, 2021-07-14:
I want to implement FTP on IOTS2 module through circulePython.
Because OTA updates the underlying firmware, I can't update the main code of code.py, which bothers me a lot.

Referring to some of the routines in mircuitPython, one of them didn't work on my board.

`from setuptools import setup`
Post-run prompt "No module name 'setuptools'"
What do I need to solve？

Neradoc, 2021-07-14:
> I want to implement FTP on IOTS2 module through circulePython.
> Because OTA updates the underlying firmware, I can't update the main code of code.py, which bothers me a lot.

I believe #3454 might allow what you want to do, by uploading the file to the board, then resetting to run a script that replaces the old `code.py` with the new one and reset again to run the new `code.py` from then on. You have to make the drive writable too, that's covered in [the learn guides](https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage).

tannewt, 2021-07-14:
`setuptools` is a CPython module that CircuitPython doesn't provide.

tannewt, 2021-07-14:
We can help with questions like this on [Discord](https://adafru.it/discord) or if you are in china we have a CircuitPython WeChat group. You can add me on wechat as tannewt and I can add you to the group.

closed 2021-07-14 by tannewt