adafruit/circuitpython #5004: 7.0.0-Alpha.4 (2021-07-15) can't boot on Feather STM32F405 Express.

water5, 2021-07-14:
### CircuitPython version

```python
CircuitPython 7.0.0-Alpha.4 on 2021-07-15, Feather STM32F405 Express.
```

### Code/REPL

```python
None
```

### Behavior

NeoPixel indicator short blinking (green) several times, then off, no REPL, CIRCUITPY drive not appear.

jerryneedell, 2021-07-14:
Try the 'absolute newest' build   the "S3"  https://circuitpython.org/board/feather_stm32f405_express/
7.0 .0 alpha4 did not work for me either (as you described) but the latest build does work

water5, 2021-07-14:
7.0.0-Alpha.4 (2021-07-15 github custom build) on 'STM32F411CE Black Pill' works fine.

jerryneedell, 2021-07-15:
Ah - I found that 7.0.0 Alpha.4 does actually boot and mount but it took quite a long time -- maybe 20 - 30 seconds after pressing the RESET button
```

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.4 on 2021-07-08; Adafruit Feather STM32F405 Express with STM32F405RG
>>> 

```

also same with newest. build
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.4-67-ge3bc800bb on 2021-07-15; Adafruit Feather STM32F405 Express with STM32F405RG
>>> 

```

jepler, 2021-08-18:
I see the same, with #5172 (just in case it was a tinyusb bug that was subsequently fixed):
```
Aug 18 10:30:31 localhost kernel: [2395904.328570] usb 3-1.3: new full-speed USB device number 110 using xhci_hcd
Aug 18 10:30:31 localhost kernel: [2395904.447250] usb 3-1.3: New USB device found, idVendor=0483, idProduct=df11, bcdDevice=22.00
Aug 18 10:30:31 localhost kernel: [2395904.447255] usb 3-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 18 10:30:31 localhost kernel: [2395904.447257] usb 3-1.3: Product: STM32  BOOTLOADER
Aug 18 10:30:31 localhost kernel: [2395904.447259] usb 3-1.3: Manufacturer: STMicroelectronics
Aug 18 10:30:31 localhost kernel: [2395904.447260] usb 3-1.3: SerialNumber: 20843696424D
Aug 18 10:31:55 localhost kernel: [2395988.136101] usb 3-1.3: USB disconnect, device number 110
Aug 18 10:31:56 localhost kernel: [2395989.893231] usb 3-1.3: new full-speed USB device number 111 using xhci_hcd
Aug 18 10:32:12 localhost kernel: [2396005.857330] usb 3-1.3: New USB device found, idVendor=239a, idProduct=805a, bcdDevice= 1.00
Aug 18 10:32:12 localhost kernel: [2396005.857334] usb 3-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 18 10:32:12 localhost kernel: [2396005.857335] usb 3-1.3: Product: Feather STM32F405 Express
Aug 18 10:32:12 localhost kernel: [2396005.857336] usb 3-1.3: Manufacturer: Adafruit Industries LLC
Aug 18 10:32:12 localhost kernel: [2396005.857337] usb 3-1.3: SerialNumber: 420052000F504D4254363220
Aug 18 10:32:15 localhost kernel: [2396008.559669] cdc_acm 3-1.3:1.0: ttyACM1: USB ACM device
Aug 18 10:32:21 localhost kernel: [2396013.965380] usb-storage 3-1.3:1.2: USB Mass Storage device detected
Aug 18 10:32:21 localhost kernel: [2396013.965822] scsi host9: usb-storage 3-1.3:1.2
Aug 18 10:32:24 localhost kernel: [2396017.568679] scsi host9: scsi scan: INQUIRY result too short (5), using 36
Aug 18 10:32:24 localhost kernel: [2396017.568687] scsi 9:0:0:0: Direct-Access     Adafruit Feather STM32F40 1.0  PQ: 0 ANSI: 2
Aug 18 10:32:24 localhost kernel: [2396017.568997] sd 9:0:0:0: Attached scsi generic sg2 type 0
Aug 18 10:32:27 localhost kernel: [2396020.271462] sd 9:0:0:0: [sdc] 4089 512-byte logical blocks: (2.09 MB/2.00 MiB)
Aug 18 10:32:28 localhost kernel: [2396021.172777] sd 9:0:0:0: [sdc] Write Protect is off
Aug 18 10:32:28 localhost kernel: [2396021.172783] sd 9:0:0:0: [sdc] Mode Sense: 03 00 00 00
Aug 18 10:32:29 localhost kernel: [2396022.073096] sd 9:0:0:0: [sdc] No Caching mode page found
Aug 18 10:32:29 localhost kernel: [2396022.073102] sd 9:0:0:0: [sdc] Assuming drive cache: write through
Aug 18 10:32:34 localhost kernel: [2396027.494522]  sdc: sdc1
Aug 18 10:32:39 localhost kernel: [2396032.899412] sd 9:0:0:0: [sdc] Attached SCSI removable disk
```

jepler, 2021-08-18:
What do y'all have in `code.py`?  I bisected the problem I was seeing to the merge of #3191 (between beta.2 and rc.0) but it went away after moving my `code.py` out of the way.  `code.py` contained some code to use canbus, but there was no working bus.

The code I had was:

```python
import struct

import board
import canio
import digitalio

# If the CAN transceiver has a standby pin, bring it out of standby mode
if hasattr(board, 'CAN_STANDBY'):
    standby = digitalio.DigitalInOut(board.CAN_STANDBY)
    standby.switch_to_output(False)

# If the CAN transceiver is powered by a boost converter, turn on its supply
if hasattr(board, 'BOOST_ENABLE'):
    standby = digitalio.DigitalInOut(board.CAN_STANDBY)
    standby.switch_to_output(True)

# Use this line if your board has dedicated CAN pins. (Feather M4 CAN and Feather STM32F405)
can = canio.CAN(rx=board.CAN_RX, tx=board.CAN_TX, baudrate=250_000, auto_restart=True)
# On ESP32S2 most pins can be used for CAN.  Uncomment the following line to use IO5 and IO6
#can = canio.CAN(rx=board.IO6, tx=board.IO5, baudrate=250_000, auto_restart=True)
listener = can.listen(matches=[canio.Match(0x408)], timeout=.9)

old_bus_state = None
old_count = -1

while True:
    bus_state = can.state
    if bus_state != old_bus_state:
        print(f"Bus state changed to {bus_state}")
        old_bus_state = bus_state

    message = listener.receive()
    if message is None:
        print("No messsage received within timeout")
        continue

    data = message.data
    if len(data) != 8:
        print(f"Unusual message length {len(data)}")
        continue

    count, now_ms = struct.unpack("<II", data)
    gap = count - old_count
    old_count = count
    print(f"received message: count={count} now_ms={now_ms}")
    if gap != 1:
        print(f"gap: {gap}")
```

jepler, 2021-08-18:
.. and I can go all the way back up to 7.0.0-alpha.6 with no `code.py` and all is well.

PR #5175 (2021-08-20): canio: Run background tasks while waiting for message reception
While I don't have enough information to determine if this is what affected the original reporter, I had an stm32f405 feather that readily reproduced the problem. I eventually discovered that the problem affected me only when I had a program that used canio and was spending all its time waiting in `common_hal_canio_listener_receive` for a packet that never arrived. This was due to missing background task & interrupt key checking in the blocking loop in `common_hal_canio_listener_receive`.

I added the extra code to both esp32-s2,  [... cut]
commit 371f166d (2021-08-20): canio: Run background tasks while waiting for message reception

Closes: #5004
commit cabe96e1 (2021-08-20): canio: Run background tasks while waiting for message reception

Closes: #5004
closed 2021-08-20 by microDev1