adafruit/circuitpython #5008: RP2040: adding display code causes HID to break

dhalbert, 2021-07-16:
### CircuitPython version

```python
7.0.0-alpha.4 on Tiny2040
```

### Code/REPL

```python
import board
import busio
import displayio
import digitalio
from adafruit_st7789 import ST7789
import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
import usb_midi
import adafruit_midi
from adafruit_midi.note_on          import NoteOn
from adafruit_midi.note_off         import NoteOff

#  if you want to use this as an HID keyboard, set keyboard_mode to True
#  otherwise, set it to False
keyboard_mode = True
#  if you want to use this as a MIDI keyboard, set midi_mode to True
#  otherwise, set it to False
midi_mode = False

#  change keyboard shortcuts here
#  defaults are shortcuts for save, cut, copy & paste
#  comment out ctrl depending on windows or macOS
if keyboard_mode:
    keyboard = Keyboard(usb_hid.devices)
    #  modifier for windows
    ctrl = Keycode.CONTROL
    #  modifier for macOS
    #  ctrl = Keycode.COMMAND
    key0 = Keycode.S
    key1 = Keycode.X
    key2 = Keycode.C
    key3 = Keycode.V
    shortcuts = [key0, key1, key2, key3]

#  change MIDI note numbers here
if midi_mode:
    midi = adafruit_midi.MIDI(midi_out=usb_midi.ports[1], out_channel=0)
    midi_notes = [60, 61, 62, 63]

# Release any resources currently in use for the displays
displayio.release_displays()

#  spi display setup
#spi = board.SPI()
spi = busio.SPI(clock=board.GP2, MOSI=board.GP3)
tft_cs = board.GP1
tft_dc = board.GP7

display_bus = displayio.FourWire(
    spi, command=tft_dc, chip_select=tft_cs, reset=board.GP0
)

#  display setup
display = ST7789(display_bus, width=240, height=240, rowstart=80)

#  bitmap setup
bitmap = displayio.OnDiskBitmap(open("/parrot-240-sheet.bmp", "rb"))

# Create a TileGrid to hold the bitmap
parrot0_grid = displayio.TileGrid(bitmap, pixel_shader=displayio.ColorConverter(),
                                    width=1, height=1,
                                 tile_height=240, tile_width=240,
                                 default_tile=0,
                                 x=0, y=0)

# Create a Group to hold the TileGrid
group = displayio.Group()

# Add the TileGrid to the Group
group.append(parrot0_grid)

# Add the Group to the Display
display.show(group)

#  digital pins for the buttons
key_pins = [board.A0, board.A1, board.A2, board.A3]

#  array for buttons
keys = []

#  setup buttons as inputs
for key in key_pins:
    key_pin = digitalio.DigitalInOut(key)
    key_pin.direction = digitalio.Direction.INPUT
    key_pin.pull = digitalio.Pull.UP
    keys.append(key_pin)

p = 0 #  variable for tilegrid index
a = 0 #  variable for tile position

#  states for buttons
key0_pressed = False
key1_pressed = False
key2_pressed = False
key3_pressed = False

#  array for button states
key_states = [key0_pressed, key1_pressed, key2_pressed, key3_pressed]

while True:
    #  default tile grid position
    parrot0_grid[a] = p

    #  iterate through 4 buttons
    for i in range(4):
        inputs = keys[i]
        #  if button is pressed...
        if not inputs.value and key_states[i] is False:
            #  tile grid advances by 1 frame
            p += 1
            #  update button state
            key_states[i] = True
            #  if a midi keyboard...
            if midi_mode:
                #  send NoteOn for corresponding MIDI note
                midi.send(NoteOn(midi_notes[i], 120))
            #  if an HID keyboard...
            if keyboard_mode:
                #  send keyboard output for corresponding keycode
                #  the default includes a modifier along with the keycode
                print(ctrl)
                print(shortcuts[i])
                keyboard.send(ctrl, shortcuts[i])
                print(keyboard.send(ctrl, shortcuts[i]))
            #  if the tile grid's index is at 9...
            if p > 9:
                #  reset the index to 0
                p = 0
        #  if the button is released...
        if inputs.value and key_states[i] is True:
            #  update button state
            key_states[i] = False
            #  if a midi keyboard...
            if midi_mode:
                #  send NoteOff for corresponding MIDI note
                midi.send(NoteOff(midi_notes[i], 120))
```

### Behavior

```
224
25
Traceback (most recent call last):
  File "code.py", line 124, in <module>
  File "adafruit_hid/keyboard.py", line 120, in send
  File "adafruit_hid/keyboard.py", line 90, in press
  File "adafruit_hid/keyboard.py", line 125, in _add_keycode_to_report
TypeError: function takes 1 positional arguments but 2 were given
```

### Description

From discord user cmora111; code is from https://learn.adafruit.com/kitty-toe-bean-paw-keypad-color-tct/overview, adapted for TinyRP2040 pinout.

The backtrace error is very odd. A simpler version of the program without the display support does not cause the error, so there may be memory corruption.

cmora111, 2021-07-18:
I have tested the same code on CP 6.x and it works fine.

dhalbert, 2021-07-18:
Re-tested on a QT Py RP2040, and pruned the program a bit. I've revised the original post above.

The errors vary, so this appears to be memory corruption, and I think it's due to the `OnDiskBitmap`.

jepler, 2021-07-18:
It seems likely to be related to the palette changes I introduced in 7. Assign it to me if you like.

dhalbert, 2021-07-18:
> It seems likely to be related to the palette changes I introduced in 7. Assign it to me if you like.

You are more the expert on this, sure, thanks. I am still doing the DMA debugging.

cmora111, 2021-07-19:
What release will the fix be in?

closed 2021-07-19 by dhalbert

tannewt, 2021-07-20:
> What release will the fix be in?

It'll be in the next 7.0.0 alpha. Before that you can follow the Absolute Latest link from circuitpython.org/downloads and snag the latest build. (It takes an hour or two after merge but is then available.)

cmora111, 2021-07-20:
Thank you very much!!

PR #5017 (2021-07-19): OnDiskBitmap improvements
 * Fix out of memory write in constructor
 * Allow constructing from filename, rather than from open file object, to improve usability

Closes: #5008
Closes: #4197