adafruit/circuitpython #5016: PulseOut constructor sometimes leaves the line high

dmcomm, 2021-07-19:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.4-115-g3c25d8e2a on 2021-07-18; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import board
import pulseio
import pwmio
import time

while(True):
	pwm = pwmio.PWMOut(board.GP0, frequency=38000, duty_cycle=2**15)
	pulseOut = pulseio.PulseOut(pwm)
	time.sleep(5)
	pulseOut.deinit()
	pwm.deinit()
	time.sleep(1)
```

### Behavior

Sometimes the line is high after creating the PulseOut object. I guess the same cause as #4908 . Though sounds like some of this code might be going away in #3264 .

closed 2021-07-20 by tannewt

dmcomm, 2021-07-20:
@tannewt it's not fixed (I don't think the new code gets called by the constructor).

dhalbert, 2021-07-20:
@DavePutz Should the constructor call send() with no pulses to get things in the right state?

DavePutz, 2021-07-20:
I think the issue here is that anytime a PWMOut is created with a duty
cycle it starts outputting immediately. I don't think
that can be changed - there are library examples that depend on that
behavior. So, the best we can do is turn off the
PWM when a PulseOut is constructed, meaning there will likely be some
(unwanted) output on the pin. I'll make
that change and submit a new PR.

dmcomm, 2021-07-21:
Fixed now, thanks :)

@DavePutz I agree the brief unwanted output on the pin is expected behaviour with the current API. I asked in #3264 whether it could be removed in the new API.