adafruit/circuitpython #5020: ESP32S2 keeps resetting when in `SafeMode`

microDev1, 2021-07-19:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.4-109-g29aab69fb on 2021-07-17; microS2 with ESP32S2
```

### Code/REPL

```python
# Following is a forced way to enter safe mode.
# This also happens when safe mode is triggered automatically
# (i.e. due to crash, unhandeled exception etc.). 
import microcontroller
microcontroller.on_next_reset(microcontroller.RunMode.SAFE_MODE)
microcontroller.reset()
```

### Behavior

- The onboard led goes `quick blue pulse` -> `yellow` -> `purple` and reset.
- The `CIRCUITPY` drive keeps appearing and disappearing repeatedly and I am unable to catch hold of `REPL`.
- Happens irresepective of `TinyUF2` presence.

microDev1, 2021-08-01:
I ran a traceback, it says `StoreProhibited` and points to the following line. Looks like `out_report_buffer` is `NULL`.
 https://github.com/adafruit/circuitpython/blob/98cd989c16ab82bff10002359b09d4027a198a97/shared-module/usb_hid/Device.c#L255

dhalbert, 2021-08-01:
It looks like safe mode is not initializing the HID devices. This maybe the host trying to set the keyboard LEDs or something like that. I'll take a look.

dhalbert, 2021-08-17:
What is your host computer OS? I could not reproduce this on Ubuntu 20.04 with later-than-alpha.6, but it might be due to a delayed callback from TinyUSB.

In any case, I added code to PR #5151 to avoid a write when there is no buffer, which would prevent this.

dhalbert, 2021-08-18:
Closing via #5151. Please reopen if this shows up again.

closed 2021-08-18 by dhalbert

microDev1, 2021-08-19:
Thanks @dhalbert, this is fixed now.