adafruit/circuitpython #5023: print_exception function missing from sys module

jscholes, 2021-07-20:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.4 on 2021-07-08; Adafruit Feather nRF52840 Express with nRF52840
```

### Code/REPL

```python
import sys
sys.print_exception
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'print_exception'

### Description

We were using the `sys.print_exception` function to add tracebacks to our log entries, similar to what CPython allows.  Unfortunately this (and a whole other bunch of stuff) seems to have been removed from the `sys` module around about commit https://github.com/adafruit/circuitpython/commit/b35fa44c8adeae93000ca9bc413ea6e64456fd94.

`print_exception` is included according to the [latest MicroPython docs](https://docs.micropython.org/en/latest/library/usys.html), although their module is called `usys` and not just `sys`.  I did try to import `usys` instead:

```
>>> import usys                                                                 
Traceback (most recent call last):                                              
  File "<stdin>", line 1, in <module>                                           
ImportError: no module named 'usys'                                             
```

I also tried the [traceback module from micropython-lib](https://github.com/micropython/micropython-lib/blob/master/python-stdlib/traceback/traceback.py), although that just calls `sys.print_exception` anyway.  It also uses `sys.exc_info` in its `print_exc` function, which is also not present in CircuitPython.

If these facilities have been removed intensionally, is there some other recommended way in CircuitPython to capture a traceback for reuse?  Alternatively, if they are still present, where can they be located?

anecdata, 2021-07-20:
duplicate of https://github.com/adafruit/circuitpython/issues/4748 I bvelieve

jscholes, 2021-07-20:
@anecdata Thanks, and apologies for the duplicate.  I am curious how merging in a MicroPython update removed so many things from the `sys` module, despite them still being present in the `usys` MicroPython module?

tannewt, 2021-07-20:
> @anecdata Thanks, and apologies for the duplicate. I am curious how merging in a MicroPython update removed so many things from the `sys` module, despite them still being present in the `usys` MicroPython module?

I removed `print_exception` during the merge because it was not CPython compatible. We hadn't validated `sys` previously. What else did I remove?

jscholes, 2021-07-20:
@tannewt I think I was reading things wrong in terms of how much was removed from the `sys` module.  As a minimum, `print_exception` and `atexit` seem to no longer be there, but are still present in MicroPython.  Additionally (and I'm not overly clear on this one), `exc_info` is used by some MicroPython modules, but not documented under `usys`.  So I'm not sure if it actually exists or not.

> I removed print_exception during the merge because it was not CPython compatible.

I assume `atexit` was removed for the same reason, because in CPython the same functionality is fulfilled by the dedicated `atexit` module?  I understand the desire for compatibility, but it seems problematic to remove working functionality with no replacement.

microDev1, 2021-07-20:
`exc_info` is disabled by default in both `CPY` & `MPY`.
https://github.com/adafruit/circuitpython/blob/ce7301527aa94b3a4f17dbb02f4952ff6fec8ce7/py/mpconfig.h#L1303-L1307

tannewt, 2021-07-20:
> I assume `atexit` was removed for the same reason, because in CPython the same functionality is fulfilled by the dedicated `atexit` module? I understand the desire for compatibility, but it seems [...]

Correct. It should be relatively easy to move to a dedicated and compatible `atexit` native module. I don't think it's problematic because all of the source code is available to anyone who wants to add it in a compatible way.

Merging it as-is would potentially lead to new code that doesn't work in CPython with Blinka. Supporting that incompatibility is a burden I'm not willing to assume to have the functionality. Supporting APIs has a bigger impact than simply the API boundary itself. API choices cascade into libraries and example code and are harder to change once they've been supported. I'd rather get it right up front than have to go back and find all the places to change.

microDev1, 2021-07-29:
Converting this to a feature request for CPython `atexit` module. We already have an issue open for `print_exception()` (#4748).

closed 2021-08-23 by microDev1