adafruit/circuitpython #5033: Failed to make a CircuitPython build with ULAB_MAX_DIMS set to 4 in ulab.h

nikita-kotsehub, 2021-07-21:
### CircuitPython version

```python
CircuitPython 7.0.0 Alpha 4 (not sure, but I cloned it from https://github.com/adafruit/circuitpython.git)
```

### Code/REPL

```python
git clone https://github.com/adafruit/circuitpython.git
cd circuitpython
git submodule sync --quiet --recursive
git submodule update --init

# Install pip if it is not already installed (Linux only)
sudo apt install python3-pip

# Install needed Python packages from pypi.org.
pip3 install -r requirements-dev.txt

# Force a particular version of click. It does not get updated to the
# proper version sometimes
pip3 install --upgrade click==7.1.2

# Changing ULAB_MAX_DIMS to 4 in extmod/ulab/code/ulab.h

make -C mpy-cross

cd ports/raspberrypi
make BOARD=arduino_nano_rp2040_connect
```

### Behavior

../../extmod/ulab/code/numpy/numerical/numerical.c:284:72: error: macro "RUN_SUM" requires 7 arguments, but only 5 given
                 RUN_SUM(uint8_t, array, results, rarray, _shape_strides);
                                                                        ^
../../extmod/ulab/code/numpy/numerical/numerical.c:286:71: error: macro "RUN_SUM" requires 7 arguments, but only 5 given
                 RUN_SUM(int8_t, array, results, rarray, _shape_strides);
                                                                       ^
../../extmod/ulab/code/numpy/numerical/numerical.c:288:73: error: macro "RUN_SUM" requires 7 arguments, but only 5 given
                 RUN_SUM(uint16_t, array, results, rarray, _shape_strides);
                                                                         ^
../../extmod/ulab/code/numpy/numerical/numerical.c:290:72: error: macro "RUN_SUM" requires 7 arguments, but only 5 given
                 RUN_SUM(int16_t, array, results, rarray, _shape_strides);
                                                                        ^
Command '['arm-none-eabi-gcc', '-E', '-DNO_QSTR', '-Ibuild-arduino_nano_rp2040_connect/tmp', '-DCIRCUITPY_FULL_BUILD=1', '-DMICROPY_PY_ASYNC_AWAIT=1', '-DCIRCUITPY_AESIO=0', '-DCIRCUITPY_ALARM=1', '-DCIRCUITPY_ANALOGIO=1', '-DCIRCUITPY_AUDIOBUSIO=1', '-DCIRCUITPY_AUDIOBUSIO_I2SOUT=1', '-DCIRCUITPY_AUDIOBUSIO_PDMIN=1', '-DCIRCUITPY_AUDIOIO=0', '-DCIRCUITPY_AUDIOPWMIO=1', '-DCIRCUITPY_AUDIOCORE=1', '-DCIRCUITPY_AUDIOMIXER=1', '-DCIRCUITPY_AUDIOMP3=1', '-DCIRCUITPY_BINASCII=1', '-DCIRCUITPY_BITBANG_APA102=0', '-DCIRCUITPY_BITBANGIO=1', '-DCIRCUITPY_BITOPS=1', '-DCIRCUITPY_BLEIO_HCI=1', '-DCIRCUITPY_BLEIO=1', '-DCIRCUITPY_BLE_FILE_SERVICE=0', '-DCIRCUITPY_BOARD=1', '-DCIRCUITPY_BUSDEVICE=1', '-DCIRCUITPY_BUILTINS_POW3=1', '-DCIRCUITPY_BUSIO=1', '-DCIRCUITPY_BUSIO_SPI=1', '-DCIRCUITPY_CAMERA=0', '-DCIRCUITPY_CANIO=0', '-DCIRCUITPY_DIGITALIO=1', '-DCIRCUITPY_COMPUTED_GOTO_SAVE_SPACE=0', '-DCIRCUITPY_CONSOLE_UART=0', '-DCIRCUITPY_COUNTIO=1', '-DCIRCUITPY_DISPLAYIO=1', '-DCIRCUITPY_BITMAPTOOLS=1', '-DCIRCUITPY_FRAMEBUFFERIO=1', '-DCIRCUITPY_VECTORIO=1', '-DCIRCUITPY_DUALBANK=0', '-DCIRCUITPY_ENABLE_MPY_NATIVE=0', '-DCIRCUITPY_ERRNO=1', '-DCIRCUITPY_ESPIDF=0', '-DCIRCUITPY__EVE=1', '-DCIRCUITPY_FREQUENCYIO=0', '-DCIRCUITPY_GAMEPAD=0', '-DCIRCUITPY_GAMEPADSHIFT=0', '-DCIRCUITPY_GNSS=0', '-DCIRCUITPY_I2CPERIPHERAL=0', '-DCIRCUITPY_IPADDRESS=0', '-DCIRCUITPY_JSON=1', '-DCIRCUITPY_KEYPAD=1', '-DCIRCUITPY_MATH=1', '-DCIRCUITPY_MEMORYMONITOR=0', '-DCIRCUITPY_MICROCONTROLLER=1', '-DCIRCUITPY_MSGPACK=1', '-DCIRCUITPY_NEOPIXEL_WRITE=1', '-DCIRCUITPY_NETWORK=0', '-DCIRCUITPY_NVM=1', '-DCIRCUITPY_OS=1', '-DCIRCUITPY_IMAGECAPTURE=1', '-DCIRCUITPY_PEW=0', '-DCIRCUITPY_PIXELBUF=1', '-DCIRCUITPY_PS2IO=0', '-DCIRCUITPY_PULSEIO=1', '-DCIRCUITPY_PWMIO=1', '-DCIRCUITPY_RAINBOWIO=1', '-DCIRCUITPY_RANDOM=1', '-DCIRCUITPY_RE=1', '-DCIRCUITPY_REQUIRE_I2C_PULLUPS=1', '-DCIRCUITPY_RP2PIO=1', '-DCIRCUITPY_RGBMATRIX=1', '-DCIRCUITPY_ROTARYIO=1', '-DCIRCUITPY_ROTARYIO_SOFTENCODER=1', '-DCIRCUITPY_RTC=1', '-DCIRCUITPY_SAMD=0', '-DCIRCUITPY_SDCARDIO=1', '-DCIRCUITPY_SDIOIO=0', '-DCIRCUITPY_SERIAL_BLE=0', '-DCIRCUITPY_SHARPDISPLAY=1', '-DCIRCUITPY_SOCKETPOOL=0', '-DCIRCUITPY_SSL=0', '-DCIRCUITPY_STAGE=0', '-DCIRCUITPY_STORAGE=1', '-DCIRCUITPY_STRUCT=1', '-DCIRCUITPY_SUPERVISOR=1', '-DCIRCUITPY_SYNTHIO=1', '-DCIRCUITPY_TERMINALIO=1', '-DCIRCUITPY_TIME=1', '-DCIRCUITPY_TOUCHIO_USE_NATIVE=0', '-DCIRCUITPY_TOUCHIO=1', '-DCIRCUITPY_UHEAP=0', '-DCIRCUITPY_USB=1', '-DUSB_NUM_IN_ENDPOINTS=8', '-DUSB_NUM_OUT_ENDPOINTS=8', '-DCIRCUITPY_USB_CDC=1', '-DCIRCUITPY_USB_CDC_CONSOLE_ENABLED_DEFAULT=1', '-DCIRCUITPY_USB_CDC_DATA_ENABLED_DEFAULT=0', '-DCIRCUITPY_USB_HID=1', '-DCIRCUITPY_USB_HID_ENABLED_DEFAULT=1', '-DCIRCUITPY_USB_MIDI=1', '-DCIRCUITPY_USB_MIDI_ENABLED_DEFAULT=1', '-DCIRCUITPY_USB_MSC=1', '-DCIRCUITPY_USB_MSC_ENABLED_DEFAULT=1', '-DCIRCUITPY_USB_VENDOR=0', '-DUSB_NUM_ENDPOINT_PAIRS=8', '-DCIRCUITPY_USTACK=0', '-DCIRCUITPY_ULAB=1', '-DCIRCUITPY_WATCHDOG=1', '-DCIRCUITPY_WIFI=0', '-DINTERNAL_FLASH_FILESYSTEM=1', '-DQSPI_FLASH_FILESYSTEM=0', '-DSPI_FLASH_FILESYSTEM=0', '-DCIRCUITPY_CREATOR_ID=0x239A', '-DCIRCUITPY_CREATION_ID=0x00CF', '-DUSB_INTERFACE_NAME="CircuitPython"', '-DUSB_VID=0x239A', '-DUSB_PID=0x00CF', '-DUSB_MANUFACTURER="Arduino"', '-DUSB_MANUFACTURER_8="Arduino"', '-DUSB_PRODUCT_16="Arduino Nano RP2"', '-DUSB_PRODUCT="Arduino Nano RP2040 Connect"', '-DCFG_TUD_CDC=2', '-DUSB_HIGHSPEED=0', '-DCIRCUITPY_DEBUG=0', '-DLONGINT_IMPL_MPZ', '-DRASPBERRYPI', '-DPICO_ON_DEVICE=1', '-DPICO_NO_BINARY_INFO=0', '-DPICO_TIME_DEFAULT_ALARM_POOL_DISABLED=0', '-DPICO_DIVIDER_CALL_IDIV0=0', '-DPICO_DIVIDER_CALL_LDIV0=0', '-DPICO_DIVIDER_HARDWARE=1', '-DPICO_DOUBLE_ROM=1', '-DPICO_FLOAT_ROM=1', '-DPICO_MULTICORE=1', '-DPICO_BITS_IN_RAM=0', '-DPICO_DIVIDER_IN_RAM=0', '-DPICO_DOUBLE_PROPAGATE_NANS=0', '-DPICO_DOUBLE_IN_RAM=0', '-DPICO_MEM_IN_RAM=0', '-DPICO_FLOAT_IN_RAM=0', '-DPICO_FLOAT_PROPAGATE_NANS=1', '-DPICO_NO_FLASH=0', '-DPICO_COPY_TO_RAM=0', '-DPICO_DISABLE_SHARED_IRQ_HANDLERS=0', '-DPICO_NO_BI_BOOTSEL_VIA_DOUBLE_RESET=0', '-DTUD_OPT_RP2040_USB_DEVICE_ENUMERATION_FIX=1', '-DCFG_TUSB_MCU=OPT_MCU_RP2040', '-DCFG_TUD_MIDI_RX_BUFSIZE=128', '-DCFG_TUD_CDC_RX_BUFSIZE=256', '-DCFG_TUD_MIDI_TX_BUFSIZE=128', '-DCFG_TUD_CDC_TX_BUFSIZE=256', '-DCFG_TUD_MSC_BUFSIZE=1024', '-O3', '-DNDEBUG', '-I../../devices/ble_hci', '-I.', '-I../..', '-I../lib/mp-readline', '-I../lib/timeutils', '-Iboards/arduino_nano_rp2040_connect', '-Iboards/', '-isystem', 'sdk/', '-isystem', 'sdk/src/common/pico_base/include/', '-isystem', 'sdk/src/common/pico_binary_info/include/', '-isystem', 'sdk/src/common/pico_stdlib/include/', '-isystem', 'sdk/src/common/pico_sync/include/', '-isystem', 'sdk/src/common/pico_time/include/', '-isystem', 'sdk/src/common/pico_util/include/', '-isystem', 'sdk/src/rp2040/hardware_regs/include/', '-isystem', 'sdk/src/rp2040/hardware_structs/include/', '-isystem', 'sdk/src/rp2_common/hardware_adc/include/', '-isystem', 'sdk/src/rp2_common/hardware_base/include/', '-isystem', 'sdk/src/rp2_common/hardware_claim/include/', '-isystem', 'sdk/src/rp2_common/hardware_clocks/include/', '-isystem', 'sdk/src/rp2_common/hardware_divider/include/', '-isystem', 'sdk/src/rp2_common/hardware_dma/include/', '-isystem', 'sdk/src/rp2_common/hardware_flash/include/', '-isystem', 'sdk/src/rp2_common/hardware_gpio/include/', '-isystem', 'sdk/src/rp2_common/hardware_irq/include/', '-isystem', 'sdk/src/rp2_common/hardware_i2c/include/', '-isystem', 'sdk/src/rp2_common/hardware_pio/include/', '-isystem', 'sdk/src/rp2_common/hardware_pll/include/', '-isystem', 'sdk/src/rp2_common/hardware_resets/include/', '-isystem', 'sdk/src/rp2_common/hardware_rtc/include/', '-isystem', 'sdk/src/rp2_common/hardware_spi/include/', '-isystem', 'sdk/src/rp2_common/hardware_sync/include/', '-isystem', 'sdk/src/rp2_common/hardware_timer/include/', '-isystem', 'sdk/src/rp2_common/hardware_uart/include/', '-isystem', 'sdk/src/rp2_common/hardware_watchdog/include/', '-isystem', 'sdk/src/rp2_common/hardware_xosc/include/', '-isystem', 'sdk/src/rp2_common/pico_multicore/include/', '-isystem', 'sdk/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/include/', '-isystem', 'sdk/src/rp2_common/pico_stdio/include/', '-isystem', 'sdk/src/rp2_common/pico_printf/include/', '-isystem', 'sdk/src/rp2_common/pico_float/include/', '-isystem', 'sdk/src/rp2_common/pico_platform/include/', '-isystem', 'sdk/src/rp2_common/pico_runtime/printf/include/', '-isystem', 'sdk/src/rp2_common/pico_bootrom/include/', '-isystem', 'sdk/src/rp2_common/pico_unique_id/include/', '-Isdk_config', '-I../../lib/tinyusb/src', '-I../../supervisor/shar [... cut ...]
make: *** [../../py/mkrules.mk:74: build-arduino_nano_rp2040_connect/extmod/ulab/code/numpy/numerical/numerical.o] Error 1

### Description

I tried making a build where ULAB_MAX_DIMS is 4 to support four dimensions. However, I get the errors above when trying to make a build. I could easily build the stock build (no changes), and also after changing ULAB_MAX_DIMS to 3. Hence, I think the issue is with the 4th dimension, although, according to the comments in ulab.h,  "Possible values lie between 1, and 4, inclusive."

jepler, 2021-07-22:
@v923z can you take a look?

v923z, 2021-07-22:
> @v923z can you take a look?

I'm on it.

jepler, 2021-07-22:
We could potentially add all the settings to the CI over in micropython-ulab once this is sorted, let me know if you have an interest in me doing that.

v923z, 2021-07-22:
Do you mean that the dimensions could be checked out, one by one? I think that wouldn't be a bad idea. I would have to re-write the test scripts, though. I think, at the moment they are really geared for 2D.

v923z, 2021-07-22:
@jepler @nikita-kotsehub I think I have found the error. It is fixed in https://github.com/v923z/micropython-ulab/pull/428. New release coming in a couple of minutes.

PR #5028 (2021-07-22): ulab: move documentation to the right location
Update ulab to 3.3.1.  This brings in some bugfixes and most importantly moves all the documentation items (as far as I could tell) to the right places.

Originally I thought this was going to be complicated due to the need to rename `ulab.ndarray` to `ulab.numpy.ndarray` with changes to both repos in parallel, but ultimately I found a way to do it compatibly, with a second step in ulab to remove an alias of the `ndarray` type that can be done after this merge—but nobody spends time with their CI in the red.

~~This ends up being complicated, beca [... cut]
closed 2021-07-22 by tannewt