adafruit/circuitpython #5041: Raspberry Pi Pico should have default SPI, I2C, UART pins

shuyuan-liu, 2021-07-23:
Hi! I'm using CircuitPython 7.0.0-alpha.5-8-g779b5c185 on a Raspberry Pi Pico.

Despite [the Pico's datasheet](https://datasheets.raspberrypi.org/pico/pico_datasheet.pdf) labelling GP16-19 as default for SPI, GP4&5 as default for I2C, and GP0&1 as default for UART, there is no `board.SPI()`, `board.I2C()`, or `board.UART()` available.

The pinout diagram on [Getting Started with Raspberry Pi Pico and CircuitPython](https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/pinouts) seems outdated as it only includes the default pins for UART but not SPI or I2C. The page also states that "The Pico does not label specific pins as the defaults to use for I2C, SPI, or UART connections," whereas default pins for those functions *are* marked in the datasheet.

dhalbert, 2021-07-23:
We discussed this in #4121 and made a decision not to do this, for the reasons given there.

tannewt, 2021-07-23:
We should revisit this. As @shuyuan-liu points out, the Pico datasheet designates defaults now:

[image]

dhalbert, 2021-07-23:
The datasheet, the SDK doc, and MicroPython all use somewhat different "default" pins. Our conclusion before was that since it's not obvious which is the most authoritative for CircuitPython, and there are no silkscreen markings, that this would be a continuing support issue. I'm not sure that situation has changed.

tannewt, 2021-08-04:
Ok, closing this because the board itself doesn't label the default busses.

dhalbert, 2026-11-01:
I would like to revisit this despite closing this and #4121.

1. The pinout picture as supplied by Raspberry Pi above does mark defaults, even though they are ascetic when it comes to board markings.
2. GPIO0/GPIO1 seems very common for UART.
3. The Adafruit Cowbell doubler uses GPIO4/5 for STEMMA/QT I2C, same as the default in the pinout picture.
4. We spend too much time explaining why we don't have defaults.
5. It makes code less portable across boards.
6. Zephyr also defines some defaults similar or the same as above.

So I propose adding the defaults above, including `board.I2C()` and `board.STEMMA_I2C()` as the same thing using 4/5.

tannewt, 2026-11-02:
Fine with me. They've evolved a bunch since they released.

closed 2026-11-07 by dhalbert