adafruit/circuitpython #5043: Maxlen of Bluetooth name? Too long names result in a crash during start_advertising

ThomasAtBBTF, 2021-07-23:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.4 on 2021-07-08; Adafruit ItsyBitsy nRF52840 Express with nRF52840
```

### Code/REPL

```python
# boot.py
usb_cdc.enable(console=True, data=False)
usb_midi.disable()

#code.py
...

devicename = "123456789A123456789B123457"
# this are 27 characters (no error in advertisement with names which are 26 characters long)

...

_bleio.adapter.name = devicename  # I don't know if the device name here or bleow is neccessary!
ble = adafruit_ble.BLERadio()
ble.name = devicename # I don't know if the device name here or above is necessary!
...
    print("Start advertizing:", self.service)
...
if self.isconnected:
  ...
else:
  print("unconnected", self.ble, self.ble.name, self.ble.connections)
...

...
ble.start_advertising(self.advertisement)
# In that call the Error below is thrown.

Repl:
unconnected <BLERadio object at 2001c8a0> 123456789A123456789B1234567 ()
Start advertizing: HID
Traceback (most recent call last):
  File "code.py", line 336, in <module>
  File "code.py", line 174, in check
  File "/lib/adafruit_ble/__init__.py", line 195, in start_advertising
ValueError: Invalid BLE parameter

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

The next reload will often crash the board (the serial port over USB is disconnected)
And only reset(I think) or removing and reapplying USB-Power reboots the board.
```

### Behavior

When I use device names with 27 characters or more a Runtime Error is thrown during Advertisement.
[image]

In this picture, you see a crash with devicename 30 characters long.,

### Description

Error when allowing user-specific DeviceNames.
(I will limit them to 26 Characters in the code)user-specific

### Additional information

During narrowing down this issue I found another strange bug which I will file later.

ThomasAtBBTF, 2021-07-23:
My suggestion would be, either to quietly slice the name in the setter to the max allowable length or raise an error in the setter and not let the error come up in start_advertizing.

tannewt, 2021-07-23:
I like the idea of validating the name when it is set. That is clearest for the setter on how to fix it.

ThomasAtBBTF, 2021-07-23:
Fine with me, but is 26 characters really the threshold?
Is the length of other parameters in the report not changing the threshold?
I read somewhere that there is a length-limit of 31 bytes for a report.
But I don't understand this in enough detail to make a statement here.

tannewt, 2021-07-26:
I'm not sure where the limit is being imposed. Figuring that out is part of the task here.

31 bytes is the limit for legacy advertisements IIRC.

tannewt, 2021-07-27:
What do you mean by Bluetooth Beacon? Do you have an example product that does it?

BLE 5 does have extended advertisements that would support up to ~512 bytes IIRC.

ThomasAtBBTF, 2021-07-28:
Bluetooth beacons are used for indoor navigation for blind and visually impaired persons. 
Here are articles describing the functionality:
https://www.lighthouseguild.org/vision-health/a-hybrid-indoor-positioning-system-for-the-blind-and-visually-impaired-using-bluetooth-and-google-tango/

https://www.irjet.net/archives/V7/i4/IRJET-V7I4609.pdf

tannewt, 2021-07-29:
How are you getting the name of a BLE beacon? My impression is that they may simply advertise an ID that an app can lookup for a name. `NRF Connect` will list the contents of advertisements around you.

dhalbert, 2021-12-09:
The defined maximum length of the name is 248 bytes, but whether you can use that depends on the advertising capabilities of the peripheral. The actual length that fits depends on what else is in the advertising packet (e.g. various UUIDs), including in the extended advertisement. So I don't think we can validate the length when `.name` is set. We could possibly throw a better error instead of `Invalid BLE parameter`.

The nRF SD adds the default name to the advertisement by itself, so it's not even that we are assembling the advertisement and it's too long. We'd have to keep track of the default name length and also see that there is not another name given.

I am going to move this to Long Term, because I don't see a short-term fix, unfortunately.