adafruit/circuitpython #5044: Hard Crash after multiple reloads.

ThomasAtBBTF, 2021-07-23:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.4 on 2021-07-08; Adafruit ItsyBitsy nRF52840 Express with nRF52840
```

### Code/REPL

```python
print("first line.")
s = input("first input\n")
```

### Behavior

The code above runs fine as expected!
But when you press CTRL-D at the input several times,
there will be a hard crash of the board with no output to repl. (The serial USB-Device "goes-away") 
Similar to the crash reported in #5043 
Note that once the board with the nRF-52840 is powered several "unknown" Bluetooth devices are advertised.
I would like a possibility (in boot.py ?) to disable Bluetooth until the Application in code.py is ready and initialized.
  

### Additional information

Actually also this code:
print("first line.")
Will after several ctrl-D's in fast sequence hard crash the board!

ThomasAtBBTF, 2021-07-23:
And it seems, that not only ciruitpython and the USB-stack stops working
also, the usual "unknown Device" advertisements of the nRF52840 are not sent anymore.

jepler, 2021-07-23:
This is probably the same as #5018.

closed 2021-07-23 by jepler

ThomasAtBBTF, 2021-07-23:
I still would like to be able to actively start Bluetooth in code.py when turned off in boot.py

jepler, 2021-07-23:
@ThomasAtBBTF Thanks! I overlooked this element of the issue because I read it too quickly.

It would be great if you enter a fresh feature request for these items (maybe just one, as they do seem closely related/interdependent). I do agree that we have room to refine how the bluetooth "flow" goes and how to control it from code, though I only know the basics and haven't used it much for myself.

> I still would like to be able to actively start Bluetooth in code.py when turned off in boot.py

>  I would like a possibility (in boot.py ?) to disable Bluetooth until the Application in code.py is ready and initialized.

ThomasAtBBTF, 2021-07-23:
Will do so.
Thank you!
And: My gut feeling tells me, that the "Hit ctrl-d repeatedly bug" is related to communication with the Nordic API when bringing up CircuitPython after a reload or reset.
The advertisement of the "stock-devices" needs somehow be stopped "nicely" and CP needs to take over.