adafruit/circuitpython #5047: audiobusio.I2SOut: audio is initially "crackly"

PaintYourDragon, 2021-07-23:
### CircuitPython version

```python
6.3, 7.0 alpha, tested on both Feather M4 and Raspberry Pi Pico
```

### Code/REPL

```python
import board
import audiobusio
import audiocore

FILE = 'sweep.wav'

# Feather M4
BIT_CLOCK_PIN = board.TX
LR_CLOCK_PIN = board.D10
DATA_PIN = board.D11
OUTPUT = audiobusio.I2SOut(BIT_CLOCK_PIN, LR_CLOCK_PIN, DATA_PIN)

DATA = open(FILE, 'rb')
AUDIO = audiocore.WaveFile(DATA)

OUTPUT.play(AUDIO, loop=True)
while True:
    pass
```

### Behavior

No error, but sound output is initially “crackly.” This eventually clears up, but the timing of this seems random: could be halfway through the first play, could be after several loops through. Restarting the code (either through reset or touching code.py) causes the crackle to manifest anew.

Using single-channel 3W I2S amp:
https://www.adafruit.com/product/3006
Minimal wiring of power, L/R, data and clock. Gain and SD are left floating, which documentation suggests is OK (and tying them various ways didn’t change results anyway).

### Description

Mentioning this here as there was some talk of audio being an RP2040 problem … but I’m experiencing this on Feather M4 as well. Maybe RP2040 is a red herring and it’s I2S in general?

### Additional information

sweep.wav used in the example can be downloaded here:
https://www.dropbox.com/s/81t4ibplpd1799j/sweep.wav?dl=0

tannewt, 2021-07-23:
@dhalbert is looking into this. What is the sample rate of the wav?

PaintYourDragon, 2021-07-23:
16 KHz 16-bit mono

dhalbert, 2021-08-12:
This should be fixed by #5079 on RP2040 but may still exist on SAMD51, and I will be looking at that.

closed 2021-08-23 by tannewt

nathanielbutts, 2021-08-29:
I'm still having this issue on Pico RP2040, using 7.0.0-beta.0.  There is a difference in the crackling between beta.0 and alpha.6.  On alpha.6 it was the same crackling for the entire length of play back.  On beta.0 it cycles, kinda like:
 `_|¯|_|¯|_|¯|______|¯|_|¯|_|¯|______|¯|_|¯|_|¯|_`

Code I am using:
`import board
import audiocore
import audiomp3
import audiobusio
import digitalio
import time

f = open("/music/15-03.mp3", "rb")
mp3 = audiomp3.MP3Decoder(f)
a = audiobusio.I2SOut(board.GP27, board.GP28, board.GP26)

print("playing")
a.play(mp3)
while a.playing:
  time.sleep(.001)
print("stopped")`

Hardware is Raspberry Pico and [Waveshare Pico-Audio Hat](https://www.waveshare.com/wiki/Pico-Audio)

nathanielbutts, 2021-08-29:
> I'm still having this issue on Pico RP2040, using 7.0.0-beta.0. There is a difference in the crackling between beta.0 and alpha.6. On alpha.6 it was the same crackling for the entire length of play back. On beta.0 it cycles, kinda like:
> `_|¯|_|¯|_|¯|______|¯|_|¯|_|¯|______|¯|_|¯|_|¯|_`
> 
> Code I am using:
> 
> ```
> import board
> import audiocore
> import audiomp3
> import audiobusio
> import digitalio
> import time
> 
> 
> f = open("/music/15-03.mp3", "rb")
> mp3 = audiomp3.MP3Decoder(f)
> a = audiobusio.I2SOut(board.GP27, board.GP28, board.GP26)
> 
> print("playing")
> a.play(mp3)
> while a.playing:
>   time.sleep(.001)
> print("stopped")
> ```
> 
> Hardware is Raspberry Pico and [Waveshare Pico-Audio Hat](https://www.waveshare.com/wiki/Pico-Audio)
> 
> _edit_ I have a audio amp breakout board here somewhere. If I get the time I'll try with this board instead of the hat.

Same issue using my MAX98357 Breakout board.  Thought it might be a related to being a 2 channel file, but converting to one channel didn't make a change.  Modified my script to run audiocore.WaveFile and still using the I2S output, and it works as expected.

dhalbert, 2021-08-29:
@nathanielbutts Could you open a new issue, specific to MP3 (it sounds like it's only MP3?)

nathanielbutts, 2021-08-30:
> @nathanielbutts Could you open a new issue, specific to MP3 (it sounds like it's only MP3?)

Absolutely.  Opened [Issue #5226](https://github.com/adafruit/circuitpython/issues/5266)