adafruit/circuitpython #5049: Adding the ability to control the state and behavior of the Bluetooth (sub-)system of the nRF52 chips in boot.py

ThomasAtBBTF, 2021-07-23:
While working on code for HID devices on the nRF52840 ItsyBitsy I am observing that as soon as power is applied to the board, it starts advertising (unknown to Windows) devices.
Only after code.py starts and executes code to present an HID device these "unknown devices" are "going away" and the device prepared in code.py gets advertised.
I think it would be helpful and allow a cleaner customer experience if the behavior of the default Bluetooth (sub-)system can be modified in boot.py as nowadays USB devices can be configured there.

If these, currently unknown to me devices, are necessary or helpful for OverTheAir updates or other useful things for customers, it would be good to add a API to allow the possibility in code.py to "to turn on" the original functionality in a user product.

ThomasAtBBTF, 2021-07-27:
Yes, I am talking about BLE HID devices.
And no to the second part of your question:
I would like to disable Bluetooth altogether in boot.py until CircuitPython started up and inside code.py all is ready to advertise the HID device which is implemented in code.py

ThomasAtBBTF, 2021-08-03:
How does #5089 fix this problem?
Or: can I "reset the chip" without rebooting CircuitPython?
Maybe we need the possibility to call: supervisor.enable_ble_workflow() which can be called in code.py,  
after supervisor.disable_ble_workflow() was called in boot.py

tannewt, 2021-08-03:
What are you trying to do overall? It would be ok to have it on and advertise until user code advertises something else. If a device connects before the user code, then the user code can create the HIDService and the central will pick it up.

#5089 does the minimal API to turn it all off. I don't really want to add more complicated APIs until we know what we want to be able to do.

tannewt, 2021-08-03:
If/when we add that detail, I'd put it in `_bleio` because the detailed control doesn't matter when user code can't control ble. The current disable is meant to be a broad shutoff.

ThomasAtBBTF, 2021-08-05:
Overall, I am trying to use a nRF52 board with CircuitPyton.
And when powering on the board, I want the Bluetooth transmissions of the board to be off / disabled.
Until the python code in code.py is ready to advertise the intended features of the product.

ThomasAtBBTF, 2021-08-05:
I just do not see what kind of services the board shall advertise while no code is running.

closed 2021-08-05 by tannewt

tannewt, 2021-08-05:
> Overall, I am trying to use a nRF52 board with CircuitPyton.
> And when powering on the board, I want the Bluetooth radio / transmissions of the board to be off / disabled.
> Until the python code in code.py is ready to advertise the intended features of the project / product.

`supervisor.disable_ble_workflow()` will allow for this. It turns off background advertising and creation of the file transfer service and the nordic uart service for circuitpython serial connectivity. You can still advertise from user code.

> I just do not see what kind of services the board shall advertise while no code is running.

The background advertisements will advertise the file transfer service (0xfebb) and the nordic uart service in the scan response.

ThomasAtBBTF, 2021-08-05:
Ok, great!
I will try this out first thing tomorrow morning.
I think, I just did not understand what "disable_ble_workflow" would actually do. 
I thought, that it will stop all Bluetooth activities until reset.

PR #5089 (2021-08-05): Add ability to disable BLE workflow
Call `supervisor.disable_ble_workflow()` and the BLE workflow will
be disabled until the chip is reset.

This also includes a couple fixes:
1. Terminals can now be deinit by setting the tilegrid to NULL. This
  prevents using the tilegrid before display is init.
2. Fix BLE serial send amount when sending more than a single packet.

Fixes #5049
commit 1e225610 (2021-08-05): Add ability to disable BLE workflow

Call `supervisor.disable_ble_workflow()` and the BLE workflow will
be disabled until the chip is reset.

This also includes a couple fixes:
1. Terminals can now be deinit by setting the tilegrid to NULL. This
  prevents using the tilegrid before display is init.
2. Fix BLE serial send amount when sending more than a single packet.

Fixes #5049