adafruit/circuitpython #5057: “All timers in use” from AudioOut() after 4 soft reloads on PewPew M4

cwalther, 2021-07-25:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.5-42-g0a3c75c43 on 2021-07-25; PewPew M4 with samd51g19
```

### Code/REPL

```python
import audioio, board; a = audioio.AudioOut(board.SPEAKER)
```

### Behavior

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.5-42-g0a3c75c43 on 2021-07-25; PewPew M4 with samd51g19
>>> import audioio, board; a = audioio.AudioOut(board.SPEAKER)
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.5-42-g0a3c75c43 on 2021-07-25; PewPew M4 with samd51g19
>>> import audioio, board; a = audioio.AudioOut(board.SPEAKER)
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.5-42-g0a3c75c43 on 2021-07-25; PewPew M4 with samd51g19
>>> import audioio, board; a = audioio.AudioOut(board.SPEAKER)
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.5-42-g0a3c75c43 on 2021-07-25; PewPew M4 with samd51g19
>>> import audioio, board; a = audioio.AudioOut(board.SPEAKER)
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.5-42-g0a3c75c43 on 2021-07-25; PewPew M4 with samd51g19
>>> import audioio, board; a = audioio.AudioOut(board.SPEAKER)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>>
```

### Description

Creating an `audioio.AudioOut` instance, then soft-reloading using `^D` works four times, the fifth time it raises a `RuntimeError: All timers in use`. Apparently something is not properly being reset in the soft reload.

This appears to be specific to the PewPew M4, maybe related to its display initialization. I cannot reproduce it on an ItsyBitsy M4 Express using the same firmware version. (It can however be reproduced by installing the PewPew M4 firmware on the ItsyBitsy, for those who don’t have a PewPew M4.)

cwalther, 2021-07-25:
`git bisect` points to 1a076d34c9d197c1e1420117f856a5d9aa2e8581 as the commit that introduced the bug. Probably the change of included modules, let me check if I can narrow it down further.

cwalther, 2021-07-25:
It’s the disabling of `CIRCUITPY_PWMIO` that causes the issue.

It looks like [`common_hal_audioio_audioout_construct()`](https://github.com/adafruit/circuitpython/blob/0a3c75c43d61f8a988e31a87b40450ec772d0c31/ports/atmel-samd/common-hal/audioio/AudioOut.c#L234) sets up a timer but [`audioout_reset()`](https://github.com/adafruit/circuitpython/blob/0a3c75c43d61f8a988e31a87b40450ec772d0c31/ports/atmel-samd/common-hal/audioio/AudioOut.c#L100) doesn’t reset it and instead relies on [`pwmout_reset()`](https://github.com/adafruit/circuitpython/blob/0a3c75c43d61f8a988e31a87b40450ec772d0c31/ports/atmel-samd/common-hal/pwmio/PWMOut.c#L92) to do it? Or is someone supposed to call [`common_hal_audioio_audioout_deinit()`](https://github.com/adafruit/circuitpython/blob/0a3c75c43d61f8a988e31a87b40450ec772d0c31/ports/atmel-samd/common-hal/audioio/AudioOut.c#L340) on a soft reset and that is not happening?

That’s all the investigation I’m going to do for now. I’m guessing it should be reproducible on any SAMD board with `CIRCUITPY_AUDIOIO = 1`and `CIRCUITPY_PWMIO = 0`.

microDev1, 2021-07-25:
> Or is someone supposed to call common_hal_audioio_audioout_deinit() on a soft reset and that is not happening?

The `ctrl+d` routine calls [`reset_port()`](https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/supervisor/port.c#L327) which calls `module_reset()` functions, module `module_deinit()` functions are not called.

deshipu, 2021-07-26:
I could work around this by enabling pwmio and disabling analogio instead.

cwalther, 2021-07-26:
In my testing, just enabling pwmio without disabling anything worked. But I didn’t test all languages.

PR #5127 (2021-08-14): Reset timers separate from pwmio
This prevents timer leakage on builds without pwmio.

Fixes #5057
commit 67c6932c (2021-08-14): Reset timers separate from pwmio

This prevents timer leakage on builds without pwmio.

Fixes #5057
closed 2021-08-14 by microDev1