adafruit/circuitpython #5062: Empty boot_out.txt

lesamouraipourpre, 2021-07-26:
### CircuitPython version

```python
adafruit-circuitpython-pimoroni_picolipo_16mb-en_GB-20210725-a9735fb.uf2
```

### Code/REPL

```python
> ls -l /media/user/CIRCUITPY/boot_out.txt
```

### Behavior

-rw-r--r-- 1 user user 0 Jan  1  2020 /media/purpleankh/CIRCUITPY/boot_out.txt

### Description

The latest UF2 from S3 is producing an empty `boot_out.txt`. This is possibly board specific as my PyPortal is fine.

`adafruit-circuitpython-pyportal_pynt-en_GB-20210725-a9735fb.uf2`
`Adafruit CircuitPython 7.0.0-alpha.5-44-ga9735fbe7 on 2021-07-25; Adafruit PyPortal with samd51j20`

> ls -l /media/user/CIRCUITPY/boot_out.txt
-rw-r--r-- 1 user user 100 Jan  1  2000 /media/purpleankh/CIRCUITPY/boot_out.txt

I don't know if it's important but the years are different on the files.

### Additional information

This was originally reported on Discord by #David.G (possibly with a MagTag)

lesamouraipourpre, 2021-07-26:
Further Testing.

I've set the contents of `boot_out.txt` to `Hello` so that it triggers the rewrite, but it sets it to an empty file again.

Connecting the REPL, I get the correct VERSION string:
`Adafruit CircuitPython 7.0.0-alpha.5-44-ga9735fbe7 on 2021-07-25; Pimoroni Pico LiPo (16MB) with rp2040`

lesamouraipourpre, 2021-07-26:
Testing UF2s from alpha-5 forward

```
-rw-r--r-- 1 user user 91 Jan  1  2020 /media/user/CIRCUITPY/boot_out.txt
Adafruit CircuitPython 7.0.0-alpha.5 on 2021-07-21; Pimoroni Pico LiPo (16MB) with rp2040

-rw-r--r-- 1 user user 104 Jan  1  2020 /media/user/CIRCUITPY/boot_out.txt
Adafruit CircuitPython 7.0.0-alpha.5-8-g779b5c185 on 2021-07-23; Pimoroni Pico LiPo (16MB) with rp2040

-rw-r--r-- 1 user user 104 Jan  1  2020 /media/user/CIRCUITPY/boot_out.txt
Adafruit CircuitPython 7.0.0-alpha.5-6-g9c8c4a301 on 2021-07-23; Pimoroni Pico LiPo (16MB) with rp2040

-rw-r--r-- 1 user user 0 Jan  1  2020 /media/user/CIRCUITPY/boot_out.txt
adafruit-circuitpython-pimoroni_picolipo_16mb-en_GB-20210723-beda1f7.uf2
```
The `beda1f7` build seems to the one that introduced the problem.
REPL gives `Adafruit CircuitPython 7.0.0-alpha.5-11-gbeda1f799 on 2021-07-23; Pimoroni Pico LiPo (16MB) with rp2040`

lesamouraipourpre, 2021-07-26:
Testing all the boards I have to hand only two are writing data to `boot_out.txt`

* atmel-samd - PyPortal
* atmel-samd - QT PY

The following all write an empty `boot_out.txt`

* esp32s2 - FunHouse
* nrf52 - Feather Bluefruit Sense
* rp2040 - Pimoroni Pico Lipo 16MB
* rp2040 - Raspberry Pi Pico RP2040

All tested with `7.0.0-alpha.5-44-ga9735fbe7`

dhalbert, 2021-07-26:
This sounds like it might be sleep-related, as all those boards have sleep implementations and `atmel-samd` does not yet.

evildave666, 2021-07-27:
You can add stm to the list of architectures this is occurring on. I've had this issue on local builds for 2 stm32f4 based boards (stm32f411ce_blackpill and stm32f411ce_blackpill_with_flash) since the weekend.

dhalbert, 2021-08-01:
I did a bisect on this on ESP32-S2, and it's this commit that's causing the empty `boot_out.txt`:

https://github.com/adafruit/circuitpython/compare/cf97793af%5E...adafruit:cf97793af (_Add supervisor.get_previous_traceback() function._)

dhalbert, 2021-08-01:
Making this change causes `boot_out.txt` to appear again. A `NULL` was passed before the commit mentioned above
```c
        //bool found_boot = maybe_run_list(boot_py_filenames, &result);
        bool found_boot = maybe_run_list(boot_py_filenames, NULL);
```

Tagging @cwalther for interest and perhaps some insight.

I tried one other thing, which was to insert a one second delay after the `filesystem_flush()` in `run_boot_py()`, and that did not help. I thought perhaps the new exception handling was overwriting the data we wanted to write to the file.

cwalther, 2021-08-02:
Hm, could be that I broke something there - I probably didn’t check whether `NULL` had any special meaning beyond “I’m not interested in the result”. I’ll take a look.

cwalther, 2021-08-02:
I think I figured it out. PR forthcoming.

```diff
diff --git a/main.c b/main.c
index c6801dca7..b35c874c0 100755
--- a/main.c
+++ b/main.c
@@ -633,11 +633,13 @@ STATIC void __attribute__ ((noinline)) run_boot_py(safe_mode_t safe_mode) {
 
     static const char * const boot_py_filenames[] = STRING_LIST("settings.txt", "settings.py", "boot.py", "boot.txt");
     bool skip_boot_output = false;
+    #ifdef CIRCUITPY_BOOT_OUTPUT_FILE
+    FIL file_pointer;
+    #endif
 
     if (ok_to_run) {
 
         #ifdef CIRCUITPY_BOOT_OUTPUT_FILE
-        FIL file_pointer;
         boot_output_file = &file_pointer;
 
         // Get the base filesystem.
```

We were keeping a pointer to an out-of-scope local variable, which was clobbered by the newly added `result`. Replacing `&result` by `NULL` probably caused the compiler to optimize out the unused `result`, preventing that.

PR #5088 (2021-08-03): Fix empty boot_out.txt (#5062)
A pointer to the FIL structure was kept after it went out of scope. Apparently this happened to work until cf97793 added the `result` variable that clobbered it.

Tested on ItsyBitsy nRF52840 Express.

Fixes #5062.
commit 09e68461 (2021-08-03): Fix empty boot_out.txt

A pointer to the FIL structure was kept after it went out of scope.
Apparently this happened to work until cf97793 added the `result`
variable that clobbered it.

Fixes #5062
closed 2021-08-03 by dhalbert