adafruit/circuitpython #5063: Drive wiped Raspberry Pico

DartLazer, 2021-07-26:
### CircuitPython version

```python
CircuitPython 7.0.0-alpha.4
```

### Code/REPL

```python
-
```

### Behavior

I plugged the PiCo into my computer to continue my project and all my files on it are gone. It now contains a code.py with hello world in it and an empty lib folder.

- Why did this happen?
- Is there any way to get my files back?

dhalbert, 2021-07-26:
This can happen because the filesystem became corrupted, which can occur for a variety of reasons. This is a support question: please ask further in https://adafru.it/discord or https://forums.adafruit.com/viewforum.php?f=60.

closed 2021-07-26 by dhalbert

DartLazer, 2021-07-26:
Thank you. I managed to recover the code. It was in the mu editors log in local appdata /python/mu

I eventually wanna put this project in production. Is there a good practise to prevent these file system corruption errors in the future?

Thanks

dhalbert, 2021-07-26:
Please ask in one of those other places and we'll be happy to talk about how corruption might happen. We avoid doing support questions here because it causes email to be sent to a lot of people for each comment.