adafruit/circuitpython #5084: Vectorio doesn't seem to display rotation properly

FoamyGuy, 2021-07-30:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.5 on 2021-07-21; Adafruit PyPortal with samd51j20
```
```

### Code/REPL

```python
import board
import displayio
import vectorio

main_group = displayio.Group()
board.DISPLAY.rotation = 0
#board.DISPLAY.rotation = 90
#board.DISPLAY.rotation = 180
#board.DISPLAY.rotation = 270
board.DISPLAY.show(main_group)

rect = vectorio.Rectangle(width=50, height=1)
rect_palette = displayio.Palette(2)
rect_palette[0] = 0xFFFFFF
rect_palette[1] = 0xFFFFFF

vector_shape = vectorio.VectorShape(shape=rect, x=50, y=50, pixel_shader=rect_palette)
main_group.append(vector_shape)

while True:
    pass
```

### Behavior

When the display is at rotation `0` the rectangle gets drawn as expected a 50px wide, 1px tall line:
[image]

When the display rotation is `90` we get a filled in rectangle, looks like 50px tall and over 100px wide, not sure of exact size:
[image]

When the display rotation is `180` we get a bigger filled rectangle, not sure of exact size:
[image]

When the display rotation is `270` we get a similar rectangle as the `90` degree one, but rotated 90 degrees:
[image]

### Description

I'm not sure how to describe the exact relation, but I believe the x,y location that the tilegrid is placed at is determining the size and shape of the incorrectly drawn larger rectangles on all of the rotations except for `0`

### Additional information

These images depict a PyPortal but this issue has been seen on the MacroPad as well. I believe it would be affecting all devices that support displayio. I suspect there is something inside the core implementation of `vectorio` that is not handling the rotation properly

FoamyGuy, 2021-07-30:
@WarriorOfWire I'm curious if you have any ideas on cause or potential solutions for supporting rotations other than `0`

WarriorOfWire, 2021-07-31:
Thanks for the report! I recall this worked in the dim mists of time. Will dig up a uc, set up this computer for development and take a look.

WarriorOfWire, 2021-07-31:
So the issue is in the way it's computing the dirty area + a displayio semantic. This works fine, but it has really sharp edges.

```python
main_group = displayio.Group()
display = _get_initialized_display()
display.brightness = 1
#display.rotation = 0
display.rotation = 90
#display.rotation = 180
#display.rotation = 270
display.show(main_group)

rect = vectorio.Rectangle(width=20, height=6)
rect_palette = displayio.Palette(2)

# rect_palette.make_transparent(0)     <---------- Change this

rect_palette[0] = 0x00FFFF
rect_palette[1] = 0xFFFF00
```

If you run the above you will see a blue streak up to the yellow rectangle of the appropriate size. What you're seeing is the "dirty rectangle" from the shape. Iirc displayio isn't set up to support multiple dirty areas so you gotta invalidate the whole area from where you were to where you're going.

The blue area is the dirty rect, the yellow is the rectangle.

If you make palette[0] transparent it will do the thing you expect. When palette[0] is not transparent, vectorio dutifully draws in the box. I think it is not well-documented that _you must set palette[0] transparent_.

I will propose a change to the API wherein palette[0] is the color of the shape rather than requiring people to set up the right palettes.