adafruit/circuitpython #5091: RP2040 DEBUG=1 builds don't work

dhalbert, 2021-08-03:
Version: fails with 7.0.0 main at this time and earlier

RP2040 board builds built with DEBUG=1 don't work: they crash. The crash can be seen with `gdb`:
```
(gdb) mon reset
Resetting target
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
isr_hardfault () at sdk/src/rp2_common/pico_standard_link/crt0.S:99
99	decl_isr_bkpt isr_hardfault
(gdb) bt
#0  isr_hardfault () at sdk/src/rp2_common/pico_standard_link/crt0.S:99
#1  <signal handler called>
#2  0x1001d160 in port_init () at supervisor/port.c:97
#3  0x1001cfc2 in main () at ../../main.c:773
(gdb) 
```
If I set a breakpoint at `main()`, it does get past `main()`, and then soon hits the `decl_isr_bkpt isr_hardfault`.

MicroPython built as `CMAKE_ARGS = -DMICROPY_BOARD=$(BOARD) -DCMAKE_BUILD_TYPE=Debug` does not have this problem.

(I originally thought this was some kind of problem with the Segger J-Link and gdb, but that's because I never tried this with a non-debug build. In fact gdb works fine with a non-debug build, but it's not so useful.)

tannewt, 2021-08-03:
This is weird because this is where the memory is being initialized: https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/supervisor/port.c#L97

DavePutz, 2021-08-22:
I ran some tests and found the if the -Og in the Makefile under the "ifeq ($(DEBUG), 1)"  with -O3 resulted in a successful (and debugable) build. @tannewt ;  since a normal build also uses -O3 would this be an acceptable change to PR?

tannewt, 2021-08-23:
@DavePutz Yup, fine with me.

microDev1, 2021-08-24:
Closing... fixed in #5210.

closed 2021-08-24 by microDev1