adafruit/circuitpython #5092: PWM-generated tones may be off-frequency at higher frequencies

dhalbert, 2021-08-04:
### CircuitPython version

```python
recent 7.0.0 versions
```

### Code/REPL

```python
# strange how the sound jumps around at around 1800 and on up
# Tested with CircuitPython 7, not with 6
from time import sleep
from adafruit_macropad import MacroPad
macropad = MacroPad()
for freq in range(200,10000, 50):
    print(f"freq: {freq}")
    macropad.start_tone(freq)
    sleep(.45)
    macropad.stop_tone()
    sleep(.05)
```

### Behavior

he tone jumps randomly at 1850 and up.

### Additional information

Original report here: https://forums.adafruit.com/viewtopic.php?f=59&t=181842

Check PWM frequency calculation and setting.

RufusVS, 2021-08-13:
Through additional testing I found out that frequencies as low as 666 and 670 drop out, though adjacent frequencies play, if this helps.

jepler, 2021-08-24:
```
    uint32_t best_numerator = 0;
    uint32_t best_denominator = 0;
    uint32_t best_error = system_clock;
        
    for (uint32_t denominator = 0xffff; denominator > 0; denominator--) {
        uint32_t numerator = (denominator * sample_rate) / system_clock;
        uint32_t remainder = (denominator * sample_rate) % system_clock;
```

For sample rates above around 65536, the math overflows at `(denominator * sample_rate)`

jepler, 2021-08-24:
Because the macropad sine wave contains 100 samples, problems would start at around 65536/100 Hz

dhalbert, 2021-08-24:
I have been looking at the `PWMOut` frequency setting, which is pretty grainy at lower frequencies.

jepler, 2021-08-24:
It looks like the DMA will happily update the PWM _register_ faster than the PWM's carrier frequency, but you're right that at higher sample rates maybe not all the samples would actually make it out via PWM

PR #5218 (2021-08-25): Fix audio sample rate on rp2040
There's a simple fix, that uses 64 bit arithmetic where necessary. But I chose to make a more complex fix, because I also noticed that it could take up to 100ms for the sample rate to be found! The new algorithm is faster (I think it runs in log(max_denominator) time, sort of like gcd) and can also give _slightly_ more accurate results due to comparing the frequencies as floating point instead of integers—it only makes a difference in cases where the sample rate was already being met to within 1Hz, so not a big difference.

Closes: #5092
closed 2021-08-25 by microDev1